wills106/homeassistant-solax-modbus #379: [BUG]

MrRickwi, 2023-04-11:
**Describe the bug**
In the log there are a lot of entries reporting a blocking interface call.

**Mandatory details**
1. Integration Version: 2023.03.1
2. HA Version: 2023.4.2
3. Inverter Brand: SolaX 7 / Q-Cells
4. Plugin Used: 
5. First 6 digits of Serial:
6. Connection Method: RS485 / Modbus / USB

**Detailed Error Log**
In the log, I see a lot of entries like this one:

``
2023-04-11 17:21:07.709 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 318: return self._client.read_holding_registers(address, count, **kwargs)
``

**Additional context**
None

wills106, 2023-04-11:
This is listed as No2 for [Known Issues](https://github.com/wills106/homeassistant-solax-modbus#known-issues) and a workaround.

MrRickwi, 2023-04-11:
Sorry, didn't see that section. Thanks for the hint. However, would it make sense to remove that blocking call?

wills106, 2023-04-11:
There aren't any active developers connecting over Serial / RS485-USB.

My Inverter has built in Ethernet, so I have never had a need to connect with an RS485 - USB adaptor.

MrRickwi, 2023-04-11:
So, now there are ;-) Will play around with the code and see if I find a fix for that...

wills106, 2023-05-29:
Any joy with improving the Serial connection part of the Integration?

MrRickwi, 2023-05-30:
Unfortunately I had no time to do some work for that...

ZdenekM, 2023-06-02:
I just briefly looked at the code - the thing is that (at least) methods `read_holding_registers` and `read_input_registers` will have to become async, and the `return self._client.read_xxx_registers(address, count, **kwargs)` shall become something like `return await asyncio.run_in_executor(None, functools.partial(self._client.read_xxx_registers, address, count, **kwargs))` and all methods where this is called from should also become async.

FloCodeRian, 2023-11-11:
I´m experiencing the same problem.  Hope there is a fix for it. Blows me up my logs :/ 

2023-11-11 11:31:17.657 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 350: return self._client.read_input_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
2023-11-11 11:31:17.685 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 350: return self._client.read_input_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
2023-11-11 11:31:17.715 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 350: return self._client.read_input_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues

wills106, 2023-11-11:
There is a link to a workaround on the second post of this issue.

FloCodeRian, 2023-11-21:
> This is listed as No2 for [Known Issues](https://github.com/wills106/homeassistant-solax-modbus#known-issues) and a workaround.

Hi there, I did the insert the fix some day´s ago, several restart later, but it didn't worked for me. My logfile´s are still blown up because of: 

2023-11-21 19:31:35.736 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 344: return self._client.read_holding_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
2023-11-21 19:31:35.767 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 350: return self._client.read_input_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues

wills106, 2024-02-22:
I have just released [2024.02.6b3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.6b3) does the work @Zwer2k has done on the async reard / writes prevent the need to hide the blocking call warnings?
[Read the Docs - Detected Blocking Call](https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#detected-blocking-call)

FloCodeRian, 2024-02-22:
Seems like working well for me. Can´t see any blocking calls unto the last running 4 hours

THX

MrRickwi, 2024-02-23:
Testes the Release Version ans it works perfectly without any need to configure logging. Thanks!