wills106/homeassistant-solax-modbus #384: [IDEA] poll registers at different rates

i-jk, 2023-04-19:
**Goal**
Get significantly more frequent updates for important data from the inverter.

**Background**
Presently this integration requests all the modbus registers in every poll, and since there are many of them we are therefore limited to a poll rate that doesn't overload the weak processors in the inverters. Sure, some can go faster than others, but nonetheless I'd like to propose the following way to potentially speed them all up to get semi real-time data.

IMHO there are 2 types of register data:

* Slow - data that doesn't change very frequently, or isn't that interesting. I'm suggesting slower update for these is fine.
* Fast - data that is very interesting and changes very frequently. I'd like to see changes as fast/often as possible

**Suggested approach**
Rather than poll ALL registers every 15 seconds, perhaps we can poll a small subset very frequently, and the rest at a reduced rate. 

One way to make this work, with an example polling rate of 5 seconds, could be:
1. Every poll we always request the 10-20 or so "fast" registers (e.g power, voltage, current and mode data).
* The other, slow registers are broken into small batches (e.g. 5-10 registers) a batch is requested together with the fast group.
* So we request all "fast" + current batch of "slow" every poll.
* The batches of "slow" registers are requested round robin until we have got them all updated, then we start again.

Since slow ones don't change often or aren't as important it's ok if they take `number of slow registers` / `slow batch size` * `polling interval` seconds to fully update all of them. (I'm not sure what these numbers were work out to?)

----
So could this work or is it madness? Are there better ways of getting faster data from the inverters without them freaking out?

I'd be happy to test and help with code if this might be worth pursuing.

infradom, 2023-04-20:
Thanks for the idea ! In my opinion, it would make sense if all fast registers are grouped together. This may depend heavily on the brand if inverter. I have not checked this yet. But if fast registers are spread over the address space, we might en up with more read-block operations than in the current model, which might even increase the load. 
So it depends... I am a bit worried about the added complexity which might introduce new errors and make it more difficult to debug. But it looks feasible to me, we might need to add a 'fast' attribute for registers that need to be polled frequently.
So yes, might be a good idea for some inverters, but is it worth the added complexity?

wills106, 2023-04-20:
Ideally you want to read registers in the largest block size possible. This way you are putting less demand on the inverter.

Most of the docs mention 1second space between the next request. So the smaller the block size of a single request the more requests we need to do.

It also doesn't help that certain Inverters such as Solis aren't fully Modbus compliant.  Modbus RTU and TCP are both able to support a single request of over 120 16bit registers in a single transaction. But unfortunately Inverters such as Solis in the docs recommend only 50 registers and it seems practically closer to 40 in reality.

dakaix, 2023-08-15:
Would love to see this as added as well, I have been trying to increase the polling frequency for specific sensors (such as PV production, battery input/output) but so far with limited success. This is to ensure other loads can respond as quickly as possible when production reduces.

wills106, 2023-09-24:
Going to close this down for now as it would potentially increase the number of requests to the Inverter and would need manual grouping registers to poll quicker.

If your Inverter is struggling to keep up, have a look to see if there is a faster baud rate you can use?

closed 2023-09-24 by wills106

gec75, 2024-04-08:
I also need this, I added a pull request for it.
It works very well for me
https://github.com/wills106/homeassistant-solax-modbus/pull/853