wills106/homeassistant-solax-modbus #385: [BUG] Gen2 Missing Today Grid Import Total Today Grid Export Total for Dashboard Integration

jsantosdias, 2023-04-22:
**Describe the bug**
Integration works great with my Solax SKSU5000. Gen2. I have been able to configure everything, except Energy Dashboard.
I searched and activated several sensors that were inactive. 
I didn't find "Today Grid Import Total" and "Today Grid Export Total" for Dashboard Integration as said in Wiki. I enabled "Grid Import Total" and "Grid Export Total", but both stay always at 0, even if Grid Import and Grid Export work great. Without those sensores I can't use energy dashboard. Please add those missing sensores for Gen 2 Solax.

**Mandatory details**
1. Integration Version: 2023.03.1
2. HA Version: 2023.4.6
3. Inverter Brand: Solax
4. Plugin Used: 
5. First 6 digits of Serial: U50EF2XXXXXXXX
6. Connection Method: Ethernet

jsantosdias, 2023-04-25:
Thank you, alienatedsec, for the reply. What config do you mean? The list of sensors?

alienatedsec, 2023-04-25:
A screenshot from the `energy configuration` config page

[image]

jsantosdias, 2023-04-27:
[image]

This is what I have. I can't use grid values from Solax, because they are always 0, so I am using calculation from Solax integration. 

Also, Today's Solar shows a lot of solar energy at night. 

I think the solution is to make new integrations of sensors from Grid and solar...

jsantosdias, 2023-04-27:
I am using HASS Solax integration (Wifi) with calculations:

`  - sensor:
      - name: "Consumo_Rede"
        unit_of_measurement: "W"
        device_class: power
        state: >
          {% if states('sensor.solax_8caa5352_exported_power') > "0" %}
            {{ 0 }}
          {% else %}
            {{ states('sensor.solax_8caa5352_exported_power') | float(default=0) | abs }}
          {% endif %}`
But I think using data from modbus would be more accurate...

alienatedsec, 2023-04-27:
Do you have those two sensors, which I believe can be used for your import and export?
https://github.com/wills106/homeassistant-solax-modbus/blob/a4287309c409e996202cbe5b173327853346540f/custom_components/solax_modbus/plugin_solax.py#L3519-L3521

and

https://github.com/wills106/homeassistant-solax-modbus/blob/a4287309c409e996202cbe5b173327853346540f/custom_components/solax_modbus/plugin_solax.py#L3504-L3506

closed 2025-10-04 by github-actions[bot]