wills106/homeassistant-solax-modbus #386: [FR] Write support Growatt

wills106, 2023-04-25:
I Have connection after upgrade to [2023.03.2b23] is it possible to add function to change operation types:
1 Grid First - where the system prioritises export before sending to battery/house
2 Battery First - where the system prioritises the charging of batteries before anything else.
3 Load First - where the system prioritises supplying the house before charging the batteries or exporting to the grid.

https://www.amosplanet.org/wp-content/uploads/2022/12/%E9%80%9A%E4%BF%A1%E5%8D%8F%E8%AE%AEMAX_Growatt-Inverter-Modbus-RTU-Protocol_II-V1_13_20190116-English.pdf

Page 41 register 118

_Originally posted by @super-bedst in https://github.com/wills106/homeassistant-solax-modbus/issues/377#issuecomment-1501741461_

wills106, 2023-04-25:
@super-bedst
What do the timing windows do on "Grid First"? Does it prevent you from discharging the battery or does it prevent you from charging the battery?

wills106, 2023-04-29:
@super-bedst have you been able to look at the new write support in [2023.03.2b24](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b24) yet?

wills106, 2023-04-29:
I have added in Load Times into [2023.03.2b25](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b25) also renamed the Stop Times to End Times.

super-bedst, 2023-04-29:
@wills106 i have installed 2023.03.2.b25  it now, and done a quick test, ad it seems to work. i Hope i can fine more time in next week to test more deeply :-)

super-bedst, 2023-04-29:
I miss the kwh information for battery charge and discharge and the percentage of battery capacity. i am not 100% sure on the register, maybee 1044-1066 (page51) or 112-120 (page 40)

wills106, 2023-04-29:
I'll have a look into those registers.

Do your other registers look ok? Are they scaled correctly?

Ie Solar PV is 1900w instead of 19000w etc. Unless you happen to have a 19kW system?

wills106, 2023-05-01:
1044 - 1050 are already implemented.
I can add the ones from 1052 - 1065 in.
I'll have a look at those others as well.

wills106, 2023-05-06:
@super-bedst I have just released [2023.05.2b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.05.2b2) and added in 1052 - 1062
Shouldn't "Battery SOC" already be returning the Battery capacity as a percentage?

super-bedst, 2023-05-07:
Hi
Total House Load (Local load consumption is a better name) and Grid Power are both factor 10 bigger than real.
an another funny thing. if i set the charge time to 13:15, after 20 sec it will be persented in the list as 15:13, i lookes like it switches between hours and mins.

If it is possible if the actual value in the inverter can be shown as beginning value in the dropdown for editing the value

super-bedst, 2023-05-07:
Hi
I also miss the export limit switch and the limit set, register 122 and 123.

wills106, 2023-05-08:
> Total House Load (Local load consumption is a better name)

I have kept them consistent between all Inverter brands.

> export limit switch and the limit set, register 122 and 123

Is 123 really +/- 1000w? That seems low to me.

wills106, 2023-07-31:
Can this be closed down? Or are there still issues?

wills106, 2023-07-31:
I'm going to close this issue down on consolidate to #471

closed 2023-07-31 by wills106