wills106/homeassistant-solax-modbus #388: [BUG]

bluedevil500, 2023-04-27:
**Mandatory details**
1. Integration Version: current version
2. HA Version: Home Assistant 2023.4.6 Supervisor 2023.04.1 Operating System 10.0 Frontend 20230411.1 - latest 
3. Inverter Brand: SolaX X3 G4.2
4. Plugin Used: 
5. First 6 digits of Serial: H34A12
6. Connection Method: Off-Grid

**Detailed Error Log**
Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:484
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 07:03:00 (1 occurrences)
Last logged: 07:03:00

Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 409, in read_modbus_data
    res = self.read_modbus_registers_all()
  File "/config/custom_components/solax_modbus/__init__.py", line 493, in read_modbus_registers_all
    res = res and self.read_modbus_block(block, 'holding')
  File "/config/custom_components/solax_modbus/__init__.py", line 484, in read_modbus_block
    if ((descr.ignore_readerror != True) and (descr.ignore_readerror !=False)) : self.data[descr.key] = descr.ignore_readerror # return something static
AttributeError: 'dict' object has no attribute 'ignore_readerror'

the error occurred this morning

[image]

wills106, 2023-04-27:
Have you just updated the Integration and got this error?
Or have you been on the same version for a while and this error has just started?

bluedevil500, 2023-04-28:
HA was rebuilt yesterday on version 10 incl. your version 2023.03.1.23

wills106, 2023-05-06:
Are you still getting this error? Or has it gone now?

wills106, 2023-05-29:
Closing down as no further response to issue.
if you are still having issues feel free to reopen or to create a new issue.

closed 2023-05-29 by wills106