wills106/homeassistant-solax-modbus #389: Function request

bluedevil500, 2023-04-27:
Question:

Is the battery heating function already in the planning stage?

[image]

wills106, 2023-04-27:
Do you own the 3kWh batteries with the built in heating element then?

bluedevil500, 2023-04-28:
Our system consists of three 12KW SolaX inverters with 4 x T30 batteries each.
Once master and two as slave in parallel operation. The heating function is integrated.

wills106, 2023-04-28:
Ok I just wanted to double check you had 3kWh batteries and you weren't expecting the heating to function on the 5.8kWh ones.
I'll add in support.

wills106, 2023-04-28:
> Once master and two as slave in parallel operation.

Have you tried out Parallel mode in the Integration?
It's not fully implemented yet, but you should see the combined PV Production etc.

bluedevil500, 2023-04-29:
The T58 has no heater but only the T30 battery. Attached is a screenshot of the parallel mode. Runs fine both in on-gird and off-grid mode.
[image]

wills106, 2023-04-29:
I don't think that screenshot is the actual Parallel Mode.
The sensors should start PM

If you reconfigure the Integration with the Paralele Mode button ticked you should be able to see the combined output of all of the Inverters from polling the Master Inverter.

I'll work on adding in the Battery Heating Support.

wills106, 2023-04-29:
I have added in Battery Heating Times into [2023.03.2b25](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b25)

bluedevil500, 2023-04-30:
Hey Will,

thank you very much for your great work you have helped me a lot. if it does not bother you, it would be great if you still realize the following Fuktionen:

- Security
- Grid parameters
- ExternalGen
- HotStandby Setting
- External BAT FUNC
- PV Connection

these are the last functions that would be missing.

Unfortunately I am not fit in Python, otherwise I would have done it myself.
[image]
[image]

wills106, 2023-05-03:
Are the Grid Parameters something that really needs to be controlled from the Integration / people should be altering?

bluedevil500, 2023-05-03:
i agree, this is not for endsusers.
It would have been very helpful for me as an installer.

closed 2023-05-03 by bluedevil500

wills106, 2023-05-03:
I could look at adding an extra tick box for "Warning Installers" then the more dangerous stuff could go in there?

Which would actually be useful to you?

bluedevil500, 2023-05-04:
a "Warning installer" checkbox would probably be the most plausible. The "Security" and "Grid parameters" should probably be in there. Unfortunately, what we are currently missing the most is the "Charger" entry "Charger upper limit (%). The background to this is that there is the case that, for example, only the master has PV modules connected and the slavers not. In order to achieve an even charging of the batteries, this value must be adjusted daily in order to evenly approach the 100% of all batteries.

wills106, 2023-05-04:
There is a bit of info on Upper Limit on #363

ChequeM8, 2023-06-18:
I purchased my T58 batteries around January 2023 and Solax inform me they have the battery heating functionality.  I haven't tried it yet as my inverter needs a firmware update.

wills106, 2023-06-19:
I was under the impression only the T30 batteries have the heating element.