wills106/homeassistant-solax-modbus #391: Lost all modbus controls after doing fresh install of ubuntu server, hacs, home assistant and this modbus install

UK-JustinSmall, 2023-04-29:
I am getting following on debug logging, after reload

2023-04-29 19:07:40.059 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-04-29 19:07:40.062 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-04-29 19:09:12.001 ERROR (MainThread) [frontend.js.latest.202304111] :0:0 ResizeObserver loop completed with undelivered notifications.
2023-04-29 19:10:58.763 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2023-04-29 19:10:58.863 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2023-04-29 19:10:58.867 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find serial number, even not for MIC
2023-04-29 19:10:58.870 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2023-04-29 19:15:22.676 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2023-04-29 19:15:22.767 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2023-04-29 19:15:22.772 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find serial number, even not for MIC
2023-04-29 19:15:22.776 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown

Any ideas on how to get this fixed? I cannot see any solax entities other than sensor.solax_inverter_temperature and that has no value assigned either.

shaft321, 2023-04-29:
I can 2nd this. Today at around 1pm i lost all the sensors. I'm running  HA on Oracle VM VirtualBox
Home Assistant 2023.4.6
Supervisor 2023.04.1
Operating System 10.1

UK-JustinSmall, 2023-04-29:
I just managed to fix this on my side. With a little help.

Try going into modbus configuration and changing Polling frequency to 10 and check the Power Options (EPS) box. Did that and as if by magic 187 entities have appeard and are showing up on the Overview screen

UK-JustinSmall, 2023-04-29:
still getting 

2023-04-29 19:15:22.772 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find serial number, even not for MIC

2023-04-29 19:15:22.776 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown

2023-04-29 19:45:38.132 ERROR (MainThread) [frontend.js.latest.202304111] :0:0 ResizeObserver loop completed with undelivered notifications.

though so keeping bug open

shaft321, 2023-04-29:
> 

i'll give that a go. thanks.

shaft321, 2023-04-29:
changing the poll freq to 10 and checking the EPS didn't work for me. I'm still only seeing 4 entities. :(

wills106, 2023-04-29:
Are you both using a PocketWiFi v1 or v2 by any chance?

shaft321, 2023-04-29:
> Hi, I'm using pocketwifi v3.0 (black one)
I've just opened a new issue with all the info you ask for.

UK-JustinSmall, 2023-04-29:
I'm using the black one too v3. Appears two entities missing and 6 with no values assigned for my dashboard namely 

sensor.solax_grid_import_total - Not showing
sensor.solax_grid_export_total - Not showing
sensor.solax_pv_power_1 - Present but no value
sensor.solax_pv_current_1- Present but no value
sensor.solax_pv_voltage_1- Present but no value
sensor.solax_pv_power_2- Present but no value
sensor.solax_pv_current_2- Present but no value
sensor.solax_pv_voltage_2- Present but no value

wills106, 2023-04-30:
I would have a look at https://github.com/wills106/homeassistant-solax-modbus/issues/393#issuecomment-1528881968 and do the same.

UK-JustinSmall, 2023-04-30:
Great thank you the pv values populated today when the sun came out :-)

Seems my black dongle is on version 3.001.02 emailed solax for a firmware update beyond  V3.004.03

See if that resolves the issue with the other two values sensor.solax_grid_import_total and sensor.solax_grid_export_total not being available

wills106, 2023-05-02:
Did you manage to get you PocketWiFi updated?

UK-JustinSmall, 2023-05-02:
They say they have updated it, but when I log into the inverter it is still showing as 3.001.02

However, I removed the modbus application, and re-installed it. Checked two boxes Emergency Power Options (EPS) and Dry Contact Box (Gen4) and now it is showing 217 entities so appears to have fixed this issue

closed 2023-05-02 by UK-JustinSmall