wills106/homeassistant-solax-modbus #393: [BUG]Lost all sensors

shaft321, 2023-04-29:
**Describe the bug**
the integration has been working flawlessly for a few weeks with my Solax Hybrid G4 inverter but today at around 13:00 it stopped working. I can no longer see all the sensors. I have tried reinstalling but still the same issue.

**Mandatory details**
1. Integration Version: 2023.03.2b25
2. HA Version: 2023.3.6
3. Inverter Brand: Solax Hybrid G4
4. Plugin Used: not sure what this means.
5. First 6 digits of Serial: H4502A
6. Connection Method: TCP (IP address)

**Detailed Error Log**
2023-04-29 22:36:09.364 INFO (MainThread) [custom_components.solax_modbus.config_flow] validating base: {'name': 'SolaX', 'interface': 'tcp', 'read_modbus_addr': 1, 'plugin': 'solax', 'scan_interval': 15, 'read_eps': False, 'read_dcb': False, 'read_pm': False}
2023-04-29 22:36:09.365 INFO (MainThread) [custom_components.solax_modbus.config_flow] validating base config for SolaX: pre: {'name': 'SolaX', 'interface': 'tcp', 'read_modbus_addr': 1, 'plugin': 'solax', 'scan_interval': 15, 'read_eps': False, 'read_dcb': False, 'read_pm': False}
2023-04-29 22:36:17.594 INFO (MainThread) [custom_components.solax_modbus.config_flow] validating host: returning data: {'host': '192.168.5.101', 'port': 502}
2023-04-29 22:36:17.594 INFO (MainThread) [custom_components.solax_modbus.config_flow] title configflow solax_modbus name: {'name': 'SolaX', 'interface': 'tcp', 'read_modbus_addr': 1, 'plugin': 'solax', 'scan_interval': 15, 'read_eps': False, 'read_dcb': False, 'read_pm': False, 'host': '192.168.5.101', 'port': 502}
2023-04-29 22:36:17.595 INFO (MainThread) [homeassistant.setup] Setting up solax_modbus
2023-04-29 22:36:17.595 INFO (MainThread) [homeassistant.setup] Setup of domain solax_modbus took 0.0 seconds
2023-04-29 22:36:17.597 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solax
2023-04-29 22:36:17.667 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: trying to determine inverter type
2023-04-29 22:36:18.078 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2023-04-29 22:36:18.080 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX 0x0 serial number before potential swap: None
2023-04-29 22:36:18.157 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2023-04-29 22:36:18.158 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX 0x300 serial number before potential swap: None
2023-04-29 22:36:18.158 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find serial number, even not for MIC
2023-04-29 22:36:18.159 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2023-04-29 22:36:18.171 INFO (MainThread) [homeassistant.components.select] Setting up select.solax_modbus
2023-04-29 22:36:18.176 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solax_modbus
2023-04-29 22:36:18.179 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8 inverter_temperature to block with start 0x8
2023-04-29 22:36:18.179 INFO (MainThread) [custom_components.solax_modbus.sensor] SolaX returning input block: 0x8 0x9 [8]
2023-04-29 22:36:18.179 INFO (MainThread) [custom_components.solax_modbus.sensor] computedRegs: {}
2023-04-29 22:36:18.183 INFO (SyncWorker_5) [homeassistant.loader] Loaded button from homeassistant.components.button
2023-04-29 22:36:18.185 INFO (SyncWorker_1) [homeassistant.loader] Loaded number from homeassistant.components.number
2023-04-29 22:36:18.191 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solax_modbus entity: sensor.solax_inverter_temperature
2023-04-29 22:36:18.210 INFO (MainThread) [homeassistant.setup] Setting up button
2023-04-29 22:36:18.212 INFO (MainThread) [homeassistant.setup] Setup of domain button took 0.0 seconds
2023-04-29 22:36:18.213 INFO (MainThread) [homeassistant.setup] Setting up number
2023-04-29 22:36:18.214 INFO (MainThread) [homeassistant.setup] Setup of domain number took 0.0 seconds
2023-04-29 22:36:18.229 INFO (MainThread) [homeassistant.components.button] Setting up button.solax_modbus
2023-04-29 22:36:18.233 INFO (MainThread) [homeassistant.components.number] Setting up number.solax_modbus
2023-04-29 22:36:18.236 INFO (MainThread) [custom_components.solax_modbus.button] hub.wakeuButton: SolaxModbusButtonEntityDescription(key='battery_awaken', device_class=None, entity_category=None, entity_registry_enabled_default=True, entity_registry_visible_default=True, force_update=False, icon='mdi:battery-clock', has_entity_name=False, name='Battery Awaken', translation_key=None, unit_of_measurement=None, allowedtypes=0, register=86, command=1, blacklist=None, write_method=1, value_function=None, autorepeat=None)
2023-04-29 22:36:18.241 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new button.solax_modbus entity: button.solax_battery_awaken
2023-04-29 22:36:18.242 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new button.solax_modbus entity: button.solax_unlock_inverter
2023-04-29 22:36:18.243 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new button.solax_modbus entity: button.solax_unlock_inverter_advanced
2023-04-29 22:36:33.205 INFO (MainThread) [custom_components.solax_modbus] exception Modbus Error: [Connection] ModbusTcpClient(192.168.5.101:502): Connection unexpectedly closed 1.7881393432617188e-05 seconds into read of unbounded read bytes without response from unit before it closed connection : SolaX cannot read input registers at device 1 position 0x8
NoneType: None

**Additional context**
I have pinged the ip address of the inverter to make sure and do get a response so the ip address is correct.

wills106, 2023-04-29:
Did you update anything on HomeAssistant around this time?

It might be worth trying a power cycle on the Inverter. Either do it during the day when Solar PV is active, or if you have enough battery capacity you could try it now.

shaft321, 2023-04-29:
I updated home assistant yesterday. yes i have enough battery. but what is a power cycle?

wills106, 2023-04-29:
Turn the Inverter Off and Back On again.

I have noticed a few people have lost connection lately and needed to Power Cycle their Inverters.
I need to try and look to see if there is any pattern. Might be a Firmware issue on the Pocket WiFi for example?

shaft321, 2023-04-29:
ah ok thank you. i'll try that now once i figure out the correct process of turning it off and on. I'm a bit nervous i might mess things up.

wills106, 2023-04-29:
If you use the LCD on the Inverter you should see something like "System Switch" in the menu.
You can use that to turn the Inverter off, then turn the main breakers off. Then when the LCD display has gone off you can turn it on again.

When you use the "System Switch" there maybe a clunk. That's just the contactors and is normal. Similar to when the Inverter wakes up in the morning, it probably cycles through the contactors (at least on Gen3)

wills106, 2023-04-29:
Or can you log into the PocketWiFi and reboot it?

I don't own one, so I don't know.

shaft321, 2023-04-29:
ok. thanks.   I'll try it in the morning and will let you know.

shaft321, 2023-04-29:
I couldn't wait till the morning. good news....I've restarted the inverter and reinstalled the integration and I now have all 180 entities back. 
thank you Will.

closed 2023-04-29 by shaft321

wills106, 2023-04-29:
Do you know what Firmware your PocketWiFi is on?

shaft321, 2023-04-29:
3.001.02

wills106, 2023-04-30:
The minimum officially supported by SolaX is meant to be V3.004.03 and I believe there is a newer version available.
So it might be worth emailing SolaX to ask them to upgrade the PocketWiFi for you.