wills106/homeassistant-solax-modbus #396: Solax Modbus: Unable to set values

otava5, 2023-05-05:
**Describe the bug**
After the remote firmware update of the inverter, the values cannot be set. Always after setting, the value will be returned.
Ejecting, reinserting dongle/restarting router/restarting HA didn't help

**Mandatory details**
1. Integration Version: 2023.05.1
2. HA Version: 2023.1.5
3. Inverter Brand: Solax X3 G4 Hybrid
4. Plugin Used:
5. First 6 digits of Serial: H34C10
6. Connection Method: Wifi Dongle 3.0 black

wills106, 2023-05-06:
Have you tried "Unlocking" the Inverter as mention just before the known issues section of the [Readme](https://github.com/wills106/homeassistant-solax-modbus#known-issues)?

otava5, 2023-05-06:
Thank you sir it helped ;-)
Can it be locked back??

wills106, 2023-05-06:
Not at the moment.
I can create a "Select" so you can choose between the two Unlocked modes and Locked.

otava5, 2023-05-06:
It would be great...
Thanks for the advice, we can close...

wills106, 2023-05-06:
I have added select "Lock State" into the new Beta release [2023.05.2b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.05.2b1)

wills106, 2023-05-29:
Closing down as [2023.05.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.05.2) addresses this issue.

closed 2023-05-29 by wills106