wills106/homeassistant-solax-modbus #398: [BUG] Bad items and Divions needs to be higher for growatt hybrid sph3600

fontexD, 2023-05-06:
Grid power seems to be wrong i got a esp on my Meter directly connected, currently my solax says 0W where it should say quite higher number

https://ibb.co/k66B2Bw

these also seems to be wrong
https://ibb.co/mhkCHkw

wills106, 2023-05-08:
Just released [2023.05.2b3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.05.2b3) that should correct "Total House Load".

fontexD, 2023-05-08:
awesome!

i tried to fix it myself by changing the scale but after i changed the number adding a  0 it totaly failed to load the integration

what did u change ? if u can show :)

wills106, 2023-05-08:
https://github.com/wills106/homeassistant-solax-modbus/commit/86a87fe23188951fa0c11beacb4f717862c3b2f4#diff-046c299de9c82c23a2b80aab480e4764b6d768dbf26f783a29263fda92774c37R1637

closed 2023-07-31 by wills106