wills106/homeassistant-solax-modbus #399: Can't see 'today's' entities in Home Assistant's Energy Dashboard

maxdraki, 2023-05-07:
**Describe the bug**
(Firstly - Thank you for this integration - I was trying to manually extract all the data from my inverter before finding this integration - I added the integration, configured it and magically loads of statistics were available! Nice work - thank you!)

I can't see 'today's' entities in Home Assistant's Energy Dashboard. 
While I have loads of statistics, I can't actually see the metrics you refer to on the [wiki](https://github.com/wills106/homeassistant-solax-modbus/wiki/Energy-Dashboard's#electricity-grid) -  specifically:

**Electricity Grid**
Grid consumption - `sensor.solax_today_s_import_energy`
Return to grid - `sensor.solax_today_s_export_energy`

**Solar Panels**
Solar production - `sensor.solax_today_s_solar_energy`

**Mandatory details**
1. Integration Version: 2023.05.1
2. HA Version: 2023.5.2
3. Inverter Brand: SolaX
4. Plugin Used: 
5. First 6 digits of Serial: U50EHB
6. Connection Method: Ethernet

Here's a screenshot of the entities - filtered by 'solax' and ordered by 'entity ID'
I would expect the statistics to show up here

- Disclaimer - I'm pretty new to this, so maybe getting confused. Any help would be much appreciated. Thank you

[image]

jsantosdias, 2023-05-07:
Hello! Same with me. I have already submitted bug... github.com/wills106/homeassistant-solax-modbus/issues/385 
But you can make "integration" in Configuration --> Services --> Helpers New Helper

wills106, 2023-05-08:
If you enable "Grid Import Total" and "Grid Export Total" you should be able to use them in the dashboard.

maxdraki, 2023-05-08:
Thank you @wills106 - yes, I managed to get that working, but was getting confused about the names of the statistics. Appreciate the help - and kudos for the integration! Thank you

wills106, 2023-05-08:
I need to update the Wiki for most pages.

That Energy Dashboard page is only relevant for the Gen3 & Gen3 SolaX at the moment.
Need give examples for the other brands Growatt, Sofar & Solis etc.

wills106, 2023-05-08:
Just updated the Wiki so I'll mark this issue as closed.

closed 2023-05-08 by wills106