wills106/homeassistant-solax-modbus #400: Sofar: reading serial number from address xxxx failed; other address may succeed

sanderbaks, 2023-05-07:
Hi All,

First:
Inverter: SOFAR 12KTL-G3 fw: SS2ES112N3U010
Logger: LSE-3 fw: ME_0C_270A_1.05

I can access it with a web browser.
I really want to put the data into my Home Assistant install. I cant get it to work and it is driving me mad!

Used the "Sofar" configuration setting and the "Sofar_old" config.  

It cant read the inverter.

What can I do? How can I find the correct paramer?

thanks!

Sander

darkrain-nl, 2023-05-07:
HI there,

I have the same inverter, you should not use sofar_old, but the normal sofar plugin.

The serial should then be read fine, keep in mind the inverter needs to be online, so there needs to be sun....

sanderbaks, 2023-05-07:
What port did you use! 8899?----------------Sander BaksCommercial & editorial photographerwww.sanderbaks.comwww.hotelcreatives.comT: 06-46188558

darkrain-nl, 2023-05-07:
The port you should use is TCP 502

sanderbaks, 2023-05-07:
Still does not work .. and I still have some sun..

Error:

Sofar: reading serial number from address 0x445 failed; other address may succeed

darkrain-nl, 2023-05-07:
Ok what device are you using to get the data out of the inverter?
Is this one of the listed WaveShare devices or are you trying to do that via the wifi of the inverter itself?

sanderbaks, 2023-05-07:
I am using a the Sofar LSE-3 logger.  So ethernet.
I can access the logger directly. So cables/connection etc etc are working. Ip is correct
I thought that it was Port 8899

I also see the device using the solarmanpro website.

It is so strange this issue

darkrain-nl, 2023-05-07:
Hmm ok I'm using a WaveShare device...

But is that port exposing the RS485 (modbus) protocol, cause that is what this integration expects...

sanderbaks, 2023-05-07:
I guess not ..  I think I ma on the wrong track with this one.

wills106, 2023-05-07:
Some data loggers expose Modbus over TCP, some don't.
Unfortunately I don't fully know how each brand of Inverter and sub component needs to connect.

It seems your data logger doesn't provide Modbus over TCP and like @darkrain-nl mentions you possibly need a Waveshare to connect to Modbus over RS485 (RTU) on the Inverter. Or you could use one of those cheap USB - RS485 adaptors.

sanderbaks, 2023-07-05:
> 502

> Hmm ok I'm using a WaveShare device...
> 
> But is that port exposing the RS485 (modbus) protocol, cause that is what this integration expects...

Hi!  It is me again.  I found that the connection should send Modbus and using Solax it does find something.
I just need to modify the settings I guess to be able to see stuff

sanderbaks, 2023-07-09:
@wills106 

According to specs by Solarman, this LSE-3 logger is communicating RS485.
If not, is there a way to check? Some kind of plugin to catch raw data coming from this IP/Port?

Thanks!

[image]

wills106, 2023-09-03:
I'm going to close this down as it isn't an Integration issue.
Feel free to start a discussion if you are still having issues connecting. Hopefully another Sofar owner can assist.

closed 2023-09-03 by wills106