wills106/homeassistant-solax-modbus #401: [BUG] - Getting randomly false readings - peaks - Solax X3 G4 / WaveShare

shaarkys, 2023-05-07:
**Describe the bug**
Getting randomly false readings - peaks, see examples - it's happening after I have switched to Waveshare RS485 to RJ45 Ethernet DIN Rail Mount - RS485 TO ETH (B) , because previously used Wifi Dongle v3 even on any latest FW dongle version get disconnected randomly (using HA in parallel to Homey , so my idea was to get data into HA from WaveShare and to Homey via Wifi Dongle v3)

**Mandatory details**
1. Integration Version: 2023-05-2b2 (this happens since switching to WaveShare) - 2 weeks ago
2. HA Version:  2023.5.2
3. Inverter Brand: Solax X3 G4
4. Plugin Used:
5. First 6 digits of Serial:H34A10
6. Connection Method: Waveshare RS485 to RJ45 Ethernet DIN Rail Mount - RS485 TO ETH (B)

**Detailed Error Log**
`2023-05-07 16:48:58.457 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 410, in read_modbus_data
    res = self.read_modbus_registers_all()
  File "/config/custom_components/solax_modbus/__init__.py", line 494, in read_modbus_registers_all
    res = res and self.read_modbus_block(block, 'holding')
  File "/config/custom_components/solax_modbus/__init__.py", line 485, in read_modbus_block
    if ((descr.ignore_readerror != True) and (descr.ignore_readerror !=False)) : self.data[descr.key] = descr.ignore_readerror # return something static
AttributeError: 'dict' object has no attribute 'ignore_readerror'
2023-05-07 16:52:19.620 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 579, in state
    numerical_value = int(value)
ValueError: invalid literal for int() with base 10: '2023-05-07 10:50:49'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 204, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 349, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 146, in async_update_listeners
    update_callback()
  File "/config/custom_components/volkswagencarnet/__init__.py", line 344, in async_write_ha_state
    or str(self.state or STATE_UNKNOWN) != str(prev.state)
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 583, in state
    raise ValueError(
	
	2023-05-07 17:24:59.977 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 410, in read_modbus_data
    res = self.read_modbus_registers_all()
  File "/config/custom_components/solax_modbus/__init__.py", line 494, in read_modbus_registers_all
    res = res and self.read_modbus_block(block, 'holding')
  File "/config/custom_components/solax_modbus/__init__.py", line 485, in read_modbus_block
    if ((descr.ignore_readerror != True) and (descr.ignore_readerror !=False)) : self.data[descr.key] = descr.ignore_readerror # return something static
AttributeError: 'dict' object has no attribute 'ignore_readerror'
2023-05-07 17:27:28.648 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 579, in state
    numerical_value = int(value)
ValueError: invalid literal for int() with base 10: '2023-05-07 17:25:27'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 204, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 349, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 146, in async_update_listeners
    update_callback()
  File "/config/custom_components/volkswagencarnet/__init__.py", line 344, in async_write_ha_state
    or str(self.state or STATE_UNKNOWN) != str(prev.state)
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 583, in state
    raise ValueError(
	
	2023-05-07 22:22:33.943 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 410, in read_modbus_data
    res = self.read_modbus_registers_all()
  File "/config/custom_components/solax_modbus/__init__.py", line 494, in read_modbus_registers_all
    res = res and self.read_modbus_block(block, 'holding')
  File "/config/custom_components/solax_modbus/__init__.py", line 485, in read_modbus_block
    if ((descr.ignore_readerror != True) and (descr.ignore_readerror !=False)) : self.data[descr.key] = descr.ignore_readerror # return something static
AttributeError: 'dict' object has no attribute 'ignore_readerror'`

**Additional context**
Using HA connected via WaveShare via COM port and Homey Athom connected via Wifi Dongle v3, not sure if that might be a problem

Waveshare settings (not sure if multihost might be causing any such issues, currently having it disabled for futher tests.)

[image]

shaarkys, 2023-05-07:
Today between - 18-19

[image]

[image]

shaarkys, 2023-05-07:
Day before

[image]

shaarkys, 2023-05-07:
Also tried with higher baudrate configured as well on Solax inverter and got rather higher frequency of issues - switched back to baudrate  19200

shaarkys, 2023-05-08:
Reviewed text as I mixed connection methods between HA and Athom Homey, sorry - not it's reflecting right state.

jevburchell, 2023-05-15:
I'm seeing similar behaviours.  Entities dropping to zero or sky-high for a few minutes each day.  S3-WiFi-ST with 485 hub interface.

shaarkys, 2023-05-16:
> edit: yes it is,

But it's a pain ;-( I gave up as I would have to fix up 10 entries daily.

wills106, 2023-05-20:
@shaarkys Did these errors only start when you were using both the Pocket WiFi and the Waveshare?
The Inverter possibly doesn't like two active connections to it.

Are you trying to control the Inverter from both HA and Homey, or do you only want the read only data going to Homey for example?

If you only need one master another option to try is disconnect from the Pocket WiFi through Modbus.
Use the Waveshare with the settings shown on the Wiki connected to the master Home Automation system.
Use https://github.com/infradom/ha-addon-modbusspy to passively monitor the Modbus communications between the Waveshare and the Inverter.
Then connect the read only Home Automation system to the Modbus TCP connection modbusspy provides.

alienatedsec, 2023-05-20:
> I'm seeing similar behaviours.  Entities dropping to zero or sky-high for a few minutes each day.  S3-WiFi-ST with 485 hub interface.  

@jevburchell is this the @jumpy07 design?

jevburchell, 2023-05-20:
> > I'm seeing similar behaviours.  Entities dropping to zero or sky-high for a few minutes each day.  S3-WiFi-ST with 485 hub interface.
> 
> @jevburchell is this the @Jumpy07 design?

It is yes, I've had a chance now to do a little bit of troubleshooting and haven't seen the behaviour since.  Are you also running this setup?

alienatedsec, 2023-05-20:
@jevburchell no, I followed my own logic https://github.com/alienatedsec/solis-ha-modbus-cloud but your connection method is also great.

I have seen your video (I assume it's yours) - great work. 

What have you changed to see the improvement? As soon as there are communication issues between Waveshare (the hub in your instance) and the inverter, there are issues with the integration registering false readings.

jevburchell, 2023-05-20:
Ah brilliant it's yourself.  Actually your waveshare design spurred me on to look more closely at building a Modbus integration so cheers :)  And yeah I put out a video yesterday, a bit David Brent but some of it might be useful to folk.

Have you seen any data irregularities on HA Solax with your Waveshare design?  

I have a couple of theories, but what has worked so far is to completely remove any trace of the Solis HA integration.  I was still using that for a couple of data points but actually completely getting rid of it seems to have solved the issue.  Data collisions between that integration's requests and the datalogger seem to be the likely cause.  Possibly any proprietary integration would have the same effect on different inverter brands, it's hard to tell.

If I see anything pop up again I'll let you know, and if so I'll be looking into aliasing between poll rates, software flow control and power draw of the Modbus unit.

alienatedsec, 2023-05-20:
> Have you seen any data irregularities on HA Solax with your Waveshare design?

@jevburchell Yes, a very early stage with previous versions it was regular - see #340 

Nowadays it's less frequent - once a month on average.

jevburchell, 2023-05-20:
>  - see #340

I actually remember reading this thread before ordering any parts to build, I hadn't connected the dots that this might be the same issue.  You go into way more detail that I'm able to at this end as the 485 hub I'm using doesn't have nearly as many options to configure.  

I'm a little unclear on what the bottom line is/was for you on that one - do you have a firm idea of why it was happening on your setup?  I see block size and polling rate were key factors

alienatedsec, 2023-05-20:
> do you have a firm idea of why it was happening on your setup?

@jevburchell some improvements to integration helped a lot, and as you mentioned, reducing the block size was the key. I am suspecting a reduced baud rate of `9600` when compared with other inverter brands can be a bottleneck.  I may try increasing it both on the inverter and on the `TCP Server` of the Waveshare. The `TCP Client` serial config should match the datalogger defaults `9600`. It is likely to be difficult to set different serial settings on the Hub for each master (?)

jevburchell, 2023-05-20:
@alienatedsec  The hub is fairly dumb and will simply act as an intermediary between masters and slave - so as long as all three nodes agree on their baud rate then it will happily chug along providing hardware flow control (the limit of which I am unsure).  It's straightforward to change the baud rate in the Elfin device, and I believe equally so in the inverter - but do you know how it's done in the datalogger?  Just having a look at its interface now and I don't see an option for it

alienatedsec, 2023-05-20:
> @alienatedsec  The hub is fairly dumb and will simply act as an intermediary between masters and slave - so as long as all three nodes agree on their baud rate then it will happily chug along [...]

@jevburchell 
That's why it will have to stay on 9600. Potentially, we will need to ask Solis about datalogger settings amendments. It is possible on DLS-W and DLS-L in the config_hide page.

jevburchell, 2023-05-20:
Ah.  Ok, I suspect I know what the answer to that will be..

alienatedsec, 2023-05-20:
> Ah.  Ok, I suspect I know what the answer to that will be..  

I have now two waveshares at different baud rates

TCP Server at 19200 and the inverter the same
TCP Client at 9600 to match dataloggers defaults

Works fine (so far). The cloud is reporting as normal and the integration is polling data. The inverter is not supporting anything higher than 38400.

wills106, 2023-05-20:
Can we please try and keep Solis issues and SolaX issues separate. They might appear to be linked but it's hard to track when the discussion is mixed and between so many different issues and discussions pages.

wills106, 2023-05-20:
Thanks, I have set all of the other posts to hidden, so they aren't deleted and can still be viewed. But it's cleaned up this issue.

shaarkys, 2023-05-20:
Thank you for your reaction.

> when you were using both the Pocket WiFi and the Waveshare?
> The Inverter possibly doesn't like two active connections to it.

Well, I was first using only Wifi Dongle and connected it from Homey and from HA (container) via your integration. Unfortunately Dongle is loosing from time to time connectivity to the Inverter and require restart (later on I learned that it's ok to disconnect from wifi connection and the dongle will "self" reboot). So those disconnections were the main reason why I purchased Waveshare reading everywhere it's much better choice. Unfortunately when I have both Homey and HA hooked to the Waveshare, it also seem to loose inverted connectivity randomly. Also Waveshare restart solves that but didn't find the way to automate it. Ironically now with Wifi restart (via ssh forcing Wifi disconnect), it's more reliable compared to Waveshare, which is causing those nasty peaks in the statistics (which is 2nd main reason I'm using HA and your plugin).

With Wifi dongle I got disconnections but never those peaks (will switch back not for testing purposes, having Homey and HA connected via Wifi Dongle.

> do you only want the read only data

I'm so far only reading the data, setting only via HA exceptionally, eg. setting charge/discharge times during Winter etc. - not having yet any more advanced automation. Not sure whether your proposal would be working for me.

**I'm just wondering, can't be those fully random peaks somehow filtered out / dropped ? Seems it's really happening just few times a day ? Maybe only one time ?**

Today, again the same happened

[image]

When looking into details, 14:52 there is this "drop" eg for PV Solar power

[image]

Also I saw you hide the comments from https://github.com/wills106/homeassistant-solax-modbus/issues/401#issuecomment-1549607084 and https://github.com/wills106/homeassistant-solax-modbus/issues/401#issuecomment-1547881532, I thought they speak about the same issue?

shaarkys, 2023-05-20:
On Homey, date are fine for today...but the data analysis is incomparable to HA and your fantastic plugin ;-(

[image]

alienatedsec, 2023-06-01:
@shaarkys Do you use earthing or grounding connection on your Waveshare? I will be removing mine soon to check if this will help my spiking issues. Look at the below for more details, and the reasoning for my question.

https://github.com/alienatedsec/solis-ha-modbus-cloud/issues/13#issuecomment-1570830574

shaarkys, 2023-06-01:
@alienatedsec , interesting and yes, I do use Pgnd, will subscribe your thread to see your results. ;-)

[image]

alienatedsec, 2023-06-01:
@shaarkys is yours PoE too?

shaarkys, 2023-06-01:
> PoE too

non PoE variant... so it could be different then

alienatedsec, 2023-06-01:
@shaarkys no, that could be still related as PoE and non-PoE could act the same if earthing is connected.

@wills106 I know yours is direct ethernet, but do you know anyone with the Waveshare?

wills106, 2023-06-01:
@alienatedsec If yours are POE have you double checked the POE board is seated correctly? #216 
They have been known to be supplied loose. So might only just be hanging in there?

@infradom is using one with his SolaX X3 Gen4 I think he was the first to use one, that's where the info in the Wiki came from.

alienatedsec, 2023-06-01:
> @alienatedsec If yours are POE have you double checked the POE board is seated correctly?

I had no problems with supplying power, and it never lost a ping once. I can double-check, but also would like to follow our theory on earthing those devices. @infradom do you use the below connection?

[image]

wills106, 2023-06-01:
I believe he's only using 485B & 485A as that's what he wrote in the Wiki, but he can confirm.

infradom, 2023-06-02:
Sorry for the late reply, I am indeed only using the RS485A and RS485B wire. Protective ground/Earth is not connected in my setup. I personally did not like to connect is as this might create grounding loops (unlikely however). My setup is non-PoE, so in this case a grounding loop is more likely than with PoE.

wills106, 2023-06-19:
> ;-(
> 
> [image]

Can you please let me know how we are meant to troubleshoot this? You have provided no context other than a screenshot.

I am assuming you still have [two active Modbus connections](https://github.com/wills106/homeassistant-solax-modbus#known-issues) to your Inverter?

Have you investigated using [ha-addon-modbusspy](https://github.com/infradom/ha-addon-modbusspy) to reduce your connections to your Inverter?

The PocketWiFi has never been [officially supported](https://github.com/wills106/homeassistant-solax-modbus#support-modbus-over-rs485--tcp) by this Integration and I still don't recommend it.

Do these issues go away if you remove the PocketWiFi?

If the issue revolves around the PocketWiFi there isn't much I can do about it.

shaarkys, 2023-06-19:
Thank you for your reply.

> how we are meant to troubleshoot this

I posted update with two intentions : 
**1st** - hopefully someone reading along and will say = YES, same issue ! ;-) So we will have more data... 
**2nd,** isn't there a way to filter such "peaks" / non-senses (https://www.statology.org/excel-average-excluding-outliers/)  ? Eg. setting some peaks, above which data will be ignored (probably by entity , or key entities) so simply option to drop updates which significantly differs from previous one... I know size of inverter differs but still there are some theoretical limits etc. (just thinking out loud)
Eg. on example today : 
[image]
or
[image]

> I am assuming you still have [two active Modbus connections](https://github.com/wills106/homeassistant-solax-modbus#known-issues) to your Inverter?

By two connection, do you mean 2 clients connected eg. to the Waveshare or by 2 connections you mean two devices connected to the inverter - eg in my case, PocketWifi AND WaveShare.
In any way, would you rather recommend eg. 2 clients connected to 1 Waveshare ? The problem is I would still like to have remote/cloud monitoring and yes, I have been thinking about possible conflicts, question is then why manufacturer allows that. 

> The PocketWiFi has never been [officially supported](https://github.com/wills106/homeassistant-solax-modbus#support-modbus-over-rs485--tcp) by this Integration and I still don't recommend it.

I have switched to Waveshare now and Homey to PocketWifi, to see what will be the difference. 

> Do these issues go away if you remove the PocketWiFi?

I will have to try as I said however, cloud monitoring, remote upgrades, consultancies with manufacturer etc. is always a plus but yes, I got it and understand it.

A pity there isn't some script to cleanup HA database by significant figures.... in easy way, not finding time manually and editing each entry ;-(

shaarkys, 2023-06-20:
Yesterday I switched sources - HA got data from WavesShare and Homey from Wifi dongle to test / to compare the results. Unfortunately even I have restarted HA (using Docker), it got just a little bit later full nonsense from the WaveShare - like 1-2 hours later - strange is that I nave not faced this behavior in Homey itself when reading data from WaveShare, it simply worked.

Now I'm testing again both (Homey and HA) reading data from Wifi Dongle to see, what will happen.

Any idea @wills106 what could be causing such misbehave, while Waveshare works almost fine with Homey, it does not with HA ? Using 5 sec poll, ModBus TCP on 502.

Does this provide some additional insights ? 

So summary and possible next steps : 
1] Homey reading data from Waveshare and HA from WifiPocket 3.0. - got random peaks
2] Homey reading data from Wifipocket and HA from Waveshare - got random disconnects on Homey (Wifi reconnect works to revive the WifiPocket) and got full nonsenses in HA from WaveShare (as docummented on screenshots)
3] WIll try to shutdown fully Waveshare and use Wifipocket as only source
4] Will try to disconnect GND from Waveshare as someone pointed in the thread
5] Will disconnect Wifipocket and read data from WaveShare only

Btw I have tried also longer polls, like 10 sec, it doesn't seems to provide better results.

I will probably disable WaveShare as next step and will try reading data only from Wifi Dongle as the data it got when I used WaveShare on HA, it scares me little bit... 

[image]
[image]
[image]

Log prior to integration reload and after (marked it in the log)... did not changed anything else for the logging purposes, just another reload of HA integration.

```
2023-06-19 22:25:50.206 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] Solax X3 G4: reading serial number from address 0x0 failed; other address may succeed
2023-06-19 22:25:55.369 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] Solax X3 G4: reading serial number from address 0x300 failed; other address may succeed
2023-06-19 22:25:55.370 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] Solax X3 G4: cannot find serial number, even not for MIC
2023-06-19 22:25:55.370 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2023-06-19 22:25:55.646 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform sun is taking over 10 seconds.
2023-06-19 22:25:55.791 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Solar West' for forecast_solar integration not ready yet: Cannot connect to host 148.251.178.234:443 ssl:False [None]; Retrying in background
2023-06-19 22:25:55.792 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Nová Véska' for forecast_solar integration not ready yet: Cannot connect to host 148.251.178.234:443 ssl:False [None]; Retrying in background
2023-06-19 22:25:55.793 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Solar East' for forecast_solar integration not ready yet: Cannot connect to host 148.251.178.234:443 ssl:False [None]; Retrying in background
2023-06-19 22:25:56.908 WARNING (MainThread) [homeassistant.config_entries] Config entry 'Electricity Spot Rate in CZK/kWh' for cz_energy_spot_prices integration not ready yet: Error communicating with API: ; Retrying in background
2023-06-19 22:26:52.415 WARNING (MainThread) [custom_components.solax_modbus] no local data file found after 5 tries - is this a first time run ?
2023-06-19 23:01:23.335 WARNING (MainThread) [custom_components.solax_modbus] Solax X3 G4: read failed at 0x84: bootloader_version 
2023-06-19 23:01:23.336 WARNING (MainThread) [custom_components.solax_modbus] Solax X3 G4: read failed at 0x85: rtc 
2023-06-19 23:01:23.337 WARNING (MainThread) [custom_components.solax_modbus] Solax X3 G4: read failed at 0x8b: charger_use_mode 
2023-06-19 23:01:23.337 WARNING (MainThread) [custom_components.solax_modbus] Solax X3 G4: read failed at 0x8c: manual_mode 
2023-06-19 23:01:23.338 WARNING (MainThread) [custom_components.solax_modbus] Solax X3 G4: read failed at 0x8d: battery_type 
2023-06-19 23:01:23.338 WARNING (MainThread) [custom_components.solax_modbus] Solax X3 G4: read failed at 0x8e: battery_charge_float_voltage 
2023-06-19 23:01:23.339 WARNING (MainThread) [custom_components.solax_modbus] Solax X3 G4: read failed at 0x8f: battery_discharge_cut_off_voltage 
2023-06-19 23:01:23.339 WARNING (MainThread) [custom_components.solax_modbus] Solax X3 G4: read failed at 0x90: battery_charge_max_current 
2023-06-19 23:01:23.340 WARNING (MainThread) [custom_components.solax_modbus] Solax X3 G4: read failed at 0x91: battery_discharge_max_current 
2023-06-19 23:01:23.340 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 428, in read_modbus_data
    res = self.read_modbus_registers_all()
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 525, in read_modbus_registers_all
    res = res and self.read_modbus_block(block, 'holding')
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 502, in read_modbus_block
    val = decoder.decode_16bit_uint()
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
2023-06-20 08:25:03.415 DEBUG (MainThread) [custom_components.solax_modbus] skipping bytes 4
2023-06-20 08:25:03.415 DEBUG (MainThread) [custom_components.solax_modbus] treating register 0x114 : charge_discharge_power
2023-06-20 08:25:09.522 ERROR (MainThread) [frontend.js.latest.202306080] :0:0 ResizeObserver loop completed with undelivered notifications.
2023-06-20 08:25:13.409 WARNING (MainThread) [custom_components.solax_modbus] no local data file found after 5 tries - is this a first time run ?

```

shaarkys, 2023-06-23:
FYI, now **both** systems - HA and Homey are **using the same source - Wifi dongle**... Waveshare is still connected just not being used.

[image]

[image]

Just have to disconnect/reconnect dongle every day but at-least no false readings.

Now I wonder how to clear the peaks from the DB but that's another challenge.

shaarkys, 2023-08-01:
Yes, please, even I have no idea why Waveshare RS485 to RJ45 do not work reliably - now I have both Homey and HA connected via Wifi Dongle v3 and except of dropouts (1x per week), which can be resolved by disconnecting from Access point, it's running great.

A pity, I thought there will be something useful in the logs, so such peaks or better to say nonsense values will be fixed / not accepted, as Waveshare is always recommended for your integration. Maybe disconnecting wifi dongle would be a way but I don't want to loose remote upgrade options and cloud integration (maybe (and I hope not) if the geopolitical situation would enforce it, then I will be testing Waveshare again).

[image]

..no single peak or nonsense values for past 2 months.

Thank you.

closed 2023-08-01 by shaarkys