wills106/homeassistant-solax-modbus #402: [BUG]

craigcurtin-dev, 2023-05-10:
I have Two Solis Non-hybrid inverters - one is a 4G unit and the other semi industrial.

I have worked out the modbus registers for these 

Do you want a list of them so you can add them to this project or is it going too left field ?

I am in Australia and none of the WIFI dongles work with Modbus of any form - i have instead used a variation of the Waveshare connection to get them working

I use Node Red for most of my automations - but sometimes fall back on HA - mostly for visual feedback

Craig

**Describe the bug**
A clear and concise description of what the bug is.

**Mandatory details**
1. Integration Version:
2. HA Version:
3. Inverter Brand:
4. Plugin Used:
5. First 6 digits of Serial:
6. Connection Method:

**Detailed Error Log**
Detailed Error Log, if you press on the individual error further information is displayed.
This is the content we need, the description alone doesn't provide enough context.

If there are no log's showing for SolaX try the following depending on your language settings, you can find the full logs under:

Settings → System → Logs > at bottom of page press “LOAD FULL LOGS”

Now the full logs are loaded. If you scroll down, you will see them.
Once the full logs are shown, you can either use the search function in your browser to search for “solax” related entries or use the search entry field on top of the page.

Search for

solax

and report us the logs. Make sure to replace sensitive information by xxxx (if any)

**Additional context**
Add any other context about the problem here.

wills106, 2023-05-10:
Yes I can add them in. I was going to add in more industrial SolaX Inverters when I got time. So the addition of Solis ones would be great.

craigcurtin-dev, 2023-05-10:
Excellent thanks

Will format them up and send them over

wills106, 2023-08-14:
Just wondering if managed to get these registers formatted?

wills106, 2023-09-24:
@craigcurtin-dev Did you manage to compile the register set for these two Inverters? Or can I close this FR down?

wills106, 2024-01-10:
Closing down as no response.
Feel free to submit a PR either with the change, or register sets detailed.