wills106/homeassistant-solax-modbus #404: Solax error unrecognized inverter type - serial number : MC210TI839XXXX

KestasG, 2023-05-15:
Setup integration on home assistant and getting errors:
```
This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:4619`
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 5:47:18 PM (2 occurrences)
Last logged: 5:50:31 PM

SolaX: reading serial number from address 0x0 failed; other address may succeed

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:4684
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 5:47:19 PM (2 occurrences)
Last logged: 5:50:31 PM

unrecognized inverter type - serial number : MC210TI839XXXX
``` 
(last for digits replaced with X)

Inverter Solax X3-MIC-G2, connected by Waveshare RS232/485/422 TO POE ETH (B)

KestasG, 2023-05-15:
After adding serial number to `determineInverterType()` it started working, all values got populated except `Measured Power L1,L2,L3`.

wills106, 2023-05-15:
Did you include the X3 at the end?
Should look like:

```
elif seriesnumber.startswith('MC210T'):  invertertype = MIC | GEN2 | X3
```

KestasG, 2023-05-15:
Yes, I did. It gets other metrics for all phases, but not power:
[image]

lcintrat, 2023-07-20:
Hi,

The inverter is recognized by the integration, but I have the same issue regarding the `Measured Power L1,L2,L3` sensors, they report 0.
All other sensors are OK as shown by @KestasG 

HA version: `2023.7.1`
Integration version: `2023.06.8`
Inverter: `Solax X3-MIC G2`
Inverter S/N: `MC210T`
Interface: [RS485 / USB adapter - FT232RL](https://www.amazon.fr/dp/B073M68NGF)

wills106, 2023-08-09:
I have just released [2023.08.4b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.08.4b1) I have moved "Measured Power L1,L2,L3" to a different set of registers which could provide the values? I have also added in the extra registers for the Modbus Power Meter.

wills106, 2023-08-11:
@KestasG @lcintrat Have either of you managed to test out the Beta?
Just wondering if I can release it as a stable release or not?

KestasG, 2023-08-12:
Installed the beta, but the sun is already down. Will check the readings in the morning.

KestasG, 2023-08-13:
Unfortunately, L1, L2, L3 power is still zero.
[image]

wills106, 2023-08-14:
Might be a daft question but do you actually have CT Clamps or a Modbus Power Meter connected to your Inverter?
Also did you restart HA after updating through HACS?

The reason being is I have feed back from another user who has working L2 and L3, but having issues with L1 at the moment.

lcintrat, 2023-08-17:
I have the same results, beta is correctly loaded
```
2023-08-17 08:03:01.758 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0x704 
2023-08-17 08:03:01.758 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x704 measured_power_l1 to block with start 0x704
2023-08-17 08:03:01.758 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x706 measured_power_l2 to block with start 0x704
2023-08-17 08:03:01.758 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x708 measured_power_l3 to block with start 0x704
```
But measured values are still 0.

No power meter nor CT clamps connected to the inverter.

wills106, 2023-08-17:
If you don't have CT Clamps or a Modbus Meter connected these sensors won't return anything.

lcintrat, 2023-08-17:
Oh OK
Thank you for your help!

wills106, 2023-08-18:
@KestasG do you have CT Clamp's or a Modbus Power Meter attached to your Inverter?
If you don't I'll close this issue down as these Measured Power sensors rely on them.

KestasG, 2023-08-18:
I dont have power meter connected to inverter. I'm just wodering how solax show separate phase power on their web site.

cewbdex, 2023-12-05:
That's probably just power it sends to you. This can be closed then?

closed 2023-12-05 by KestasG