wills106/homeassistant-solax-modbus #405: Feature: Solis Inverter Time of Use Switch.

wawyed, 2023-05-15:
Hi, would it be possible to add the Time of Use Switch to enable/disable discharge/charge from the grid?

Seems like a useful feature to use in automations when you have a fixed schedule setup

wills106, 2023-05-16:
Is that different to "Energy Storage Control Switch"

wawyed, 2023-05-16:
How does that switch work? This would just be ON/OFF

alienatedsec, 2023-05-16:
@wawyed @wills106 Operating the SolaX integration will differ from operating the inverter screen. Changing one thing in the integration is equivalent to pressing many buttons and setting several other "functions" to get the same thing.

@wawyed if you want to charge or discharge from the grid, you need to set the "Energy Storage Control Switch" to "Timed Charged Discharge", to disable it, you have change it to "Auto Mode". If your times are predefined on the inverter, you dont need anything else. If your times will change according to `Octopus Agile` or `Octopus Flex` remember to set those before pressing `Update` button.

wills106, 2023-05-16:
I really need to create a "Solis Modes of Operation" page for the Wiki

wawyed, 2023-05-16:
Thanks for the explanation in that case I think I'll close this one as there's already a way to achieve what I wanted.

closed 2023-05-16 by wawyed

Cabocab, 2023-05-18:
Grid charging is enabled by bit 5 of the energy storage control switch. By default, the integration sets the grid charging to on. I was able to set grid charging to off by subtracting 32 from the default dictionary (i.e. for timed use with grid charging turned off, write 3 instead of 35 to the register).