wills106/homeassistant-solax-modbus #411: Polling frequency of the modbus fixed at 30sec

Fernando-Vaz, 2023-05-18:
Anyone confirm that?
I tried other value, less than 30sec, but polling frequency is 30sec.
Thanks

Inverter: Solax X1-Hybrid G3

KestasG, 2023-05-18:
I see values updated every 15 sec
Inverter Solax X3-MIC-G2
Waveshare RS485->TCP adapter
[image]

Fernando-Vaz, 2023-05-18:
I don’t understand...
Inverter X1-Hybrid G3
Waveshare RS485->TCP adapter
Configuration every 5sec poll and only have values every 30sec.
Solax integration version 2023.05.01
[image]
[image]

wills106, 2023-05-18:
Is there a reason you are using a Waveshare and not the built in Ethernet port? As the Gen3 supports Modbus over TCP.
What exactly do you have the Waveshare connected to on your Gen3 and what baud settings are you using?

wills106, 2023-05-19:
Has it always been stuck at 30s or is it a recent thing?
Are you trying to read from the Inverter over Modbus with anything else? Like the built in HA Modbus or something else?

Fernando-Vaz, 2023-05-19:
> Has it always been stuck at 30s or is it a recent thing? Are you trying to read from the Inverter over Modbus with anything else? Like the built in HA Modbus or something else?

It's a recent thing..
I don't try with anything else, but i can try with nodered.
Thanks

wills106, 2023-05-19:
I was more concerned you weren't trying to have two connections to the Inverter as I know the Inverters aren't happy with two active connections.
Has anything else changed?

Fernando-Vaz, 2023-05-19:
To test i will disable SolaX integration.

Fernando-Vaz, 2023-05-20:
Testing with Node-RED
* Start Address = 0
* Quantity = 20
* Every 5 sec
Result OK
Diferent values at each 5 sec.
The problem is related to the Solax integration.
Does anyone have a solution for this situation?
[image]

wills106, 2023-05-20:
Have you only recently enabled EPS mode on the Integration? Or have you always had it enabled.

Fernando-Vaz, 2023-05-20:
The configuration is allways the same. Only updat integration.
I don't know the integration version this happened.

wills106, 2023-05-20:
Are you getting the same error as #412 ? In your log files?

Fernando-Vaz, 2023-05-20:
> Are you getting the same error as #412 ? In your log files?

Yes, i have something like that.

> Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 579, in state
    numerical_value = int(value)
ValueError: invalid literal for int() with base 10: '50Hz'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 274, in async_refresh_modbus_data
    update_callback()
  File "/config/custom_components/solax_modbus/sensor.py", line 185, in _modbus_data_updated
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 585, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 649, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 591, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 583, in state
    raise ValueError(
ValueError: Sensor sensor.solax_eps_set_frequency has device class None, state class measurement unit Hz and suggested precision None thus indicating it has a numeric value; however, it has the non-numeric value: 50Hz (<class 'str'>)
2023-05-20 14:08:56.380 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved

wills106, 2023-05-20:
Try that Beta then I linked to. HA might be getting overwhelmed from the amount of errors.

Fernando-Vaz, 2023-05-20:
Well done my friend.
Now it's working with version [2023.05.2b4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.05.2b4)
Thank you
[image]

closed 2023-05-20 by wills106