wills106/homeassistant-solax-modbus #412: Hz causing exception

RenfoJu1, 2023-05-19:
Logger: homeassistant
Source: components/sensor/__init__.py:583 
First occurred: May 15, 2023 at 10:45:40 AM (35279 occurrences) 
Last logged: 12:45:39 PM

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 579, in state
    numerical_value = int(value)
ValueError: invalid literal for int() with base 10: '50Hz'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 274, in async_refresh_modbus_data
    update_callback()
  File "/config/custom_components/solax_modbus/sensor.py", line 185, in _modbus_data_updated
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 585, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 649, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 591, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 583, in state
    raise ValueError(
ValueError: Sensor sensor.solax_eps_set_frequency has device class None, state class measurement unit Hz and suggested precision None thus indicating it has a numeric value; however, it has the non-numeric value: 50Hz (<class 'str'>)

wills106, 2023-05-20:
Just released [2023.05.2b4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.05.2b4)

Fernando-Vaz, 2023-05-20:
Where i can see this log?

RenfoJu1, 2023-05-21:
> Lähetetty: Saturday, May 20, 2023 4:08:03 PM
> Vastaanottaja: wills106/homeassistant-solax-modbus ***@***.***>
> Kopio: Juha Renfors ***@***.***>; Author ***@***.***>

Thanks, it works!
The log can be found under Settings->System->Logs

Internal

wills106, 2023-05-29:
Closing down as [2023.05.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.05.2) addresses this issue.

closed 2023-05-29 by wills106