wills106/homeassistant-solax-modbus #420: unrecognized inverter type - serial number : unknown (but I can find the

johaaaanz, 2023-05-28:
After update to Home Assistant OS 10.1 https://github.com/home-assistant/operating-system/releases/tag/10.1
I got the error 
unrecognized inverter type - serial number : unknown
and all sensors are unavailable 

**Mandatory details**
1. Integration Version: 2023.05.1 
2. HA Version: 2023.5.4
3. Inverter Brand: Solax
4. Plugin Used: Dont understand/find  (sorry) Solax?
5. First 6 digits of Serial: H34A08
6. Connection Method: Serial

**Detailed Error Log**

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:4684
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 13:07:55 (1 occurrences)
Last logged: 13:07:55

unrecognized inverter type - serial number : unknown
------

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:4628
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 13:07:55 (1 occurrences)
Last logged: 13:07:55

SolaX: cannot find serial number, even not for MIC

--------

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:4621
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 13:07:55 (1 occurrences)
Last logged: 13:07:55

SolaX: reading serial number from address 0x300 failed; other address may succeed  

------

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:4619
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 13:07:49 (1 occurrences)
Last logged: 13:07:49

SolaX: reading serial number from address 0x0 failed; other address may succeed
------

**Additional context**
Add any other context about the problem here.

My inverter looks to exist:
`
elif seriesnumber.startswith('H34'):  invertertype = HYBRID | GEN4 | X3
`
so not sure why "unrecognized inverter type - serial number : unknown" fails.
and : SolaX: cannot find serial number, even not for MIC

Guess can be related to that the inverter is not found.

Running:
alpine-release 3.17.3
Linux core-ssh 6.1.21-v8 #1 SMP PREEMPT Tue Apr 25 14:21:31 UTC 2023 aarch64 Linux

I did try to update this two times and same every time.

Please let men know if I missed to add som info or what to test to fix it. Maybe hardcode:

wills106, 2023-05-28:
Are you using a Baud of 19200? Is your Inverter configured for the standard address of 1?
I would also double check you don't have the A/B wires the wrong way round.

johaaaanz, 2023-05-28:
Hi.
Yes the Baud is unchanged and still on 19200, same for Address its 1. 

Also the wires is unchanged.
All stoped working after i guess home assistant os (or core) upgrade, i tested it a few weeks ago but then i think i managed to restore to last full backup and it was sorking as long as i did not upgrade, which i did again today when it failed again.

Forgot to mention that i run om RPI 4  using ssd if that maybe matter in some way.

johaaaanz, 2023-05-28:
Just found alot of these also:

[core-ssh config]$ ha core logs
2023-05-28 22:31:35.210 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 324: return self._client.read_input_registers(address, count, **kwargs)

wills106, 2023-05-28:
> Detected blocking call

That's normal with RS485 - USB adaptors. https://github.com/wills106/homeassistant-solax-modbus#known-issues
Not ideal I know.

johaaaanz, 2023-05-28:
`Edit: Just reread this, did you have this Integration working at one point?`
Yes it have worked fine for several months before. And running automations that worked great.

Maybe its just my setup or a RPI problem caused by one of the home assistant upgrades i made as it looks like im the only one.

I did get the same result/failure once before, in the beginning of May but then I managed to restore to the old version and have not updated until now.
In hope that either a bug was found in a HA part or some one got the same in the modbus integration
 (as im terrible at python i just waited and hoped for this to get found by someone that understand this beter that I )

However now restore did not work when i tested, not sure you can rollback home assistant os upgrade (and thats why I guess the problem is related to the os upgrade more that the ha core upgrade, as all was working just fine before.

But guess its hard to place a bug on home assistant os if im the only one having a problem.

wills106, 2023-05-29:
Have you tried to power cycle the Inverter?
It doesn't happen very often but I know a couple of people have had to do that as they lost comms with the Inverter before.

johaaaanz, 2023-05-29:
`Have you tried to power cycle the Inverter?`
Thanks for the suggestion. I did try it now but still fails unfortunately.

johaaaanz, 2023-06-01:
Did a reinstall and restore of backup.. 
No clue what broke but now it works again, and im glad!

closed 2023-06-01 by johaaaanz