wills106/homeassistant-solax-modbus #421: Sofar Antireflux control

juraurban, 2023-05-28:
Hi Wills, would you be so kind and add following to allow me control export of energy to the grid.

Sofar HYD 10KTL-3PH 

[image]

Cheers
Jiri

sander85, 2023-05-31:
Hi!

I'd be also very happy if this option is added (running HYD 15KTL-3PH). Current energy prices are for some days below 0, so it would be great to control when the power flows back to the grid and when does it not.

wills106, 2023-06-04:
I have just released [2023.06.1b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.06.1b2)

sander85, 2023-06-07:
Seems to have the same problem as my own local changes. I can select different mode and/or power level, but these changes don't reach the inverter.

wills106, 2023-06-07:
It mentions precision of 10.
If you set it to 100 does it give an export of 1000w?

Although it looks like I have a different document to @juraurban what version / where did that document come fro?

juraurban, 2023-06-07:
I have tested it as well, but no change in my inverter when I change reflux control to any value - default is empty.
The same with Reflux power - default NaN > I am not even able to change it.

juraurban, 2023-06-07:
> It mentions precision of 10. If you set it to 100 does it give an export of 1000w?
> 
> Although it looks like I have a different document to @juraurban what version / where did that document come from?

I have found this document on Facebook Sofar page as latest version by date there.

Matz88, 2023-06-08:
In my case (Sofar rebranded 6KW single phase) the limitation works only with reflux power set 0, any other value is simply leading to no limitation

wills106, 2023-06-08:
> This seems to be copy-paste error:

I have corrected that in [2023.06.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.06.2) I haven't altered the scaling yet till someone can confirm it's actually reading.

> I can use service: modbus.write_register if I change the first register (1023, but send two values at the same time, for 1024 as well).

I might need to move this to a button to send the value of "Reflux Power" along with the mode of "Reflux Control"

sander85, 2023-06-08:
> I have corrected that in [2023.06.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.06.2) I haven't altered the scaling yet till someone can confirm it's actually reading.

For me it's now reading without hacking the code myself, but the reading is still 150 W, not 15000 W like configured in the inverter.

> > I can use service: modbus.write_register if I change the first register (1023, but send two values at the same time, for 1024 as well).
> 
> I might need to move this to a button to send the value of "Reflux Power" along with the mode of "Reflux Control"

Yeah, this seems like a working option.

juraurban, 2023-06-08:
When I set 1000.0 kw on Sofar Inverter ... the value Reflux power in HA displays 10000
When I set 0002.0 kw on Sofar Inverter (which actually limits to 2kw correctly) ... the value Reflux power in HA displays 0
When I set 0010.0 kw on Sofar Inverter (which actually limits to 10kw correctly) ... the value iReflux power n HA displays 100

When I change the value in HA ... does not write to Inverter.

No way for me to change Reflux control ---- in invertor have 3 options - Anti reflux, disabled, Threephase limit

wills106, 2023-06-08:
I only added the scaling to the number entity and not the corresponding sensor. I have just released [2023.06.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.06.3) with the extra scaling.

Under "Reflux Control" there should be 3 modes I have added in, I don't know if they match the wording on the Inverter. I was trying to interpret the Modbus doc.

Matz88, 2023-06-08:
I can confirm that the reflux power setting set in HA is then reflected in solarman correctly and applied to the inverter properly

About reflux control modes:
 -0 enabled -->no effect on solarman app or the inverter, remains set to 1
 -1 disabled -->no effect on solarman app or the inverter
 -2 disabled to a limit (see Reflux power) -->no effect on solarman app or the inverter, remains set to 1

juraurban, 2023-06-08:
to me: Reflux power is read properly by HA now, unfortunately when I change it reverts back in few seconds without any effect to invertor.

wills106, 2023-06-08:
Just spotted another typo ...
Just released [2023.06.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.06.4)

juraurban, 2023-06-08:
Readings work fine now! 
Unfortunately writing not all.

wills106, 2023-06-08:
I'll convert it to a write multi register mode.

Would you really only use option 0 = no grid export and option 2 + power level to export to the grid?

sander85, 2023-06-08:
> Would you really only use option 0 = no grid export and option 2 + power level to export to the grid?

It's more like option 0 with whatever power level to disable limiting at all. And option 2 with power level 0 to enable limiting and limit it to 0 W = no export.

At least this is how my automations are running right now. Price below 0 -> mode 2, power level 0.
Price above 0 -> mode 0, power level restored from input_number state (usually 15000 W).

Matz88, 2023-06-08:
ah damm i wrote them inverter in my message, corrected now

juraurban, 2023-06-09:
For me it's 
enabled + reflux power set to 0 kW (in case of electricity price < 0)
disabled (all other cases)

I won't use another option 

But still not being able to write anything to inverter (reading works fine).

jordisco, 2023-06-09:
Hello everyone and thank you for the work you are doing. I also have the same problem with my inverter. When I enter the new parameters, they are not saved. My inverter is a HYD 6000 EP, and I am very interested in being able to modify this data.

wills106, 2023-06-09:
I need to release an update to change how the Integration writes to the Sofar Inverter for "Reflux Power".
Hopefully I'll get this done over the weekend.

juraurban, 2023-06-18:
I have tested latest version, unfortunately still not being able to write.... neither set Reflux enabled/disabled, nor set value of Reflux Power.

jordisco, 2023-07-20:
> Necesito publicar una actualización para cambiar la forma en que la integración escribe en el inversor Sofar para "Reflux Power". Espero terminar esto durante el fin de semana.

Hello and thank you for your work. Do you have any idea when you will be able to solve this problem? Here in Spain, we are noticing that in the recent invoices we are being charged for excess electricity during certain hours, and I would like to be able to automate the surplus based on the selling price of electricity.

wills106, 2023-07-25:
I have just released [2023.07.4b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.4b1) does this allow you to set Reflux Control/Power?

jordisco, 2023-07-27:
In my case, even if I change the values of Relux Power and enable Reflux Control, it does not show changes in the state of the inverter.

wills106, 2023-07-28:
> it does not show changes in the state of the inverter.

Is that with the latest Beta and are you pressing the button after setting the select and number entitie?

jordisco, 2023-07-28:
> > it does not show changes in the state of the inverter.
> 
> Is that with the latest Beta and are you pressing the button after setting the select and number entitie?

Sorry, Yes it works, pressing the reflux control button after selecting the enable option

wills106, 2023-07-28:
@juraurban @sander85 Does the recent change allow you to set Antireflux ok?
If so I can add it into the stable release.

juraurban, 2023-07-28:
I will test as soon as I am back from vacation (7.8.), dont wanna break something :)

juraurban, 2023-08-06:
Hi, I am back. I have downloaded latest beta 2023.08.3 but when I press button Reflux control then error appears:
[image]

I can't change Reflux power either: 
[image]

juraurban, 2023-08-06:
Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/solax_modbus/__init__.py:410 
Integration: Home Assistant WebSocket API (documentation, issues) 
First occurred: 11:57:17 AM (13 occurrences) 
Last logged: 12:03:11 PM

[140340310757584] int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
[140340305840720] int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
[140339896731792] int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
[140340079300624] int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 205, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1957, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 1997, in _execute_service
    return await cast(
           ^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 845, in entity_service_call
    response_data = task.result()  # pop exception if have
                    ^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1192, in async_request_call
    return await coro
           ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 889, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/button/__init__.py", line 124, in _async_press_action
    await self.async_press()
  File "/config/custom_components/solax_modbus/button.py", line 86, in async_press
    if res: self._hub.write_registers_multi(unit=self._modbus_addr, address=self._register, payload=res)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 410, in write_registers_multi
    value = int(value)
            ^^^^^^^^^^
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'

juraurban, 2023-08-06:
This error originated from a custom integration.

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:409 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 11:57:17 AM (13 occurrences) 
Last logged: 12:03:11 PM

cannot treat payload scale None SofarModbusNumberEntityDescription(key='reflux_power', device_class=None, entity_category=None, entity_registry_enabled_default=True, entity_registry_visible_default=True, force_update=False, icon='mdi:battery-sync', has_entity_name=False, name='Reflux Power', translation_key=None, unit_of_measurement=None, max_value=None, min_value=None, mode=None, native_max_value=20000, native_min_value=0, native_step=100, native_unit_of_measurement=<UnitOfPower.WATT: 'W'>, step=None, allowedtypes=4096, register=None, read_scale_exceptions=None, read_scale=1, fmt='i', scale=100, state=None, max_exceptions=None, min_exceptions_minus=None, blacklist=None, write_method=3, initvalue=None, unit='_uint16', prevent_update=True)

GithRS, 2023-08-07:
Same here, I get error:
Failed to call service button/press. H format requires 0 <== number <== 65535
Log:
Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/solax_modbus/__init__.py:412
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 11:19:36 (15 occurrences)
Last logged: 23:39:00

[281473060336576] 'H' format requires 0 <= number <= 65535
[281472893216448] 'H' format requires 0 <= number <= 65535
[281472925164992] 'H' format requires 0 <= number <= 65535
[281472849769536] 'H' format requires 0 <= number <= 65535
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 226, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1974, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2011, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 870, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 942, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/button/__init__.py", line 124, in _async_press_action
    await self.async_press()
  File "/config/custom_components/solax_modbus/button.py", line 86, in async_press
    if res: self._hub.write_registers_multi(unit=self._modbus_addr, address=self._register, payload=res)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 412, in write_registers_multi
    if   typ == REGISTER_U16: builder.add_16bit_uint(value)
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 162, in add_16bit_uint
    self._payload.append(pack(fstring, value))
                         ^^^^^^^^^^^^^^^^^^^^
struct.error: 'H' format requires 0 <= number <= 65535

hruskaexe, 2023-09-13:
Hi, I have the same problem, what is the status? Thanks

wills106, 2024-02-12:
Is this still an issue? Or did it get resolved?

juraurban, 2024-02-13:
Hi, still the same issue here.
[image]

cschlipf, 2024-02-19:
Hi all,

 I am currently working on the Sofar integration and will see what I can do here in the next days...

PR #809 (2024-03-17): Sofar: Fix #421: Reflux Control
Fixes #421 

- Reflux control was not working on Sofar. This PR fixes this.

- Entities have been updated to match the inverter terminology: Reflux renamed to FeedIn:
[image]

- Modes had been wrong. Updated to correct modes: 
[image]

- Updated documentation
commit 3e1377b9 (2024-03-17): Sofar: Fix #421: Reflux Control
commit b01f4857 (2024-03-17): Merge pull request #809 from cschlipf/main

Sofar: Fix #421: Reflux Control
closed 2024-03-17 by wills106