wills106/homeassistant-solax-modbus #422: [BUG] Missing unit_of_measurement for EPS Mode Runtime

luboshl, 2023-05-29:
"EPS Mode Runtime" is displayed in Home Assistant entity as simple state value. It should be displayed the same way as "Grid Mode Runtime" that has unit_of_measurement set to "h" (hours).

[image]

1. Integration Version: 2023.05.1
2. HA Version: 2023.5.4
3. Inverter Brand: SolaX
4. Plugin Used: SolaX Inverter Modbus
5. First 6 digits of Serial: H34A10
6. Connection Method: RS485-ethernet adapter

luboshl, 2023-05-29:
I've also created PR #423

luboshl, 2023-05-29:
And one point to the discussion - both sensors should be renamed:
- EPS Mode Runtime -> EPS Mode Run Time
- Grid Mode Runtime -> Grid Mode Run Time

I can rename them of course.

wills106, 2023-06-01:
Feel free to submit any changes / Improvements to EPS mode.
I don't use it, so I haven't been able to validate any of it. I just went off the docs.

wills106, 2023-07-31:
I'm going to close this issue down. Feel free to submit any further PR's if you can find any further EPS improvements.

closed 2023-07-31 by wills106