wills106/homeassistant-solax-modbus #425: Connection parameters for Growatt MIN 4600TL-X Modbus TCP

BramNH, 2023-06-02:
I have a Growatt MIN 4600TL-X with a Growatt ShineLan-X dongle
The datasheet of the dongle says inverter communication is: USB (Modbus RTU)
and Server communication is: TCP(Modbus TCP)

I see that this project supports (but untested) Growatt TL-X models, _**does it also work over local Modbus TCP?**_
I have tried many configurations, HomeAssistant log says over port 502 (Modbus TCP) the connection is refused, and over port 80 (webserver) the connection is unexpectedly closed.

and another log message:

```
Logger: custom_components.solax_modbus.plugin_growatt
Source: custom_components/solax_modbus/__init__.py:330
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 10:50:18 (20 occurrences)
Last logged: 12:29:36

growatt: attempt to read firmware failed at 0x9
growatt: attempt to read firmware failed at 0xbb9
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_growatt.py", line 56, in _read_serialnr
    inverter_data = hub.read_holding_registers(unit=hub._modbus_addr, address=address, count=6)
  File "/config/custom_components/solax_modbus/__init__.py", line 330, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/mixin.py", line 104, in read_holding_registers
    return self.execute(
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/base.py", line 194, in execute
    raise ConnectionException(f"Failed to connect[{str(self)}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.174:502)]
```

wills106, 2023-06-02:
The Integration does support Modbus over TCP as that's what I personally use on my SolaX Inverter.
There is a possibility they are using a non standard port.

Also I can't seem to see any reference to anyone actually using that datastick for Modbus.

BramNH, 2023-06-03:
The Growatt [TL2500~6000 TL-X](https://solarclarity.compano.com/Data/Environments/000001/Attachment/Bijlage/Manufacturers/Growatt/2300%20-%20Growatt/2310%20-%20Growatt%20Inverter/2313%20-%20Growatt%20Inverter%20MIN%202500-6000TL-X/2%29%20Datasheets/EN%20MIN%202500-6000TL-X%20Datasheet.pdf?_gl=1*1xh6a2u*_ga*MTkxMjcxODI5NS4xNjg1NjA3NTkx*_ga_EF3CWREKBJ*MTY4NTc5OTM5NC4zLjAuMTY4NTc5OTM5NC4wLjAuMA..) provides an optional interface: either WiFi / LAN / 4G through a dongle that is connected to the USB port on the inverter. I haven't seen any Growatt inverter connected via LAN directly (without any dongle).

The LAN dongle obtains info from the inverter via Modbus RTU (over USB). And sends the data to the growatt servers via Modbus TCP (over LAN). The dongle has a webserver that you can access locally (port 80). I believe this is the only open port it has. See [datasheet of the dongle](https://online-accu.nl/media/pdf/7e/55/54/Growatt-Shine-Lan-X-NL.pdf).

It seems to me it is **not possible** to use Modbus (over port 502) to talk to the LAN module, thus not at all with Modbus TCP since the module is the only way to use LAN with this inverter.

Since you had TL-X in the list of supported (but untested) models, I thought it might be possible.
**_Should my MIN 4600TL-X model be supported?_**

wills106, 2023-06-03:
I think the dongle is for cloud connectivity only.
You can connect to the "Sys COM Port" Pins 3 & 4 with something like a waveshare.

The Growatt Modbus docs only mention Modbus RTU.
The default baud should be 9600, but you can change it on your Inverter.

BramNH, 2023-06-03:
Yes I thought so. I am first gonna try an interesting [project](https://github.com/johanmeijer/grott) that obtains the info from the LAN dongle and forwards it to the Growatt servers. Later I will definitely connect to the SYS COM port directly over Modbus RTU to get live data.

Thanks for the quick responses!

closed 2023-06-03 by BramNH