wills106/homeassistant-solax-modbus #427: Unable to change backflow power setting [BUG]

J14111, 2023-06-03:
**Describe the bug**
A clear and concise description of what the bug is.

**Mandatory details**
1. Integration Version: 2023.06.1b1
2. HA Version: 
3. Inverter Brand: Solis
4. Plugin Used: Solis
5. First 6 digits of Serial:
6. Connection Method: TCP

This was working fine but since a recent update I am unable to change the backflow power setting, if I try to change it I see this in the GUI:

[image]

I have tried removing and reinstalling the intergration

J14111, 2023-06-15:
This is still the same in the latest update, is anyone able to have a look please?

jojoBzh, 2023-12-05:
Hello,
I had the same problem but resolved now:
Just have a look to this : https://github.com/wills106/homeassistant-solax-modbus/issues/637
Regards
#637

wills106, 2024-02-12:
Just checked and "Backflow Power" has been updated to have the scale set at 100.
Can this be closed down now?

J14111, 2024-02-12:
Still having the same issue on the latest beta

[image]

wills106, 2024-02-12:
I miss read and the scale was still at 0.01 and not the 100 mentioned in #637 
I have just released [2024.02.3b4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.3b4) with the change.

J14111, 2024-02-12:
That seems to have fixed it thanks!

closed 2024-02-12 by J14111