wills106/homeassistant-solax-modbus #432: Use of the new entities types

wawyed, 2023-06-08:
I just saw the new version released and might be useful to update to the new entity types for updating the charge discharge times on the Solis inverters at least. Might simplify things :)

https://www.home-assistant.io/blog/2023/06/07/release-20236/#new-entities-date-time-datetime

wawyed, 2023-06-08:
Is it this one the one you are looking at ? https://www.home-assistant.io/integrations/input_datetime

wawyed, 2023-06-08:
You are probably right. The other one date is optional but I think it's for user creation not an integration.

wawyed, 2023-07-02:
Did you manage to get it working in the end? @wills106

wills106, 2023-07-03:
Not yet, at the time there was only 2 Integrations supporting time and I didn't understand how they were getting it to work.
I'll have another look to see if there is more info available now it's been out for a while.

SzosszeNET, 2024-02-11:
I'm sure there are other important pieces, just wanted to bring to your attention that it appears that this project implemented time entries for charge discharge - https://github.com/Pho3niX90/solis_modbus

wawyed, 2024-03-19:
is this something that'll ever get implemented? Just consider if it's worth moving to the other project for Solis or worth staying on this one?

wills106, 2024-03-21:
I have updated the time branch with newer code where at least the time entity loads. Even though it doesn't write anything at the min.
I'm looking at setting up a Modbus simulator in docker for testing purposes. So I'm not restarting my live HA 1,000's of times and potentially writing something I shouldn't be doing to the Inverter.

Can't comment on any timeframes.

wills106, 2024-05-28:
If anyone knows what they are doing better than I do with python I have had another attempt.

https://github.com/wills106/homeassistant-solax-modbus/tree/time

time.py is based on select.py and time.2.py is based on number.py
Neither return the value from "charger_start_time_1" need that working before progressing to writing the time.

wawyed, 2025-01-07:
Hey, with the latest firmware we can have 6 charge/discharge slots. It would be really useful to have datetime entities to simplify how you setup them by not having to change the hours minutes as separate sensors. Did you have any better idea on how to implement this?

No worries if not, just thought I'd ask as it'd be extremely useful as the https://github.com/Pho3niX90/solis_modbus does not support the new firmware slots

Regards

wills106, 2026-04-02:
Added initial support for time entities in [2026.04.2b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2026.04.2b1) Initial support is focused on SolaX AC / Hybrid Gen2 & Gen3 Inverters.