wills106/homeassistant-solax-modbus #434: [BUG]

MrRickwi, 2023-06-08:
Using the actual version of Solax integration, I discovered two problems:

1. Although the update interval is configured to 5 sec, the sensors are only updated every 30 sec. This is a problem for me because energy load balancing routines are dependent on high update rates.

2. The are repeated errors in the HA log:

2023-06-08 22:01:02.843 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 292, in async_refresh_modbus_data
    update_callback()
  File "/config/custom_components/solax_modbus/sensor.py", line 172, in _modbus_data_updated
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 590, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 654, in _async_write_ha_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 596, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 469, in state
    value = self.native_value
            ^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/sensor.py", line 193, in native_value
    return self._hub.data[self.entity_description.key]*self.entity_description.read_scale
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for *: 'datetime.datetime' and 'int'

1. Integration Version: 2023.06.4
2. HA Version: 2023.6.0
4. Inverter Brand: Solax

wills106, 2023-06-08:
I was running this fine myself on 2023.6.b4 I think.
Moved to the release version and like you mention it's broken 🤕

wills106, 2023-06-08:
I have just removed the Integration from the Integration page and added it back in again and it's work at the normal polling rate for me.

I don't know if it's related to the Python change? https://www.home-assistant.io/blog/2023/06/07/release-20236/#breaking-changes

MrRickwi, 2023-06-09:
I did as you suggested - and voila, the errors in the log are away, and the refresh rate is back at 5 sec. Strange behavior of HomeAssistant... Thanks for your work!

darkrain-nl, 2023-06-10:
Need to reopen...
When RTC is enabled we have this error:
```
This error originated from a custom integration.

Logger: homeassistant
Source: custom_components/solax_modbus/sensor.py:193
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 10:26:36 (219 occurrences)
Last logged: 10:42:11

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 292, in async_refresh_modbus_data
    update_callback()
  File "/config/custom_components/solax_modbus/sensor.py", line 172, in _modbus_data_updated
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 590, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 654, in _async_write_ha_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 596, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 469, in state
    value = self.native_value
            ^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/sensor.py", line 193, in native_value
    return self._hub.data[self.entity_description.key]*self.entity_description.read_scale
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for *: 'datetime.datetime' and 'int'
```

wills106, 2023-06-10:
I can recreate this on HA 2023.06.0 & HA 2023.06.1

CristiXP, 2023-06-10:
I am having the issue with the polling on version Home Assistant 2023.6.1 and for Solax Modbus 2023.06.5

wills106, 2023-06-10:
As long as you have RTC turned off / disabled have you tried doing what I posted in the 3rd post?

CristiXP, 2023-06-10:
Not sure how to do that, or what it means 🥲

darkrain-nl, 2023-06-10:
I'm on HA 2023.6.1 and the integration is version 2023.06.5 @wills106

CristiXP, 2023-06-10:
This value seems to be unknown since 4th of June[image]

infradom, 2023-06-10:
I believe I understand where the RTC issue comes from. I am examining it now.
I disabled RTC a long time ago since it had no value for me after I synced the clock with HA
I try to find a solution in the coming hour(s)

darkrain-nl, 2023-06-10:
Yep quickly made the changes local and they work :)

Thanks!

infradom, 2023-06-10:
i

> Yep quickly made the changes local and they work :)
> 
> Thanks!

Was  a quick fix, I am sure we can clean this code more later ...

infradom, 2023-06-10:
For my information: I would like to understand why people enable RTC. It pollutes the state change history every few seconds with useless data. So there must be a reason that I am missing.
A periodic syc_rtc automation action makes sure the clock is in sync

CristiXP, 2023-06-10:
> For my information: I would like to understand why people enable RTC. It pollutes the state change history every few seconds with useless data. So there must be a reason that I am missing.
> 
> A periodic sync_rtc automation action makes sure the clock is in sync

How and where to disable it?

darkrain-nl, 2023-06-10:
> For my information: I would like to understand why people enable RTC. It pollutes the state change history every few seconds with useless data. So there must be a reason that I am missing. A periodic sync_rtc automation action makes sure the clock is in sync

I have it enabled to do the sync once a week, is it not required to be enabled for that? Cause if that is the case I would happily disable it and still sync it once a week...

infradom, 2023-06-10:
> How and where to disable it?

It is disabled by default in recent versions. It can be disabled by clicking the entity name and then selecting the settings icon

infradom, 2023-06-10:
> I have it enabled to do the sync once a week, is it not required to be enabled for that? Cause if that is the case I would happily disable it and still sync it once a week...

Both items are independent from each other, so sync_RTC will work even with RTC disabled

darkrain-nl, 2023-06-10:
> > I have it enabled to do the sync once a week, is it not required to be enabled for that? Cause if that is the case I would happily disable it and still sync it once a week...
> 
> Both items are independent from each other, so sync_RTC will work even with RTC disabled

Thanks, I disabled it again :)

CristiXP, 2023-06-10:
> > How and where to disable it?
> 
> 
> 
> It is disabled by default in recent versions. It can be disabled by clicking the entity name and then selecting the settings icon (gear)
> 
> The sync_rtc button is much more usefull. I click it once every month, but an automation could do this of course

Ah, ok, didn't knew it was reffering to the Solax entity. Done now and it works.

infradom, 2023-06-10:
One last comment and conclusion: I prefer to disable RTC also because I log all my state changes to an influxDB database, and these RTC updates offer no value in that database (they just consume storage)
Anyway, RTC will work again in the next update, but I recommend to leave it disabled and use sync_RTC periodically

wills106, 2023-06-10:
Just released [2023.06.7](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.06.7) which resolves this issue.

closed 2023-06-10 by wills106