wills106/homeassistant-solax-modbus #437: [BUG] maximum Export Control User Limit too low with other inverters active

blb4github, 2023-06-10:
**Describe the bug**
The Maximum Export Control User Limit for my Gen2 X1 SK-SU (U30) inverter is set to 3000. the PV system connected to this inverter is 2250 Wp but I have more system with total Wp of 6022 Wp. This limit of 3000 is limiting my output to Grid.

I have changed  the MAX_EXPORT value in plugin_solax.py for my inverter ('U30') from 3000 to 6000 and this seems to do the trick. 
MAX_EXPORT = [
    ('L30E',    3000 ), # Gen2 X1 SK-TL
    ('U30',     6000 ), # Gen2 X1 SK-SU

**Mandatory details**
1. Integration Version: 2023.06.7 (latest)
2. HA Version: 2023.4.0
3. Inverter Brand: Solax
4. Plugin Used: modbus
5. First 6 digits of Serial: U30EDB 
6. Connection Method: Ethernet

**Additional context**
It would be great if this limit can be upgraded to higher values, taking into account other inverters exporting to GRID as well.
it's not a problem for the interver; I can change this value to 30.000 via the Solax (local connected) app.

infradom, 2023-06-11:
@wills106 Maybe we could add a config_entity like we did for export_control_user_limit_readscale
This would make sense for some non-standard setup's
This entity could be disabled by default. Any opinion on that ?

wills106, 2023-06-11:
So is it treating this value as a global combined export then? Sot it's a sum of both Inverters output. As you are exceeding 3000w it's starting the throttle the output of the SKSU?
As this Inverter should only be able to export 3000w on it's own. That's where this value came from.

blb4github, 2023-06-11:
It's measuring the value from the modbus kWh meter connected to the GRID wires. The inverter is indeed throttling the output when the kWh meter reports > 3000 W. Today this didn't happen as I did change th.e (max export) value to 6000W

wills106, 2023-06-14:
I have just released [2023.06.8](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.06.8)
There is a new number entity "Config Max Export" which you can enable to override the max limit the sliders will go to.

blb4github, 2023-06-14:
Great, thanks!! I have upgraded to 2023.06.08, activated and set the Config Max Export entity to a value which works for my setup (6000 W). perfect!!

blb4github, 2023-06-14:
This issue can be closed.

wills106, 2023-06-15:
I have updated the wiki for:
[Gen2 & Gen3](https://github.com/wills106/homeassistant-solax-modbus/wiki/SolaX-Gen2-&-Gen3-Modes-of-operation#export-control-user-limit)
[Gen4](https://github.com/wills106/homeassistant-solax-modbus/wiki/SolaX-Gen4-Modes-of-Operation#export-control-user-limit)

closed 2023-06-15 by wills106