wills106/homeassistant-solax-modbus #440: Growatt MIN6000TL-HX... no data at all

uspino2, 2023-06-13:
Here are my settings. I think I've double checked every option and I have gone through Wiki step by step:

**Waveshare RS485 to Eth (Firmware 1.452):**
Baud Rate: 9600
Work Mode: TCP Server
Transfer Protocol: Modbus_TCP Protocol
Modbus Gateway Type: Simple Modbus TCP to RTU

**Home Assistant:** 
Interface: TCP / Ethernet
Modbus Address: 1
Inverter Type: Growatt
(EPS, Dry Contact Box and Parallel Mode unchecked)
IP Address: 10.10.10.126
TCP Port: 502
Modbus Variant: Modbus RTU over TCP

**Log:**
```
2023-06-13 10:33:39.839 INFO (SyncWorker_7) [homeassistant.loader] Loaded growatt_server from homeassistant.components.growatt_server
2023-06-13 10:35:09.707 INFO (MainThread) [custom_components.solax_modbus.config_flow] validating base: {'name': 'Growatt_modbus', 'interface': 'tcp', 'read_modbus_addr': 1, 'plugin': 'growatt', 'scan_interval': 15, 'read_eps': False, 'read_dcb': False, 'read_pm': False}
2023-06-13 10:35:09.707 INFO (MainThread) [custom_components.solax_modbus.config_flow] validating base config for Growatt_modbus: pre: {'name': 'Growatt_modbus', 'interface': 'tcp', 'read_modbus_addr': 1, 'plugin': 'growatt', 'scan_interval': 15, 'read_eps': False, 'read_dcb': False, 'read_pm': False}
2023-06-13 10:36:51.028 INFO (MainThread) [custom_components.solax_modbus.config_flow] title configflow solax_modbus name: {'name': 'Growatt_modbus', 'interface': 'tcp', 'read_modbus_addr': 1, 'plugin': 'growatt', 'scan_interval': 15, 'read_eps': False, 'read_dcb': False, 'read_pm': False, 'host': '10.10.10.126', 'port': 502, 'tcp_type': 'rtu'}
2023-06-13 10:36:51.031 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: growatt
2023-06-13 10:36:51.036 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Growatt_modbus: trying to determine inverter type
2023-06-13 10:36:56.041 WARNING (MainThread) [custom_components.solax_modbus.plugin_growatt] Growatt_modbus: reading firmware number from address 0x9 failed; other address may succeed
2023-06-13 10:36:56.045 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Read Growatt_modbus 0x9 firmware number before potential swap: None
2023-06-13 10:36:56.045 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Growatt_modbus: trying alternative location
2023-06-13 10:37:01.052 WARNING (MainThread) [custom_components.solax_modbus.plugin_growatt] Growatt_modbus: reading firmware number from address 0xbb9 failed; other address may succeed
2023-06-13 10:37:01.053 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Read Growatt_modbus 0xbb9 firmware number before potential swap: None
2023-06-13 10:37:01.053 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] Growatt_modbus: cannot find firmware version, even not for other Inverter
2023-06-13 10:37:01.054 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized Growatt_modbus inverter type - serial number : unknown
2023-06-13 10:37:01.067 INFO (MainThread) [custom_components.solax_modbus.sensor] Growatt_modbus returning holding block: 0xf 0x33 [15, 45]
2023-06-13 10:37:01.106 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solax_modbus entity: sensor.growatt_modbus_language
2023-06-13 10:37:01.106 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.solax_modbus entity: sensor.growatt_modbus_rtc
```

The blue light in the Waveshare turns blue when the component is added, and never again. Does this mean there's a connection error between the Waveshare and Home Assistant or between my Growatt Inverter and the Waveshare?

Any help would be greatly appreciated. Thanks!

wills106, 2023-06-13:
When you setup the Integration did you Select Modbus TCP?

[image]

The Waveshare does the RTU - TCP conversion for you.

uspino2, 2023-06-13:
I just tried again, Modbus TCP, but still the same: the Link light turns blue for a few seconds, goes back to green and no entities are created. I can read RS485 data from a ESP32 + 485 module, so I know the inverter is sending the info. Do you see something unusual in these settings? Thank you for taking a look at it. 

[image]

[image]

uspino2, 2023-06-14:
Ok, I think I made some progress, with baud rate at 9600 and Modbus Gateway Type set at "Auto Query Storage Type" (Multi-GHost and Bus Conflict checkboxes both enabled). There seems to be communication activity, but I get this in the log:

```
2023-06-14 08:34:50.456 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: growatt
2023-06-14 08:34:50.461 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Growatt_Modbus: trying to determine inverter type
2023-06-14 08:34:50.508 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Read Growatt_Modbus 0x9 firmware number before potential swap: AL1.0ZABA
2023-06-14 08:34:50.508 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized Growatt_Modbus inverter type - serial number : AL1.0ZABA
2023-06-14 08:34:51.342 INFO (MainThread) [custom_components.solax_modbus.sensor] Growatt_Modbus returning holding block: 0xf 0x33 [15, 45]
```

Any suggestion? Tanks!

uspino2, 2023-06-14:
Just in case this is related to [Discussion #26](https://github.com/wills106/homsassistant-solax-modbus/discussions/26), here are the specs I could find about this particular model:

MIN 6000TL-XH
6kW
Max. Recommended PV Power: 10.000 W
Single Phase
No. Of MPP trackers: 2
Serial: QQK2000022****SS

alienatedsec, 2023-06-14:
> [image]

@uspino2 potentially unrelated and I am curious about it - have you updated the firmware to `v1,486` or did you receive it with this version? If so, when and where did you buy it?

uspino2, 2023-06-14:
> > [image]
> 
> @uspino2 potentially unrelated and I am curious about it - have you updated the firmware to `v1,486` or did you receive it with this version? If so, when and where did you buy it?

Upgraded following [these](https://github.com/alienatedsec/solis-ha-modbus-cloud/discussions/17) instructions.

wills106, 2023-06-14:
@uspino2 I'll add you Inverter in. It's a Hybrid version isn't it? Ie PV and Battery storage?

uspino2, 2023-06-14:
> @uspino2 I'll add you Inverter in. It's a Hybrid version isn't it? Ie PV and Battery storage?

Yes it is. Thanks!

alienatedsec, 2023-06-14:
> Upgraded following [these](https://github.com/alienatedsec/solis-ha-modbus-cloud/discussions/17) instructions.

Thanks, and I am glad it is working for you (at least what I see from the logs). However, I updated the guide, so it reflects on its purpose.

wills106, 2023-06-14:
I have just released [023.06.8](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.06.8) with the Firmware version. Your Inverter should be using the newer register set.

uspino2, 2023-06-14:
Great job! It all came out like magic. Some sensors are greyed out but I guess I need to wait until their values are populated.

One question though, I can't see a sensor for AcPower, the register that provides the actual energy being produced. Is there a way to see that? Thank you so much!

[image]
[image]

uspino2, 2023-06-14:
Great, let me know if I can help. This is the register for AcPower that I could read with a ESP32+485 board:

  - platform: modbus_controller
    name: "${device_name} AcPower"
    address: 40
    register_type: "read"
    unit_of_measurement: W
    device_class: power
    icon: mdi:flash
    value_type: U_DWORD
    accuracy_decimals: 1
    filters:
    - multiply: 0.1

uspino2, 2023-06-15:
For Battery integration in HA Dashboard, I'm also missing these entities: 

Home Battery Storage

```
SolaX Gen2
Energy going to the battery (kWh) - You need to enable "Battery Input Energy Total"
Energy coming out of the battery (kWh) - You need to enable "Battery Output Energy Total"
SolaX Gen3 & Gen4
Energy going to the battery (kWh) - sensor.solax_battery_input_energy_today
Energy coming out of the battery (kWh) - sensor.solax_battery_output_energy_today
```

Other than that, It's been working great since yesterday.

wills106, 2023-07-31:
I have just released [2023.07.4b5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.4b5) which redefines all Growatt Inverters. So a lot has changed.

wills106, 2023-07-31:
I'm going to close this issue down on consolidate to #471

closed 2023-07-31 by wills106

uspino2, 2023-08-01:
Great job, thanks! Everything looks fine, except "PV Voltage 1" now appears as _unavailable_...

wills106, 2023-08-01:
There is a typo at line:
https://github.com/wills106/homeassistant-solax-modbus/blob/1ade3916f61a609188306d4590e04048c318ad00/custom_components/solax_modbus/plugin_growatt.py#L2227

It should read Gen4.

Is that the only error you can see?

uspino2, 2023-08-01:
I think everything else is fine. I double checked with screen captures I took before updating. Also, no orphaned entities on my install.

wills106, 2023-08-01:
Ok I'm just adding in PV4 for the larger Inverters as I missed it out. Then I'll release an update.

wills106, 2023-08-01:
Do you just have a single battery system on your Inverter? Or do you have a second one connected? I am guessing you have two battery connections on the Inverter?

Rasmusbangp, 2023-08-08:
@uspino2 Sorry for jacking your post, but as i see it, my issue is similar to what you describe- 
Was changing the "modbus gateway type" to auto query the key to your issue? 

Ive made my own post about it and hope some can help :) 
https://github.com/wills106/homeassistant-solax-modbus/issues/485

Thank u all :)

wills106, 2023-08-09:
> Not really an issue, but I'm getting this error now...
> 
> `Growatt_Modbus: read failed at 0x08: firmware_version`

Hopefully this is corrected in [2023.08.4b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.08.4b1)

uspino2, 2023-08-09:
Corrected indeed, thanks!