wills106/homeassistant-solax-modbus #441: Solax X1Boost Gen3 none battery inverter

mnh637, 2023-06-13:
Hi,
for information, fresh install of HA on a  nuc and fresh install of solax modbus,
Home Assistant 2023.6.1
Supervisor 2023.06.1
Operating System 10.2
Frontend 20230608.0 - latest
I have two identical inverters and this works for one but not the other, I'm waiting for the inverter to be updated as they are on different firmware versions.
"ver":"3.009.03"-working Solax-Home
"ver":"3.008.10"-not working Solax-Garage

[image]
Regards
Mark

wills106, 2023-06-13:
Those versions look like the Firmware for the PocketWiFi 3.0 and not the Inverter it's self.
It's possible the non working Inverter isn't on Firmware Arm 1.37 and above.

Do you have any errors in your log for SolaX-Garage?

mnh637, 2023-06-14:
Hi,
I use 
curl -d "optType=ReadRealTimeData&pwd=SRxxxxxxx" -X POST http://192.168.99.233
curl -d "optType=ReadRealTimeData&pwd=SRxxxxxxx" -X POST http://192.168.99.198
and get 
{"sn":"SRXXGarage","ver":"3.008.10","type":4,"Data":[2493,5,34,1917,1292,1,1,28,16,4994,2,19187,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,25,0,1651,0,0,0,0,0,0,65384,65535,2570,2,60219,1,0,32,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[5.000,4,"XB3350IXXXXXXX",8,2.25,0.00,1.42,0.00,0.00,1]}tail: house.txt: file truncated
{"sn":"SRDGHome  ","ver":"3.009.03","type":4,"Data":[2497,5,52,1252,1814,1,2,21,41,4991,2,22613,0,0,0,0,0,0,0,0,0,0,0,0,0,21,40,0,0,0,0,0,0,0,0,0,0,0,0,27,85,1671,0,14080,13199,8,0,0,0,0,0,0,0,0,0,34,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[5.000,4,"XB3350IXXXXXXX",8,2.29,0.00,1.47,0.00,0.00,1]}tail: house.txt: file truncated
when http to either Pocket wifi the firmware show the same 3.001.02
but internal code is
[image]
and
[image]

[image]

not sure about logs but see below?

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:5172
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 12 June 2023 at 05:48:52 (3 occurrences)
Last logged: 05:29:56

SolaX-Garage: cannot find serial number, even not for MIC

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:5229
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 12 June 2023 at 05:48:52 (3 occurrences)
Last logged: 05:29:56

unrecognized inverter type - serial number : unknown

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:5165
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 12 June 2023 at 05:48:52 (3 occurrences)
Last logged: 05:29:56

SolaX-Garage: reading serial number from address 0x300 failed; other address may succeed

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:5163
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 12 June 2023 at 05:48:41 (4 occurrences)
Last logged: 05:29:51

mnh637, 2023-06-14:
both inverters are 
[image]
installed in Feb23, ordered in November22, no stock when ordered so had to wait for them to come in, serials are 2 digits apart

mnh637, 2023-06-19:
Hi,
I've got Solax to update the inverter so they are both now3.009.03 but still have the same issue 3 entities on one and 20 one the other?? I've deleted it restarted HA and added back in but still does not show any data ??
Thanks
Mark

wills106, 2023-06-19:
I know the Integration isn't talking to the PocketWiFI over Modbus RTU, it's using Modbus TCP. But is the system setting 'ModbusRTUEnable' Enabled on the working Inverter and disabled on the non working Inverter?
Also is the Modbus address 1 on both Inverters? Or is it different on the non working?

mnh637, 2023-06-19:
Hi,
[image]
it's set the same on both. I called the inverters Solax-Home and Solax-Garage in HA to differentiate them, and have tried enabling ModbusRTU on the none working one but made no difference.
I've asked through my installer for the Pocket wifi to be updated too.
Thanks
Mark

wills106, 2023-06-19:
If it's disabled on both then it must only apply to the RS485 connection on the Inverters then.

mnh637, 2023-06-19:
anything I could try, upgrade to HAOS 10.3 or not?

wills106, 2023-06-19:
It won't make any difference.

Only other thing I can think of trying is to power cycle the Inverter.
There still seems to be stability issues with these PocketWiFi devices. I don't know if they are internally loosing the Modbus connection, I just know some people who had a previously working setup looses connection and they have to power cycle the Inverter to get it working again.

mnh637, 2023-07-07:
Hi,
1st inverter works OK with firmware version 1.47, second does not work with 1.42, just had that updated now running 1.48 which also works.
Cheers
M

mnh637, 2023-07-15:
Hi, 
Had an inverter fail, they have replaced with one on version 1.42, which did not work,  updated to 1.48 and all OK, they also said there was a 1.50 version that was not yet available.
Closing this now
Thanks
Mark

closed 2023-07-15 by mnh637