wills106/homeassistant-solax-modbus #443: Sofar Solar HYD 15KTL-3P RS485/Modbus connection could see few entities only

ppiwowar, 2023-06-14:
**Describe the bug**
A clear and concise description of what the bug is.

**Mandatory details**
1. Integration Version:
2. HA Version: Home Assistant 2023.6.1, Operating System 10.2
3. Inverter Brand: Sofar Solar HYD 15KTL-3P
4. Plugin Used: 2023.06.8
5. First 6 digits of Serial: unknown as logger LSE-3 doesn't work
6. Connection Method:

HA(RPI4-8GB) is connected (LAN Ethernet) through Adapter (Ethernet<->RS486) directly to Sofar RS485 port.
I see only temperature entities' values. The others are 'unknown'. 

[image]
[image]

wills106, 2023-06-14:
You should have a error message in your HA log listing your Serial as an unrecognised Inverter.
What's the first 4 digits that it returns?

ppiwowar, 2023-06-14:
This all I have in the log file after a fresh ha restart:

2023-06-14 17:03:03.631 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-06-14 17:03:03.633 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration mobile_alerts which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-06-14 17:03:03.634 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

I am not sure, I have a communication problem with Sofar, as I see some data ....?

ppiwowar, 2023-06-15:
Additional info:
Connection Method: ModBus TCP using WaveShare RS485 TO ETH (B)

ppiwowar, 2023-06-15:
It took me the entire night. It was all about timeouts on WaveShare RS485 TO ETH (B).
Wills106, your integration is fine. thank you!
Closing, .... but still observing the stability of data flow.

wills106, 2023-06-15:
If you still need to make further changes can you let me know and I'll update the wiki.

Did you also see the note in the wiki:

In "More Advanced Aettings...",
- Sofar Solar set Modbus Gateway Type to "Auto query storage type"

ppiwowar, 2023-06-15:
Thanks. I do have the DIN rail mounted mode, but I have no Windows .... so I couldn't use vircom. Shooting blindly I modified:
- Enable multi-host = yes
- Rs485 conflict time gap = 50ms
It started to work then, but I am still testing ....
I am really newbie for HA - where to check for these  "blocking call inside the event loop", pls ?

wills106, 2023-06-15:
You should only get "blocking call inside the event loop" when using a RS485-USB adaptor.
It doesn't affect Modbus TCP

wills106, 2023-07-31:
Did you get this working correctly?
Just wondering if I can close this issue down?

ppiwowar, 2023-07-31:
Let me test it for a week or two. From time to time I have a strange thing - PV Energy and Export Energy meters go backwards?! Pls see HA screen dump. Also changes of entity values come late. Sometimes for some entities I need to wait for 10-20 minutes.

wills106, 2023-07-31:
I think some of the setting you need to set can only be configured through the Vircom tool. So it might be worth setting up a VM for Windows or taking the Waveshare round to a friends house running Windows so you can check all of the settings correctly.

wills106, 2023-08-16:
Can this be closed down now?
Were you able to setup your Waveshare with the Vircom tool?

ppiwowar, 2023-08-17:
Although I have not been able to setup the converter with Vircom, please do close the issue.

closed 2023-08-17 by wills106