wills106/homeassistant-solax-modbus #444: Define Solis 'S6-GR1P(2.5-6)K' Single Phase PV Inverter

wills106, 2023-06-15:
s/n is 16 characters long and starts with `1802`

I have the 4-pin S2-WL-ST datalogger (LAN and WiFi?), some versions of which are supposed to expose ModBus TCP if The Internet™ is to be believed, but mine doesn't appear to do anything like that (in fact the status page has 0 for every value it displays on there), and nmap only reports Port 80 (HTTP) and 53 (DNS) being open. It does, however, talk to Solis Cloud just fine.

I've also hooked up the comm port to a Waveshare PoE Modbus TCP DIN device, but haven't had any success reading anything of it (yet).

_Originally posted by @benlancaster in https://github.com/wills106/homeassistant-solax-modbus/discussions/188#discussioncomment-6180055_

wills106, 2023-06-15:
@benlancaster

I have just added your Inverter into the Beta [2023.06.9b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.06.9b1) I believe your Inverter is PV only and not a Hybrid?

At the moment plugin_solis is only setup with the entities define for Hybrid Inverters. I have set your Inverter as a Hybrid for the time being otherwise it wouldn't return anything.

Are you able to let me know which entities return values and which don't.
I can then remove all of the non working Modbus registers from PV Inverters.

> I have the 4-pin S2-WL-ST datalogger (LAN and WiFi?), some versions of which are supposed to expose ModBus TCP

I know on the Solis-DLS-L it uses port 8899, it might be worth checking that port just in case?

benlancaster, 2023-06-15:
Thank you!

It looks like I have some modbus gremlins to work out here first:

```
2023-06-15 04:03:02.507 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] solis_inverter: reading serial number from address 0x80ec failed; other address may succeed
2023-06-15 04:03:02.523 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] solis_inverter: cannot find serial number, even not for other Inverter
2023-06-15 04:03:02.525 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized solis_inverter inverter type - serial number : unknown
2023-06-15 04:03:07.555 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis_old] solis_inverter_old: reading serial number from address 0xbf5 failed; other address may succeed
2023-06-15 04:03:07.566 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis_old] solis_inverter_old: cannot find serial number, even not for other Inverter
2023-06-15 04:03:07.568 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis_old] unrecognized solis_inverter_old inverter type - serial number : unknown
```

I'm not sure what's going on there but I will dig some more later.

benlancaster, 2023-06-15:
> I know on the Solis-DLS-L it uses port 8899, it might be worth checking that port just in case?

Huh, well this is curious:

```
% nmap -PN -p8899,502 192.168.1.205
Starting Nmap 7.93 ( https://nmap.org ) at 2023-06-15 09:26 BST
Nmap scan report for 192.168.1.205
Host is up (0.0031s latency).

PORT     STATE  SERVICE
502/tcp  open   mbap
8899/tcp closed ospf-lite
```

So it looks like port 502 *is* open

benlancaster, 2023-06-15:
Some more weirdness - this is the web interface on the S2-WL-ST stick (hooked up via Ethernet). 

[image]

The stick is happily reporting valid and correct data to Solis Cloud, even though the status page is a bunch of zeros. It's almost like it's not using modbus but some other protocol (which would explain why I'm not getting anywhere with my Waveshare PoE Modbus box)

The stick seems to think it's a WiFi stick - there are wireless network settings elsewhere on there but I don't see an SSID that matches, nor is it capable of searching for SSIDs in range 🤷🏼‍♂️

alienatedsec, 2023-06-15:
The stick is RS485 Modbus RTU and Waveshare translates it to Modbus TCP. Recheck baud rates on the inverter and cabling as this must work.

benlancaster, 2023-06-15:
> The stick is RS485 Modbus RTU and Waveshare translates it to Modbus TCP. Recheck baud rates on the inverter and cabling as this must work.

I don't see any options on the inverter to set/check the baudrate.

alienatedsec, 2023-06-15:
It should be 9600 then. Check the slave ID ADDRESS - point 6.3.2 of the manual

https://www.ginlong.com/uploads/file/Solis_Manual_S6-GR1P(2,5-6)K_FN_EUR_V1,2(20221116).pdf

> I don't see any options on the inverter to set/check the baudrate.

benlancaster, 2023-06-15:
OK, some success!

[image]

Cross-checking with [this document](https://ginlongsolis.attachments8.freshdesk.com/data/helpdesk/attachments/production/36112313359/original/RS485_MODBUS%20Communication%20Protocol_Solis%20Inverters.pdf?response-content-type=application%2Fpdf&Expires=1686781847&Signature=Mv7vTmtYnBg6Vb8uLUhYwrA5j29fHi1bFgBLTZQ3XJI7BubHrFTfRQDWGL0nxZC-IiLXBn7U-KNiv1hQPf5a4rYUpu8bYuQl6H~MuwHJdafMQ1bUgDFWpKSz8-0OEXJqj9Pu-tyzZx-hSwOJ9aJMuIoHf~d5Qu5A5ZVw0jVvUZWBrDfdU7K7Sll4CAH6VATzr36E3fBWno5wQctjoOdU1fAOfnocgtzyz8bqgPPiNxIWK96vn2OZU8IG9vQDNNwOaphHrXThrSXDmMJAgBjN3rdmx1V~GS6pxMadPa34Tt9hmdgRYJfveTy2Ejw0Uxq6eZkk4EXDvlnzZSv0a8P1-Q__&Key-Pair-Id=APKAJ7JARUX3F6RQIXLA) and those values appear to match up.

This is using the S2-WT-SL stick, and port 8899.

I'm no closer to getting it hooked up to Home Assistant:

```
2023-06-15 11:24:59.851 DEBUG (MainThread) [custom_components.solax_modbus] setup entries - data: {}, options: {'name': 'S2-WL-ST', 'interface': 'tcp', 'read_modbus_addr': 1, 'plugin': 'solis', 'scan_interval': 15, 'read_eps': False, 'read_dcb': False, 'read_pm': False, 'host': '192.168.1.205', 'port': 8899, 'tcp_type': 'rtu'}
2023-06-15 11:24:59.852 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solis
2023-06-15 11:24:59.852 DEBUG (MainThread) [custom_components.solax_modbus] Setup solax_modbus.S2-WL-ST
2023-06-15 11:24:59.852 DEBUG (MainThread) [custom_components.solax_modbus] solax serial port /dev/ttyUSB0 interface tcp
2023-06-15 11:24:59.852 DEBUG (MainThread) [custom_components.solax_modbus] solax modbushub creation with interface tcp baudrate (only for serial): 19200
2023-06-15 11:24:59.853 DEBUG (MainThread) [custom_components.solax_modbus] S2-WL-ST: ready to call plugin to determine inverter type
2023-06-15 11:24:59.853 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] S2-WL-ST: trying to determine inverter type
2023-06-15 11:24:59.968 ERROR (MainThread) [pymodbus.logging] Connection to (192.168.1.205, 8899) failed: [Errno 113] Host is unreachable
2023-06-15 11:24:59.972 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] S2-WL-ST: attempt to read serialnumber failed at 0x80ec
Traceback (most recent call last):
File "/config/custom_components/solax_modbus/plugin_solis.py", line 53, in _read_serialnr
inverter_data = hub.read_input_registers(unit=hub._modbus_addr, address=address, count=8)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solax_modbus/__init__.py", line 342, in read_input_registers
return self._client.read_input_registers(address, count, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 116, in read_input_registers
return self.execute(
^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 191, in execute
raise ConnectionException(f"Failed to connect[{str(self)}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.205:8899)]
2023-06-15 11:24:59.979 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] S2-WL-ST: reading serial number from address 0x80ec failed; other address may succeed
2023-06-15 11:24:59.982 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] Read S2-WL-ST 0x80ec serial number: None, swapped: False
2023-06-15 11:24:59.982 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] S2-WL-ST: cannot find serial number, even not for other Inverter
2023-06-15 11:24:59.985 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized S2-WL-ST inverter type - serial number : unknown
```

That appears to suggest a network problem rather than something else, which is strange as my HA Yellow is on the same network as my computer. More digging required

alienatedsec, 2023-06-15:
> More digging required

19200 in the logs. Is that correct? It will disconnect the cloud if you use a data logger as the data comm to HA

alienatedsec, 2023-06-16:
> Cross-checking with [this [...]

@benlancaster 
It seems inaccessible to me - maybe the link to a source of the page rather than a document would be better.

Also, how are you with your connection problems?

benlancaster, 2023-06-19:
> Also, how are you with your connection problems?

Connection problems resolved. I've ditched the S2-WL-ST stick and just going straight from the inverter to the Waveshare PoE box. It took some doing to get it working but seems to be fairly bulletproof right now using the built-in Modbus integration in HA.

I've been able to poll all the stuff I care about:

 - PV String 1 Voltage & Current
 - PV String 2 Voltage & Corrent
 - Various energy generation meters
 - Generation power right now

>  It seems inaccessible to me - maybe the link to a source of the page rather than a document would be better.

I've updated [the link in the original comment](https://github.com/wills106/homeassistant-solax-modbus/issues/444#issuecomment-1592782024)

At this stage, I don't think I need a fully fledged integration, the built-in Modbus stuff is fine.

Sharing my rudimentary YAML configuration for posterity.

```yaml
  - name: "solis"
    type: tcp
    delay: 5
    timeout: 5
    host: !secret solis_ip
    port: 502
    sensors:
        - name: Solar Generation
          slave: 1
          address: 3004
          count: 2
          input_type: input
          data_type: uint32
          scan_interval: 15
          device_class: power
          unit_of_measurement: W
        - name: Solar String East Voltage
          slave: 1
          address: 3021
          count: 1
          input_type: input
          data_type: uint16
          scan_interval: 15
          device_class: voltage
          unit_of_measurement: V
          scale: 0.1
          precision: 1
        - name: Solar String East Current
          slave: 1
          address: 3022
          count: 1
          input_type: input
          data_type: uint16
          scan_interval: 15
          device_class: current
          unit_of_measurement: A
          scale: 0.1
          precision: 1
        - name: Solar String West Voltage
          slave: 1
          address: 3023
          count: 1
          input_type: input
          data_type: uint16
          scan_interval: 15
          device_class: voltage
          unit_of_measurement: V
          scale: 0.1
          precision: 1
        - name: Solar String West Current
          slave: 1
          address: 3024
          count: 1
          input_type: input
          data_type: uint16
          scan_interval: 15
          device_class: current
          unit_of_measurement: A
          scale: 0.1
          precision: 1
        - name: Solar Inverter Temperature
          slave: 1
          address: 3041
          count: 1
          input_type: input
          data_type: uint16
          scan_interval: 15
          device_class: temperature
          scale: 0.1
          precision: 1
          unit_of_measurement: °C
        - name: Solar energy today
          slave: 1
          address: 3014
          count: 1
          input_type: input
          data_type: uint16
          scan_interval: 15
          device_class: energy
          unit_of_measurement: kWh
          precision: 1
          scale: 0.1
        - name: Solar Energy All Time
          slave: 1
          address: 3008
          count: 2
          input_type: input
          data_type: uint32
          scan_interval: 500
          device_class: energy
          unit_of_measurement: kWh
```

wills106, 2023-07-29:
Just released [2023.07.4b3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.4b3) with the smaller 1-3K PV Only Inverters.
Still only configured as Hybrid. Require feedback on which Registers respond on the PV Only Inverter.

tomschlde, 2023-07-29:
Hi all,
can you please add Inverter Solis : '1801' PV only S6-GR1P1.5K like you added the 6KW Model before in [2023.07.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.1) Should run with the same settings, same type of inverter, just different power.

Added serial '1802' PV Only S6-GR1P6K, configured as a Hybrid at the moment.

TIA
Thomas

wills106, 2023-07-29:
I have added it into [2023.07.4b3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.4b3)
I just meant can you comment on Feedback which registers / sensors I need to remove from the Integration for PV only Inverters.

wills106, 2023-08-03:
Has anyone been able to test the Integration with PV only Inverters?
In the current setup which Sensors return values / what returns either nothing or unknown.

I can then remove the unused sensors from the PV Inverters.

wills106, 2023-08-16:
Closing down as no feedback.

Please feel free to issue a PR or Feature Request for any improvements.

closed 2023-08-16 by wills106