wills106/homeassistant-solax-modbus #447: [BUG] Solax Inverter suddenly not recognised

nikotime, 2023-06-17:
**Describe the bug**
A clear and concise description of what the bug is.

**Mandatory details**
1. Integration Version: 2023.06.8
2. HA Version: 2023.6.2
3. Inverter Brand: Solax
4. Plugin Used: HACS
5. First 6 digits of Serial: SVTU8R
6. Connection Method: Local IP

**Detailed Error Log**
2023-06-17 14:47:09.392 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaXModbus: reading serial number from address 0x0 failed; other address may succeed
2023-06-17 14:47:09.492 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaXModbus: reading serial number from address 0x300 failed; other address may succeed
2023-06-17 14:47:09.493 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaXModbus: cannot find serial number, even not for MIC
2023-06-17 14:47:09.493 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown

**Additional context**
Hi, as of around 11.30am today I stopped receiving any data into Home Assistant from this integration - it has been working without issue for a couple of months to this point. It stopped working whilst the rest of HA continued to run, and I moved to the latest HA 2 days prior. I get the error that it can't find the serial number any more.

Steps to try and resolve:
Rebooting
Testing beta Solax integration and downgrading to previous version
Restoring 2023.6.1 backup from few days ago
Deleting integration completely and starting again. With this I can see four entities: Battery Awaken, Inverter Temperature, Unlock Inverter, Unlock Inverter Advanced. Clicking on Unlock Inverter and rebooting doesn't do anything. 

I have also been able to curl the local IP with this command:`curl -d "optType=ReadRealTimeData&pwd=SVTU8RXXXX" -X POST http://192.168.1.XXX`

And I seem to get all the correct data:
{"sn":"SVTU8RXXX","ver":"3.008.10","type":15,"Data":`[2444,110,2697,5003,1436,1753,87,87,1268,1533,2,18736,0,120,24360,0,0,32,100,7193,0,8246,0,114,100,0,44,3793,0,0,0,0,2119,0,57930,0,61177,0,578,42,256,2628,1800,0,350,239,217,35,34,68,1,1,2815,0,18076,0,62841,65535,65386,65535,0,0,0,0,0,0,0,0,0,0,0,0,60039,65535,3000,0,62536,65535,480,0,930,0,0,0,0,119,24,89,0,0,350,0,360,0,0,0,0,0,0,9744,4366,5894,1115,258,9766,9766,0,0,0,0,1,2466,0,0,3482,3399,40386,24,21302,14389,18757,12354,16695,12360,14135,21302,14389,18757,12354,16695,12360,14135,21302,14389,18758,12354,16690,12616,13361,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,2817,4612,258,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[5.000,15,"H4502AI5XXXX",8,1.33,0.00,1.32,1.04,0.00,1]}`

Help would be really appreciated! Solax Cloud and the app continue to work

infradom, 2023-06-17:
The curl output is not relevant as this integration does not use http interface but the modbus over tcp interface.
The SVTU serial number prefix is not in the supported list yet, I guess this is a pocketlan/pocketwifi serial number instead of the inverter serial number.
You probably use a pocketwifi adapter with a Gen4 inverter and it is known that these adapters only work for certain versions in a stable way. See other threads if you are using a Gen4. Using a modbus to Ethernet adapter (e.g. Waveshare)  is also the recommended way if you do not have a compatible pocketwifi adapter.

nikotime, 2023-06-17:
Thanks @infradom . Any ideas why it would be working fine for months and then suddenly not be recognised? Not sure why / how the serial number would change

infradom, 2023-06-17:
> Thanks @infradom . Any ideas why it would be working fine for months and then suddenly not be recognised? Not sure why / how the serial number would change

This seems to be typical behaviour of the pocketwifi interface observed by many users. It was never meant  to be a modbus interface afaik. 
My biggest concern would be security however ... If I underrstood it right, the protection of this wifi network is unexisting, so anyone could hack your intallation (assuming you use the solax created wifi hotspot).
I might be wrong, since I have no hands-on experience as I stay away from this wifi interfaces for security reasons

infradom, 2023-06-17:
The serial number is not changing. your curl interface shows the serial number of the pocketlan adapter. Your serial number is further down the curl output : H450...

nikotime, 2023-06-17:
@infradom Whoops - sorry about that! Updated the original post.

nikotime, 2023-06-17:
With that very helpful steer, I note this lovely message on the main page:

Pocket WiFi 3.0 with Firmware V3.004.03 and above is only [officially supported]

My firmware is 3.001.02 so I must have been lucky in the past! I will try and scour the internet to find a zip file for the latest firmware and hope that resolves my issue, but leaving this open incase any other kind soul has an idea (or even better - where to source the firmware!)

wills106, 2023-06-17:
The only real suggestion I have is to do a power cycle of the inverter.

mnh637, 2023-06-19:
Hi,
for infomation
I've got an issue open too, I've got two inverters, identical, both pocket wifi have version 3.001.02 and the inverters did have different firmware versions with version 3.009.03 working and 3.008.10 not, Solax have updated them to the same 9.03 and still one works with HA the other does not?
Thanks
Mark

nikotime, 2023-06-19:
To confirm, Solax have updated my firmware to 3.009.03 and all works for me. I note you've raised your bug in a new thread @mnh637 so I Will close this bug.

Thanks so much @wills106 and @infradom for your support, fingers crossed it stays working for me!

MrStanda, 2023-08-11:
@nikotime where did you get your update firmware for the pocket wifi? I have the same issue but I can't find the updated firmware file

nikotime, 2023-08-11:
> @nikotime where did you get your update firmware for the pocket wifi? I have the same issue but I can't find the updated firmware file

Just email SolaX support and they had to do it remotely. I never received the files

MrStanda, 2023-08-11:
Ok, thank you for your answer