wills106/homeassistant-solax-modbus #448: Sofar

juraurban, 2023-06-18:
Sofar HYD 10KTL-3PH

hruskaexe, 2023-09-06:
Hello, I have the same inverter Sofar HYD 10KTL-3PH and the same problem - reading is OK, writing most parameters is not OK (not only battery control). 

I have the recommended modbus communicator (from your wiki) - USR-W610 RS232/RS485 to Wifi & Ethernet. I communicate with him via wifi (RTU over TCP, inverter type sofar).

- When I write the parameter, after some time the previous value is returned back. List of parameters are - Battery Active Control, Battery Minimum Capacity, Battery Minimum Capacity OffGrid, Time of Use On-Off, Time of Use Charge SOC
- When I write the reflux power for example to the value 4500 W, the value changes to 45 W, but after a while it changes and writes to the inverter to 4500 W.
- When I press the "Reflux Control" button, I get an error - failed to call the button press service, 'H' format requires 0 <= number <= 65535

Thank you for any help or support.

fontexD, 2023-10-10:
same issue here with growatt, used to work as i rmb but now it dosent

Matz88, 2024-01-17:
I have the same issue with an HYD3600, a while after setting a battery capacity, it come back to the original value, maybe some button has to be pressed to set the values like for the reflux? Anyone can tell?

cschlipf, 2024-02-19:
Duplicate of #703 

This never worked and implementation will be very risky, so I would rather opt for removing this. Can you explain your use case? Maybe there is another solution to what you would like to achive?

wills106, 2024-02-20:
I'm going to close this issue down so we can all focus on #703 as it has more recent activity.

closed 2024-02-20 by wills106