wills106/homeassistant-solax-modbus #449: [BUG] X1-Hybrid-G4 No longer retrieving any useful entities

obiphi, 2023-06-19:
Hi, HA newb here so any assistance (and forgiveness for ignorance) would be greatly appreciated!

I have two inverters; a Solax X1-Hybrid G4 (Serial starts: H4502AI) with Solax Triple Power Battery, and X1-Mini (Serial starts XM3A20I). Both were being queried with Modbus over TCP and everything was working. Then I updated HA to 2023.6.2 and both stopped retrieving any entities with useful information; everything was unknown or 'entity no longer retrieved'. I then ran `ha core update --version 2023.6.1` to downgrade to 2023.6.1 and it still didn't work, so then I downgraded to 2023.5.4 and now the X1-Boost is working again but the X1-Hybrid-G4 is still not working. The Logs contained lines like:

> 2023-06-19 10:58:54.842 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solax
> 2023-06-19 10:58:54.844 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: trying to determine inverter type
> 2023-06-19 10:58:55.062 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX 0x0 serial number before potential swap: H4502AIXXXXXX
> 2023-06-19 10:58:55.070 INFO (MainThread) [homeassistant.components.button] Setting up button.solax_modbus
> 2023-06-19 10:58:55.072 INFO (MainThread) [homeassistant.components.number] Setting up number.solax_modbus
> 2023-06-19 10:58:55.074 INFO (MainThread) [homeassistant.components.select] Setting up select.solax_modbus
> 2023-06-19 10:58:55.077 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solax_modbus
> 2023-06-19 10:58:55.081 INFO (MainThread) [custom_components.solax_modbus.button] hub.wakeuButton: SolaxModbusButtonEntityDescription(key='battery_awaken', device_class=None, entity_category=None, entity_registry_enabled_default=True, entity_registry_visible_default=True, force_update=False, icon='mdi:battery-clock', has_entity_name=False, name='Battery Awaken', translation_key=None, unit_of_measurement=None, allowedtypes=0, register=86, command=1, blacklist=None, write_method=1, value_function=None, autorepeat=None)
> 2023-06-19 10:58:55.087 INFO (MainThread) [custom_components.solax_modbus.sensor] computedRegs: {}

I can't see anything wrong with that except that it isn't getting the ~200 entities that it was getting before. Where do I go from here debugging this issue?

**Mandatory details**
1. Integration Version: 2023.06.8
2. HA Version: Currently 2023.5.4, but it was also failing on 2023.6.2 and 2023.6.1
3. Inverter Brand: Solax
4. Plugin Used: Solax Inverter Modbus
5. First 6 digits of Serial: H4502A
6. Connection Method: TCP to the Solax PocketWifi

If it's useful, calling the RESTful API on the device shows 'ver: 3.006.04' and 'type: 15'

Is the HA update likely to have caused this issue, or is that a red herring?

obiphi, 2023-06-19:
Thanks @wills106, I followed some youtube guides about how to reset the inverter but couldn't find separate DC breakers so I settled for the AC inverter plus the 'DC Switch' on the base of the unit, but the inverter never fully powered off. In lieu of this I pulled the PocketWifi USB out of the inverter (as you mentioned it was the PocketWifi with the issue), left it out for a few seconds and then put it back in. Still getting the same entries in the log, and no additional entities found. Would that level of reset be sufficient?

Any other ideas?

wills106, 2023-06-19:
Within LCD menu isn't there a system switch? It might be slightly different on your Gen4 to my Gen3

obiphi, 2023-06-19:
Yes, I spoke to Solax support and he guided me through the LCD menu to switch off the system, but it still didn't fully switch off the inverter like I naively expected it to. Still no change on the modbus returns either

gittan443, 2023-06-28:
I am also experiencing the same issue.

obiphi, 2023-06-28:
Thanks @gittan443, just to add that despite:

- Many attempts at inverter reboots 
- Many attempts at pulling the PocketWiFi module
- Getting firmware upgrades (as verified by calling the RESTful API). I now have the same firmware on both my PocketWiFi modules

I'm still experiencing the issue on one of my two inverters.

gittan443, 2023-07-06:
Ok, so yesterday I did a complete poweroff of the inverter (haven't had time until now). And now the modbus data is presented again in HomeAssistant.

Steps to poweroff:
1) Cut AC-Breaker
2) Cut DC-breaker under the SolaX inverter (on my bottom left) 
3) Power-off battery stack
4) Wait for poweroff to complete and the LCD panel go blank

Poweron:
1) Restore AC (and smart meter connection)
2) Restore DC
3) When waiting for battery is displayed, startup battery pack.
4) wait one hour with daylight, restart Home Assistant

Alfred99-HA, 2023-07-17:
Hi, am am newbi and expiernce same problems with only 4 entities with my Solax X3 G3 Inverter. Actually I did shut down only the switch and it worked once - I got more than 200 entities and was happy. However a bit later they became unavailable again. I did remove Solax inverter modbus several times and installing again, restart, reboot, remove LAN-cable, shut down the switch, removed, check the Modbus Port 502 - nothing brings me back the entities. See the log-entries below.
2023-07-17 16:41:03.033 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX X3 Hybrid 10.0 kWp: reading serial number from address 0x0 failed; other address may succeed
2023-07-17 16:41:03.057 ERROR (MainThread) [pymodbus.logging] Connection to (192.168.0.228, 502) failed: [Errno 104] Connection reset by peer
2023-07-17 16:41:03.063 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX X3 Hybrid 10.0 kWp: attempt to read serialnumber failed at 0x300
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 59, in _read_serialnr
    inverter_data = hub.read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 336, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 100, in read_holding_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 191, in execute
    raise ConnectionException(f"Failed to connect[{str(self)}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.0.228:502)]

Is it possible to configure the Serial number manually and would this help ?

The SOLAX Inverter modbus looks quite powerful and I am eager to take advantage of it.
Any help yould be highly appreciated.

wills106, 2023-07-19:
Unfortunately it still appears that the PocketWiFi is unreliable at times.
That's why it's recommended to use an RS485 adaptor like a Waveshare and bypass the PocketWiFi altogether.

Alfred99-HA, 2023-07-19:
It was a different problem - I am using a LAN connection anyway, but another gateway (neoom beaam) listening to the SOLAX modbus prevented HA to read the modbus currectly. Thus modbus connection might be "corrupted" by other means as weak WIFI as well. I am writing this in case someone else has  similar setup and can apply this as well

wills106, 2023-07-23:
Closing down as not an Integration issue and updated [Readme](https://github.com/wills106/homeassistant-solax-modbus#support-modbus-over-rs485--tcp)

closed 2023-07-23 by wills106