wills106/homeassistant-solax-modbus #452: Sph 800

fontexD, 2023-06-24:
I tried installing the solax modbus for sph 8000 3 phase

but getting these errors

growatts: read failed at 0x61: battery_voltage
15.29.29 – (ADVARSEL) SolaX Inverter Modbus (brugerdefineret integration) - beskeden forekom først kl. 14.49.54 og ses 154 gange
:0:0 Script error.
15.27.39 – (FEJL) components/system_log/__init__.py - beskeden forekom først kl. 14.50.42 og ses 11 gange
unrecognized skur inverter type - serial number : unknown
15.24.58 – (FEJL) SolaX Inverter Modbus (brugerdefineret integration) - beskeden forekom først kl. 15.15.39 og ses 2 gange
skur: cannot find firmware version, even not for other Inverter
15.24.58 – (FEJL) SolaX Inverter Modbus (brugerdefineret integration) - beskeden forekom først kl. 15.15.39 og ses 2 gange
skur: reading firmware number from address 0xbb9 failed; other address may succeed
15.24.58 – (ADVARSEL) SolaX Inverter Modbus (brugerdefineret integration) - beskeden forekom først kl. 15.15.39 og ses 2 gange
skur: reading firmware number from address 0x9 failed; other address may succeed
15.24.52 – (ADVARSEL) SolaX Inverter Modbus (brugerdefineret integration) - beskeden forekom først kl. 15.15.33 og ses 2 gange
instance name SolaX already defined or default name for non-default inverter
15.14.57 – (ADVARSEL) SolaX Inverter Modbus (brugerdefineret integration)

is the invertenr not supporter ?

i have a sph 3600 where it works perfect

exico91, 2023-07-18:
Hello fontexD, not related to your problem but i have a SPH 6000 that Im trying to get to work, can you tell me step by step what you did?  My issue is number 469. I can give you a mail in case. 
Sorry for the intrusion

fontexD, 2023-07-25:
On my topic I wrote it's not working will sph 8000 be supported? 😁 So I'm awaiting response aswell if it's gonna be added guess it's the same thing for sph6000 :) I think only sph3600 works for now

wills106, 2023-07-25:
It should work, how are you trying to connect to your Inverter?

exico91, 2023-07-25:
For the SPH 6000 i used a cat6 ethernet cable and used the RS485 port, on the other side i connected the Blue wire to A and the striped blue to B. I checked with a multimeter if there is a connection and its fine but there is no traffic on the bus, I tried with a scanner and 9600 and 115200 baud rates and nothing answers.

I checked on the inverter and there is only one menu item for the RS485 and let me set it to VPP or Battery. Tried both.

What can i check?

I should add that we have a ShineLanBox with a USB transmitter on the inverter and I cannot get it to work, I also asked the electrician that installed it and followed the instructions, the thing pairs but there is no data. Im starting to make an educated guess and say that there is a problem with the inverter.

fontexD, 2023-07-25:
Same way as I did with sph3600 same cables same devices :) using rj45 to usb ttl Sent from Android device25. jul.

exico91, 2023-07-25:
I used this pipece of software https://shop.marcomweb.it/en/shop-online/software/utility/modbus-scan-dettagli.html

It lets you scan Modbus via serial or ethernet, you can set a register, I think that i set 14 but i tried with others too. I can see the TX led on the USB adapter blinking when it scans but no RX or results on the software

wills106, 2023-07-31:
@fontexD Did you get this working with your SPH800 (SPH8000?)

closed 2023-08-14 by wills106