wills106/homeassistant-solax-modbus #453: Solis - Certain entities intermittently unknown

dolce08, 2023-06-28:
Inverter: Solis RHI-6K-48ES-5G
Integration Version: 2023.03.1.23
HA version: 2023.6.3
First 6 digits of serial: 60310
Connection method: LAN

Any release after 2023.5.x results in certain entities intermittently showing as unknown. These entities include (ones I use):
House Load Today
Grid Import Today
Grid Export Today
Battery SOC
Power Generation This Month
Power Generation This Year

Most of the others are unaffected. I've check the code and I can see any changes. block size is 40 in both cases. Debugging is enabled but nothing shows in logs. Any help would be greatly appreciated.

Graph showing issue. I reverted to 2023.5.1 at 09.46, which you can see fixes the issue.

[image]

alienatedsec, 2023-06-28:
@dolce08 what is the inverter firmware version? It has 6 digits

What solution are you using to connect via LAN?

> I reverted to 2023.5.1

Have you reverted HA or the integration?

dolce08, 2023-06-28:
@alienatedsec inverter firmware is 3D0037. 
I reverted the integration back to 2023.5.1. HA remains on the latest version. 
I’m using the Solis DLS-L LAN data logger connected directly (no wave share).

alienatedsec, 2023-06-28:
> inverter firmware is 3D0037.

This is a really old version - I guess you got your inverter mid-last year - it comes with many flaws and issues that may not sound related, but those are very relevant for the inverter to work properly - the latest stable version is 470044, so raise a ticket with Solis for an upgrade to that version if you can.

Is it the same with `2023.05.2b1`? That would be the next beta version after `2023.5.1`.

https://github.com/wills106/homeassistant-solax-modbus/compare/2023.05.1...2023.05.2b1

dolce08, 2023-06-29:
Thanks for the firmware information. I've raised a ticket.

2023.05.2b1 causes the same issue. It looks like it's the 'sleepmode = SLEEPMODE_NONE' causing the issue. The registries this was changed for are the ones which I'm having issues with

alienatedsec, 2023-06-29:
@dolce08 What's the firmware on your DLS-L stick?

dolce08, 2023-06-29:
DLS-L firmware is ME_0D_TCP_0501_5.0B

alienatedsec, 2023-06-29:
> DLS-L firmware is ME_0D_TCP_0501_5.0B

Is that what is shown when you logged in to the datalogger web interface?

dolce08, 2023-06-29:
Yes

alienatedsec, 2023-06-29:
Whats its serial number? At least the first 6 digits

dolce08, 2023-06-29:
First 6 digits are 192177

alienatedsec, 2023-06-29:
It's a newer `DLS-L` with newer firmware. However, I don't think it is still supported by Solis and even when you ask for an update, they will ask you to replace it with `S3-WIFI-ST`.

I have `ME-0C-0501-5.08U`, which I could share for you to try, but there is no comeback as I have no other versions - so I don't recommend it. Yours seems to support Modbus TCP as it is in the firmware name `ME_0D_TCP_0501_5.0B`. There are reports of Modbus TCP being disabled with other firmware versions.

I suspected `sleepmode = SLEEPMODE_NONE` to be the cause, but it is weird as this is to improve things and not make them worse. Unless `DLS-L` cannot work in that mode with this `ME_0D_TCP_0501_5.0B` firmware version.

You have several options:
1. Change to `sleepmode = SLEEPMODE_LASTAWAKE` in each release manually by editing `plugin_solis.py`
2. Ask for enhancement and to set the sleepmode when the integration is commissioned. That is some additional effort from @wills106 and @infradom 
3. Swap to any supported solution, which doesn't rely on Solis datalogger. This comes with the cost and some hands-on experience.

dolce08, 2023-06-29:
Thanks for all the information. I've manually updated sleepmode = SLEEPMODE_LASTAWAKE in the latest release and it works as expected. I'm happy with this approach. 

Yes, I know the DLS-L isn't officially supported by Solis anymore but it works for my needs. If/when it stops working I'll cross that bridge when I come to it.

Thanks again for the support.

alienatedsec, 2023-06-29:
@wills106 its worth adding a comment for `DLS-L` and reference to this issue in this section https://github.com/wills106/homeassistant-solax-modbus/wiki/Installation-Notes#solis-inverters

ciaranmcgirl, 2023-06-30:
I hit the same issue with the same details as the OP but my connection method was via the [Jumpy07 - Solis---SolisCloud-and-Home-Assistant ](https://github.com/Jumpy07/Solis---SolisCloud-and-Home-Assistant) project. 

changing the `SLEEPMODE_LASTAWAKE` resolved it.

infradom, 2023-06-30:
I can imagine that SLEEPMODE_LASTAWAKE is the better choice for entities that increment over time.
For entities that give the actual value at the current time  like current PV power or current, SLEEPMODE_NONE or _ZERO might be the better choice

ceemjay, 2023-07-27:
For the record (with the hope of getting this fixed) I am having the same problem - using the DSL-L  too. 

Should I replace all 20 or so instances to sleepmode = SLEEPMODE_LASTAWAKE?

Thanks

dolce08, 2023-07-27:
@ceemjay yes, updated all sleepmode entities to SLEEPMODE_LASTAWAKE. Takes a couple of minutes and I haven’t had any issues since.

ceemjay, 2023-07-27:
thanks have done that now. if the software isn't changed I will write a script to edit those entries for future module updates

wills106, 2023-07-27:
A number of different "sleepmode = xyz" variations have been tested. I think a lot of the Solis instabilities have been worked on either through Solis Firmware updates or Waveshare updates. (Although it's difficult to keep track of the progress on all Inverters.

> if the software isn't changed I will write a script to edit those entries for future module updates

If the majority of Solis Inverter owners think "SLEEPMODE_LASTAWAKE" is the most reliable option, why don't you just submit a PR for the change?

ceemjay, 2023-07-27:
Thanks for the response.

Is the fact that dolce08 hasn't had any problems having made the changes deemed to be enough testing to make the changes permanent?

If you are happy to make those changes permanent and the changes shouldn't impact anoyne else then let's submit a PR.  Though I am a complete newbie and dont know what a PR is or how to do it - though I am sure I could find out!

Any other Solis inverter users here have a view or PR proficient?

alienatedsec, 2023-07-27:
> Any other Solis inverter users here have a view or are PR proficient?

I will test and submit PR if `SLEEPMODE_LASTAWAKE` is to be working on Waveshare too.

alienatedsec, 2023-07-28:
PR #478 submitted

wills106, 2023-07-29:
Just released [2023.07.4b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.4b2) with the changes.

dolce08, 2023-07-31:
Yes, 2023.7.4b2 working fine, thank you. Thank you @alienatedsec for the PR too

closed 2023-07-31 by dolce08