wills106/homeassistant-solax-modbus #456: "Solax Today's Solar Energy" value is not Solar Energy but inverter Energy

blb4github, 2023-07-03:
I have doubt about the "Solax Today's Solar Energy" value. I have batteries connected to my inverter and I allow charging the batteries from mains and to me it looks like this sensor value is the AC Energy output of the inverter, not the Solar Energy. See attached graphs:
[image]

As far as I can see the same counts for "SolaX Total Solar Energy". I like to have a sensor that really gives the Solar Energy (per day / total) if possible.

_Originally posted by @blb4github in https://github.com/wills106/homeassistant-solax-modbus/discussions/125#discussioncomment-6333732_

PutoPunko, 2023-07-06:
Hi,

It is not happening to me, I have a X1-Hybrid-G3.

[image]

infradom, 2023-07-09:
My Gen4 X3 Hybrid system seems to work fine : (graphs produced by Grafana / InfluxDB combo):
[image]

blb4github, 2023-07-09:
Thanks, that looks better indeed. I have a Gen2 (SK-SU300E), I believe that's an older type.

Looking to all entities I found "Solax Today's Yield. This looks like a total Solar Energy Entity:
[image]

jsantosdias, 2023-07-09:
In My Gen2 it's similar. This photo was taken at 00:45
[image]

cewbdex, 2023-10-17:
I can try to fix this for you, if you find the modbus reference for your inverter.

finalbillybong, 2024-06-23:
Sorry to review an old thread.  This is exactly what is happening to mine. I have an X1-Hybrid-5.0 and am seeing the values of Todays Yield increasing overnight while solar generation is at zero. Did you get this solved?

closed 2025-10-05 by github-actions[bot]