wills106/homeassistant-solax-modbus #459: Strange responses: LSE-3 (Wired Lan Connection)

sanderbaks, 2023-07-05:
Hi all!

I just can't find any way to directly connect my Solar system to my Home Assistant configuration.
I spent quite some time using the Solarman Plugin, but that did not work. Helpful people told me that this plugin might be a better solution.

I own a Sofar 12KTL-G3 inverter. 3 fase, Serial number starts with SS2E.
It is connected with a datalogger from Solarman: de LSE-3. 
I can connect to the Logger using my own local network.

So, here it begins.

I use Port 8899
The IP is correct.

If you look at the screenshot, some config files do something, most do nothing.
Have a look at the screenshot.  None of the devices or entities give any information by the way.

What can I do to troubleshoot this situation?

Thanks all!

[image]

wills106, 2023-07-23:
Are you getting any errors in the error log? (Using plugin_sofar)

sanderbaks, 2023-07-23:
Yes, there is quite a few :). 

I will install de update first, and get back with the faultcodes.

sanderbaks, 2023-07-23:
> Are you getting any errors in the error log? (Using plugin_sofar)

This is the errors:
2023-07-23 16:59:01.243 WARNING (MainThread) [custom_components.solax_modbus.plugin_sofar] Sofar: reading serial number from address 0x445 failed; other address may succeed
2023-07-23 16:59:01.256 ERROR (MainThread) [custom_components.solax_modbus.plugin_sofar] Sofar: cannot find serial number, even not for other Inverter
2023-07-23 16:59:01.259 ERROR (MainThread) [custom_components.solax_modbus.plugin_sofar] unrecognized Sofar inverter type - serial number : unknown

I am using the solarman LSE-3 dongle. The specifications say it does transmit Modbus data.

Thanks for your help!!

wills106, 2023-07-23:
You don't have working communications then. Do you get the same error with plugin_sofar_old or does it return your serial number?

sanderbaks, 2023-07-23:
No, the serial number is not received.  I am also trying the Solarman plugin. that is also not working.

However, I do have a connection using the IP address. So the connection is physically working.

Is there a way to check if the modbus is sending any information at all?

wills106, 2023-07-23:
I am pretty sure you should be using plugin_sofar but I wanted to double check it wasn't using the old register set.

The Inverter won't send data by itself. It need to be queried.
Is there any option to turn on data logging either on the Inverter LCD or in a web portal on the stick?

sanderbaks, 2023-07-23:
Hi,

No, no particular option to turn it off or on.. the Solarman app is working normally

Thanks!

Sander

sanderbaks, 2023-07-23:
I am currently using the application "oModbusMaster" to manually try to read the data that is coming from the data logger.
It is really strange why all of the tools will not work.
I will post the results here.

thanks!

sanderbaks, 2023-07-24:
So, Looking at register "0x000A" or 10 in Decimals, I get a value back.
According to the yaml config I found somewhere it should be PV1 power. Not sure if this is the case.

But any how, I am communicating with the logger using Modbus RTU over TCP.

Is there anything I can do from here? Is there a register where we can predict the value to check to registers?

[image]

wills106, 2023-07-24:
Have you tried setting the Integration up with Modbus RTU over TCP instead of the default Modbus TCP?

[image]

sanderbaks, 2023-07-24:
Yes for sure I did :)

[image]

wills106, 2023-07-24:
Just checked the docs if you are reading 0x0A for PV Power you are using the old register set then (plugin_sofar_old)
Have you tried using that plugin setup with Modbus RTU over TCP?

sanderbaks, 2023-07-24:
Oh, I am sorry for the confusion. I don’t really know what parameters I am checking. I was just confirming that I am getting at least some data back.

Is there a register you are aware of the should produce a fixed and constant value? Like the serial number? This would be a register for me to focus on in order the establish a good connection..

My logger serial number is 2104857593
FW ME_0C_270A_1.05

Thanks.

Sander

wills106, 2023-07-24:
New register set:
https://github.com/wills106/homeassistant-solax-modbus/blob/22d790289699c8f1f3073275c846d792085296b5/custom_components/solax_modbus/plugin_sofar.py#L2347-L2350

Old registerset:
https://github.com/wills106/homeassistant-solax-modbus/blob/22d790289699c8f1f3073275c846d792085296b5/custom_components/solax_modbus/plugin_sofar_old.py#L1062-L1065

sanderbaks, 2023-07-24:
So, should these registers return the full serialnumber in one return? does that mean the register is ASCII? or an interger?

sanderbaks, 2023-07-24:
Ow man I see nothing that resembles any kind of Serialnumber.

wills106, 2023-07-24:
Can you read the grid frequency with 0x484 or 0xE it will be returned as 4989 to be 49.89Hz for example.

Another one to try is the Inverter state

New register:
https://github.com/wills106/homeassistant-solax-modbus/blob/22d790289699c8f1f3073275c846d792085296b5/custom_components/solax_modbus/plugin_sofar.py#L467-L481

Old register:
https://github.com/wills106/homeassistant-solax-modbus/blob/22d790289699c8f1f3073275c846d792085296b5/custom_components/solax_modbus/plugin_sofar_old.py#L107-L118

sanderbaks, 2023-07-24:
I sent a mail to Solarman (IGEN) to ask for maybe a new firmware.
Also I asked if they have any ModBus documentation

It is driving me nuts.  I can not get a respons that I can identify…
There is some form of communication. But I don’t have the knowledge to figure it out, yet.

wills106, 2023-07-24:
Maybe the data stick only talks externally in Solarman?
Do you have access to a cheap RS485 device? Even a RS485 - USB adaptor so you could talk directly to the Inverter? Then you could look into a Waveshare or USR device.

sanderbaks, 2023-07-24:
The inverter is quite far away (50 meter) and connected with utp. I need to create a quick temp solution. I will look into it----------------Sander BaksCommercial & editorial photographerwww.sanderbaks.comwww.hotelcreatives.comT: 06-46188558