wills106/homeassistant-solax-modbus #462: [BUG]

mikaabra, 2023-07-09:
**Describe the bug**
A clear and concise description of what the bug is.

Connecting to the device seems to work, but I get no relevant entities, only "language" and "rtc" entities show up.

**Mandatory details**
1. Integration Version: 2023.06.08
2. HA Version: 2023.6.3
3. Inverter Brand: Growatt MOD 15KTL3-X
4. Plugin Used: growatt
5. First 6 digits of Serial: RPHECM
6. Connection Method: modbus tcp

**Detailed Error Log**
```
2023-07-09 12:53:12.494 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt4 inverter type - serial number : DL1.0 Zxxx
2023-07-09 12:53:48.717 DEBUG (MainThread) [custom_components.solax_modbus] setup entries - data: {}, options: {'name': 'growatt4', 'interface': 'tcp', 'read_modbus_addr': 1, 'plugin': 'growatt', 'scan_interval': 15, 'read_eps': False, 'read_dcb': False, 'read_pm': False, 'host': '192.168.200.230', 'port': 502, 'tcp_type': 'tcp'}
2023-07-09 12:53:48.717 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: growatt
2023-07-09 12:53:48.717 DEBUG (MainThread) [custom_components.solax_modbus] Setup solax_modbus.growatt4
2023-07-09 12:53:48.718 DEBUG (MainThread) [custom_components.solax_modbus] solax serial port /dev/ttyUSB0 interface tcp
2023-07-09 12:53:48.718 DEBUG (MainThread) [custom_components.solax_modbus] solax modbushub creation with interface tcp baudrate (only for serial): 19200
2023-07-09 12:53:48.718 DEBUG (MainThread) [custom_components.solax_modbus] growatt4: ready to call plugin to determine inverter type
2023-07-09 12:53:48.718 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] growatt4: trying to determine inverter type
2023-07-09 12:53:48.765 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Read growatt4 0x9 firmware number before potential swap: DL1.0 Zxxx
2023-07-09 12:53:48.766 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt4 inverter type - serial number : DL1.0 Zxxx
2023-07-09 12:53:48.770 DEBUG (MainThread) [custom_components.solax_modbus] solax modbushub done {'_hass': <homeassistant.core.HomeAssistant object at 0x7fb82c0610>, '_client': <ModbusTcpClient at 0x7f7a1f2810 socket=<socket.socket fd=61, family=2, type=1, proto=6, laddr=('192.168.200.229', 42056), raddr=('192.168.200.230', 502)>, ipaddr=192.168.200.230, port=502, timeout=5.0>, '_lock': <unlocked _thread.lock object at 0x7f5e3668c0>, '_name': 'growatt4', '_modbus_addr': 1, '_seriesnumber': 'DL1.0\x00ZBAA\x00\x13', 'interface': 'tcp', 'read_serial_port': '/dev/ttyUSB0', '_baudrate': 19200, '_scan_interval': datetime.timedelta(seconds=15), '_unsub_interval_method': None, '_sensor_callbacks': [], 'data': {'_repeatUntil': {}}, 'tmpdata': {}, 'tmpdata_expiry': {}, 'cyclecount': 0, 'slowdown': 1, 'inputBlocks': {}, 'holdingBlocks': {}, 'computedSensors': {}, 'computedButtons': {}, 'sensorEntities': {}, 'numberEntities': {}, 'writeLocals': {}, 'sleepzero': [], 'sleepnone': [], 'writequeue': {}, 'plugin': growatt_plugin(plugin_name='growatt', SENSOR_TYPES=[GrowattModbusSensorEntityDescription(key='firmware_version', device_class=None, 
```

**Additional context**
Add any other context about the problem here.

Using this configuration directly in configuration.yaml works properly but I'd rather try to spend time to get this integration working rather than doing all the work just for myself which doesn't help others.

```
  - name: "growatt-tcp"
    type: tcp
    host: 192.168.200.230
    port: 502
    sensors:
      - name: "growatt-tcp power"
        unit_of_measurement: W
        device_class: power
        slave: 1
        address: 1
        input_type: input
        scan_interval: 15
        data_type: int32
        count: 2
        scale: 0.1
```

wills106, 2023-07-19:
I need to add your Firmware version into the Integration. (Growatt has a true unique serial so I can't use that)
Is your Inverter a solar PV only Inverter and not a Hybrid model?

wills106, 2023-07-19:
I have just released [2023.07.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.1) on the Growatt plugin I haven't fully setup PV Only Inverters, so it's defined as a Hybrid for now.

Could you let me know which entities return values and which don't.

I can then configure the plugin correctly to only return values that are relevant to PV Inverters.

mikaabra, 2023-07-19:
MOD 15KTL3-X is a purely PV inverter without any battery support.

I'll get back to you regarding the entities!

wills106, 2023-07-29:
Just wondering if you have managed to look into which registers / sensors work and don't work on PV only Inverters?
Then I can define the PV / Hybrids correctly.

mikaabra, 2023-07-30:
Hi, sorry, I was travelling so couldn't test this. I now did, and all values are zero. I'm though wondering if there's some old "crud" left from before in the database, considering it's still using "growatt4" even though I changed the prefix?

I get these error messages:

```
growatt4: read failed at 0xc05: total_yield
10:17:53 – (WARNING) SolaX Inverter Modbus (custom integration) - message first occurred at 10:11:40 and shows up 12 times
growatt4: read failed at 0xc05: total_yield
10:17:33 – (WARNING) SolaX Inverter Modbus (custom integration) - message first occurred at 10:10:40 and shows up 46 times
Platform solax_modbus does not generate unique IDs. ID growatt4_grid_voltage_l3 already exists - ignoring sensor.growatt4_grid_voltage_l3
10:16:52 – (ERROR) Sensor - message first occurred at 10:10:24 and shows up 6 times
input register already declared: 0xbd3 grid_current_l1
10:16:52 – (WARNING) SolaX Inverter Modbus (custom integration) - message first occurred at 10:10:24 and shows up 12 times
```

I do get 55 entities. This is for Modbus over TCP (my RS485 TCP gateway converts from Modbus RTU RS485 to Modbus TCP).

```
Sensors
Battery Charge Max Power
0.0 W
Battery Charge Power
0.0 W
Battery Current
0.0 A
Battery Discharge Max Power
0.0 W
Battery Discharge Power
0.0 W
Battery SOC
0%
Battery Voltage
0.0 V
Charger Switch
Disabled
Charger Upper SOC
0%
Grid Current
Unknown
Grid Current L1
0.0 A
Grid Current L2
0.0 A
Grid Current L3
0.0 A
Grid Frequency
0.0 Hz
Grid Power L1
0 VA
Grid Power L2
0 VA
Grid Power L3
0 VA
Grid Power VA
Unknown
Grid Voltage L1
0.0 V
Grid Voltage L2
0.0 V
Grid Voltage L3
0.0 V
Language
English
PV Current 1
0.0 A
PV Current 2
0.0 A
PV Current 3
0.0 A
PV Power 1
0.0 W
PV Power 2
0.0 W
PV Power 3
0.0 W
PV Voltage 1
0.0 V
PV Voltage 2
0.0 V
PV Voltage 3
0.0 V
Today's Grid Export
0.0 kWh
Today's Grid Import
0.0 kWh
Today's PV1 Solar Energy
0.0 kWh
Today's PV2 Solar Energy
0.0 kWh
Today's PV3 Solar Energy
0.0 kWh
Today's Solar Energy
0.0 kWh
Today's Yield
0.0 kWh
Today'sPower Generation
0.0 kWh
Total Grid Export
0.0 kWh
Total Grid Import
0.0 kWh
Total Grid Power
0 W
Total Power Generation
0.0 kWh
Total PV Power
0.0 W
Total PV1 Solar Energy
0.0 kWh
Total PV2 Solar Energy
0.0 kWh
Total PV3 Solar Energy
0.0 kWh
Total Solar Energy
0.0 kWh
Total Yield
0.0 kWh
```

mikaabra, 2023-07-30:
Guess that wasn't the problem, I deleted my inverter in SolaX, restarted HA, added again, now it picked up the new prefix. All values are still zero though. Is it using the correct API version? In "Growatt Local" I had to choose the API version correctly for it to work.

```
growatt99: read failed at 0xc03: today_yield
12:59:46 – (WARNING) SolaX Inverter Modbus (custom integration) - message first occurred at 12:57:36 and shows up 28 times
growatt99: read failed at 0xc03: today_yield
12:57:26 – (WARNING) SolaX Inverter Modbus (custom integration) - message first occurred at 12:56:56 and shows up 8 times
Platform solax_modbus does not generate unique IDs. ID growatt99_grid_voltage_l3 already exists - ignoring sensor.growatt99_grid_voltage_l3
12:56:45 – (ERROR) Sensor
input register already declared: 0xbd3 grid_current_l1
12:56:45 – (WARNING) SolaX Inverter Modbus (custom integration) - message first occurred at 12:56:45 and shows up 2 times
```

Here is some config I made locally that works correctly, if it can be some hint.

```
  - name: "growatt-tcp"
    type: tcp
    host: 192.168.200.230
    port: 502
    sensors:
      - name: "growatt-tcp power"
        unit_of_measurement: W
        device_class: power
        slave: 1
        address: 1
        input_type: input
        scan_interval: 10
        data_type: int32
        count: 2
        scale: 0.1
        unique_id: growatt-tcp-power1
      - name: "growatt-tcp power output"
        unit_of_measurement: W
        device_class: power
        slave: 1
        address: 35
        input_type: input
        scan_interval: 10
        data_type: int32
        count: 2
        scale: 0.1
        unique_id: growatt-tcp-power-output1
      - name: "growatt-tcp grid frequency"
        unit_of_measurement: Hz
        device_class: frequency
        slave: 1
        address: 37
        input_type: input
        scan_interval: 10
        data_type: int16
        scale: 0.01
        precision: 2
        unique_id: growatt-tcp-grid-frequency1
      - name: "growatt-tcp temperature"
        unit_of_measurement: "°C"
        device_class: temperature
        slave: 1
        address: 93
        input_type: input
        scan_interval: 10
        data_type: int16
        precision: 1
        scale: 0.1
        unique_id: growatt-tcp-temp1
      - name: "growatt-tcp temperature IPM"
        unit_of_measurement: "°C"
        device_class: temperature
        slave: 1
        address: 94
        input_type: input
        scan_interval: 10
        data_type: int16
        precision: 1
        scale: 0.1
        unique_id: growatt-tcp-temp-ipm
      - name: "growatt-tcp temperature boost"
        unit_of_measurement: "°C"
        device_class: temperature
        slave: 1
        address: 95
        input_type: input
        scan_interval: 10
        data_type: int16
        precision: 1
        scale: 0.1
        unique_id: growatt-tcp-temp-boost1
      - name: "growatt-tcp power output PAC1"
        unit_of_measurement: W
        device_class: power
        slave: 1
        address: 40
        input_type: input
        scan_interval: 10
        data_type: int32
        count: 2
        scale: 0.1  
        unique_id: growatt-tcp-output-pac1
      - name: "growatt-tcp power output PAC2"
        unit_of_measurement: W
        device_class: power
        slave: 1
        address: 44
        input_type: input
        scan_interval: 10
        data_type: int32
        count: 2
        scale: 0.1  
        unique_id: growatt-tcp-output-pac2
      - name: "growatt-tcp power output PAC3"
        unit_of_measurement: W
        device_class: power
        slave: 1
        address: 48
        input_type: input
        scan_interval: 10
        data_type: int32
        count: 2
        scale: 0.1
        unique_id: growatt-tcp-output-pac3
      - name: "growatt-tcp current output PAC1"
        unit_of_measurement: A
        device_class: current
        slave: 1
        address: 39
        input_type: input
        scan_interval: 10
        data_type: int16
        precision: 1
        scale: 0.1
        unique_id: growatt-tcp-output-a-pac1
      - name: "growatt-tcp current output PAC2"
        unit_of_measurement: A
        device_class: current
        slave: 1
        address: 43
        input_type: input
        scan_interval: 10
        data_type: int16
        precision: 1
        scale: 0.1
        unique_id: growatt-tcp-output-a-pac2
      - name: "growatt-tcp current output PAC3"
        unit_of_measurement: A
        device_class: current
        slave: 1
        address: 47
        input_type: input
        scan_interval: 10
        data_type: int16
        precision: 1
        scale: 0.1
        unique_id: growatt-tcp-output-a-pac3
      - name: "growatt-tcp derating mode"
        slave: 1
        address: 104
        input_type: input
        scan_interval: 10
        data_type: int16
        unique_id: growatt-tcp-derating-mode
      - name: "growatt-tcp real power percent"
        unit_of_measurement: "%"
        slave: 1
        address: 113
        input_type: input
        scan_interval: 10
        data_type: int16
        unique_id: growatt-tcp-real-power-percent
```

wills106, 2023-07-30:
Ok it looks like this Inverter is using the alternative register set.

I have just released [2023.07.4b4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.4b4) you are probably best removing the Integration and restarting HA before adding it back in. Otherwise you will probably get a load of orphaned sensors.

mikaabra, 2023-07-30:
That worked much better! Only thing I immediately noticed is that there's no "grid voltage l3" and that other values might be missing, but the values shown seem to be correct.

```
Sensors
Battery Minimum Capacity
0%
Grid Current L1
4.7 A
Grid Current L2
4.7 A
Grid Current L3
4.7 A
Grid Frequency
50.01 Hz
Grid Power L1
1,083.3 VA
Grid Power L2
1,087.5 VA
Grid Power L3
1,086.1 VA
Grid Voltage L1
230.5 V
Grid Voltage L2
231.4 V
Language
English

Battery Type
Lithium
Boost Temperature
63.7 °C
Firmware Control Version
ZBAA
Firmware Version
@DL1.
Inverter Temperature
73.7 °C
IPM Inverter Temperature
70.1 °C
RTC
Unknown
Serial Number
RPHxxxxxx
```

wills106, 2023-07-31:
> there's no "grid voltage l3"

It wasn't declared properly in the Integration.

I have just released [2023.07.4b5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.4b5) which redefines all Growatt Inverters. So a lot has changed.

wills106, 2023-07-31:
Can you try [2023.07.4b7](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.4b7)

mikaabra, 2023-07-31:
I deleted line 970 and re-tried before you sent that, and as far as I can see, all relevant values are correctly populated now.

Nice!

It assumes up to PV4, whereas I only have PV1-2, but I guess that's ok.

Thanks, great progress!

wills106, 2023-07-31:
PV3-4 should be disabled by default though.

I'm not sure yet how to detect if an Inverter has more than 2 PV channels. It's possibly off the Inverter subtype or capacity?

mikaabra, 2023-07-31:
Oh, there's one value that is wrong:

Grid Export Limit 6,553,600.0%

wills106, 2023-07-31:
Just released [2023.07.4b8](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.4b8) I was acidentally reading 'Grid Export Limit' as a 32bit register.

closed 2023-07-31 by wills106

wills106, 2023-07-31:
I didn't realise I had it on line 136 as well. I have deleted line 136 in [2023.07.4b9](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.4b9) as it now follows the standard sequence.

Although it achieves the same as what you have done locally.

mikaabra, 2023-08-02:
I updated to 2023.08.02 and now I get this error:

```
Logger: homeassistant.config_entries
Source: custom_components/solax_modbus/plugin_growatt.py:2810
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 10:40:23 (1 occurrences)
Last logged: 10:40:23

Error setting up entry growatt90 for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 390, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 109, in async_setup_entry
    plugin = importlib.import_module(f".plugin_{plugin_name}", 'custom_components.solax_modbus')
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/plugin_growatt.py", line 2810, in <module>
    allowedtypes = GEN43,
                   ^^^^^
NameError: name 'GEN43' is not defined
```

wills106, 2023-08-02:
Just corrected that in [2023.08.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.08.3)