wills106/homeassistant-solax-modbus #464: Integration failing to load with HA 2023.7.1 - cannot import name 'make_byte_string'

litinoveweedle, 2023-07-13:
**Describe the bug**
A clear and concise description of what the bug is.

**Mandatory details**
1. Integration Version: 2023.06.8
2. HA Version: 2023.7.1
3. Inverter Brand: Solax X3 G4 10kW
4. Plugin Used:
5. First 6 digits of Serial:
6. Connection Method: RS485 to Ethernet adapter

**Detailed Error Log**

Log Details (ERROR)
This error originated from a custom integration.
Logger: homeassistant.setup
Source: custom_components/solax_modbus/__init__.py:31
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 11:12:15 AM (1 occurrences)
Last logged: 11:12:15 AM

Setup failed for custom integration solax_modbus: Unable to import component: cannot import name 'make_byte_string' from 'pymodbus.utilities' (/usr/local/lib/python3.11/site-packages/pymodbus/utilities.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 215, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 813, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 31, in <module>
    from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder, Endian
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 13, in <module>
    from pymodbus.utilities import (
ImportError: cannot import name 'make_byte_string' from 'pymodbus.utilities' (/usr/local/lib/python3.11/site-packages/pymodbus/utilities.py)

**Additional context**

This happened after upgrade from 2023.6.3 to 2023.7.1, without any other changes. Google search for related issues suggest it might be related to concurent usage of the native HA modbus integration (for accessing other devices than Solax).

litinoveweedle, 2023-07-13:
I can confirm, that after revert to HA 2023.6.3 integration is working again.

litinoveweedle, 2023-07-14:
Also reference here https://github.com/home-assistant/core/issues/96521

ChristophCaina, 2023-07-14:
I had the same error log also for ZHA...
After completely rebooting the HOST, the errors were gone.

I am pretty sure that this issue isn't directly related to the 2023.7 or 2023.7.1 updates, because both integrations did run with this version - and even now with 2023.7.2 after the mentioned reboot.

So I guess, there's an issue with the bootstrap (import) that might hit under some (specific) conditions...

wills106, 2023-07-14:
I'm still on HA 2023.06.x at the moment.

Does this just happen on HAOS / Supervised or does it also happen on Core?

ChristophCaina, 2023-07-14:
I don't know.
I am running HA OS ... not sure about the other types

wills106, 2023-07-14:
Ok, I can check Core when I update mine later.

litinoveweedle, 2023-07-14:
Just some more info. I am running Supervised. Based on @ChristophCaina sugestion I decided to upgrade again and try to reboot if needed. Now latest version is HA 2023.7.2. After upgrade Solax integration is running and no machine reboot was even required. So behavior seems to be random, but was first observed in 2023.7.1

I am afraid that issue is still present, only it occurs randomly. The common denominator seems to be pymodbus and new HA 2023.7.X. Isn't there any version conflict between HA and integration requirements caused by new HA version? Maybe it depends on the order of the integration and modules being loaded?

ChristophCaina, 2023-07-17:
Hi, if the solax modbus is a custom integration, you might need to raise a ticket with them, since the pymodbus library was updated with the latest ha updates.

The devs of the integration might need to update their dependencies accordingly.

ChristophCaina, 2023-07-17:
sorry - haven't seen that this is the solax-modbus repro :D

litinoveweedle, 2023-07-22:
Ok, today I updated first Solax integration using HACS to 2023.07.1 and then I started automated HA update to 2023.7.2, which finished with HA restart. After HA start I got same error and my inverter was not monitored by HA.

So the root cause is still present, it really seems like some clash between pymodbus libraries usage.

```
Logger: homeassistant.setup
Source: custom_components/solax_modbus/__init__.py:31
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 2:17:54 PM (1 occurrences)
Last logged: 2:17:54 PM

Setup failed for custom integration solax_modbus: Unable to import component: cannot import name 'make_byte_string' from 'pymodbus.utilities' (/usr/local/lib/python3.11/site-packages/pymodbus/utilities.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 215, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 813, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 31, in <module>
    from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder, Endian
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 13, in <module>
    from pymodbus.utilities import (
ImportError: cannot import name 'make_byte_string' from 'pymodbus.utilities' (/usr/local/lib/python3.11/site-packages/pymodbus/utilities.py)
```

wills106, 2023-07-23:
I have just updated to HA 2023.7.3 and I don't get this issue. (pymodbus 3.1.3)
I have manually updated pymodbus to 3.3.1 and I still don't get this issue.

Where people are getting this issue are you using the built in modbus component to connect to another device?

wills106, 2023-07-23:
Just released [2023.07.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.2) which matches pymodbus to version in HA.

litinoveweedle, 2023-07-23:
I just updated Solax to 2023.07.3 and no issue was observed after HA restart. I will monitor this in the longer run as to me it only happened during HA upgrades. As I mentioned previously I do use default HA modbus integration to talk to other devices.

litinoveweedle, 2023-09-20:
I have new findings. After HA upgrade from 2023.8.3 to the 2023.9.2 solax-modbus version 2023.08.6 was not able to start at all, with following errors:

```
Logger: homeassistant.setup
Source: loader.py:826
First occurred: 5:38:48 PM (1 occurrences)
Last logged: 5:38:48 PM

Setup failed for custom integration solax_modbus: Unable to import component: Exception importing custom_components.solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 818, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 31, in <module>
    from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder, Endian
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
    class BinaryPayloadBuilder:
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
    self, payload=None, byteorder=Endian.Little, wordorder=Endian.Big, repack=False
                                  ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Little

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 215, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 826, in get_component
    raise ImportError(f"Exception importing {self.pkg_path}") from err
ImportError: Exception importing custom_components.solax_modbus

Logger: homeassistant.loader
Source: custom_components/solax_modbus/__init__.py:31
Integration: solax_modbus (documentation, issues)
First occurred: 5:38:48 PM (1 occurrences)
Last logged: 5:38:48 PM

Unexpected exception importing component custom_components.solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 818, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 31, in <module>
    from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder, Endian
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
    class BinaryPayloadBuilder:
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
    self, payload=None, byteorder=Endian.Little, wordorder=Endian.Big, repack=False
                                  ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Little
```

I tried to restart HA several times, but solax-modbus was failing to load. Then I checked for solax-modbus updates - 2023.09.4 brought same version of pymodbus as HA 2023.9.2. Only after updating solax-modbus to the 2023.9.4 version, which uses same version of pymodbus library as HA integration was able to start.

So there is obviously hard lock of the pymodbus library version. If you use HA native modbus integration, (so pymodbus is delivered and loaded by HA core dependencies) than solax-modbus delivered pymodbus library has to be SAME version. Otherwise solax-modbus integration is not able to load. I have no knowledge how python dependencies are resolved during HA start, but there is obviously something fundamentally wrong either on HA site or in solax integration.

wills106, 2023-09-20:
The [Readme](https://github.com/wills106/homeassistant-solax-modbus#universal-solar-inverter-over-modbus-rs485--tcp-custom_component-for-home-assistant) and release notes for [2023.09.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.09.4) explain this. You can only have one version of pyModbus installed.

It was getting too complicated trying to support 3 very different branches of pyModbus

wills106, 2023-09-22:
Is the Initial issue still an issue or has this been resolved and can be closed down now?

wills106, 2023-09-24:
Closing down as no further response. If there are still issues either reopen this or start a new issue if different to original issue.

closed 2023-09-24 by wills106