wills106/homeassistant-solax-modbus #465: [BUG] Missing attribute 'ignore_readerror'

middelink, 2023-07-13:
**Describe the bug**
I'm using a somewhat flaky modbus ethernet converter and while it works 90% of the time, sometimes it drops the connection. At that point I see the HA log show an error about a missing attribute  'ignore_readerror'.

**Mandatory details**
1. Integration Version: 2023.06.8
2. HA Version: 2023.7.1
3. Inverter Brand: Solax
4. Plugin Used: HACS?
5. First 6 digits of Serial: H34A05
6. Connection Method: TCP

**Detailed Error Log**

```
Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:516
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 10:08:39 (13 occurrences)
Last logged: 11:45:09

Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 428, in read_modbus_data
    res = self.read_modbus_registers_all()
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 525, in read_modbus_registers_all
    res = res and self.read_modbus_block(block, 'holding')
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 516, in read_modbus_block
    if ((descr.ignore_readerror != True) and (descr.ignore_readerror !=False)) : self.data[descr.key] = descr.ignore_readerror # return something static
        ^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'dict' object has no attribute 'ignore_readerror'
```

**Additional context**
Add any other context about the problem here.

infradom, 2023-07-13:
Strange, I cannot directly see a reason for such behavior, the block structures are created at startup, so I will do some thinking  .... This error occurs indeed in case a block read fails, a piece of code that is seldom reached ...
Give me some time ...

infradom, 2023-07-13:
I think I found the error... Need some time to come up with a solution ...

infradom, 2023-07-14:
Yes I know it is a very confusing piece of code. In my development fork, I just added a if statement:
```
                     if not (type(descr) is dict):
                        if ((descr.ignore_readerror != True) and (descr.ignore_readerror !=False)) : self.data[descr.key] = descr.ignore_readerror # return something static 
```

The type is only dict in case a modbus 16bit register contains 2 independent byte values. With this additional if, we ignore this case and the ignore_readerrror attribute will not work for those exceptional cases/

wills106, 2023-09-23:
This has been fixed now and can be closed down can't it?

middelink, 2023-10-12:
sorry, yes. I don't see the log lines anymore.
Many thanks!

(Then again, I also bought some USR-N510 modbus<>ethernet bridges, so I cannot state with 100% certainty the bug is fixed).

closed 2023-10-12 by middelink