wills106/homeassistant-solax-modbus #468: Unable to set export control value to 300000w

pezlin, 2023-07-15:
In the solax app and cloud I can set the export control to 300000w. In HA I can only set it 12000w.

**Mandatory details**
1. Integration Version: 2023.06.8
2. HA Version: 2023.7.1
3. Inverter Brand: Solax X3 Hybrid G4 8Kw
4. Plugin Used:
5. First 6 digits of Serial: H34A08
6. Connection Method: Pocket Wifi

**Detailed Error Log**
I try to set the value in an automation and only sees an error that max value is 12000.

wills106, 2023-07-15:
Have you read the [Wiki?](https://github.com/wills106/homeassistant-solax-modbus/wiki/SolaX-Gen4-Modes-of-Operation#export-control-user-limit)

pezlin, 2023-07-16:
Yes, but I can't figure out if that is applicable on this. Do you mean that I should use "Config Max Export"? If so, how do I do that?

The reason for asking is that I want to limit the export when electricity prices are negative and export as much as possible when prices are positive.

pezlin, 2023-07-24:
@wills106 Would you please be able to instruct me how to set Max Config Level?

pezlin, 2023-08-06:
@wills106 Could you please point me in the right direction?

wills106, 2023-08-14:
Did I answer this on FB with:

> Go to the SolaX Integration page.
> Under Diagnostic group do you see "+xy entities not shown"
> Click that, then find "Config Max Export"
> Then Cog/Gear icon.
> There is an option to enable it and press "UPDATE"

wills106, 2023-08-16:
Closing down as steps listed to turn on disabled entities are now in the readme.

closed 2023-08-16 by wills106

insipiens, 2023-08-22:
@wills106 sort of surprising Solax allows a value well in excess of the data sheet, presumably a catch all parameter for all inverters they make and limited by the specific circuitry

Anyway thanks to @pezlin as I also missed the wiki article and would have posted a futile and annoying issue, but saw this :dancers: 

Great integration! - Kofi contribution made