wills106/homeassistant-solax-modbus #471: Improve data for "Growatt SPH 10000TL3 BH-UP" Inverter

choenig, 2023-07-21:
Hi,

I don't know if this should rather be a discussion or an issue, I just post it here as I guess there are more people reading here :)

I have successfully integrated my Growatt SPH 10000TL3 BH-UP Inverter.

The serial starts with 'YA1' so i added
```
elif seriesnumber.startswith('YA1'):  invertertype = HYBRID | GEN | X3 # Hybrid SPH 4kW - 10kW 3P TL3 BH-UP
```
to get it running.

Now I already improved some of the configuration values in `plugin_growatt.py` to enhance descriptions, scales and I also added some more registers. I also reduced the `block_size` to 64, that fixed some of the received registers.

In any case I'm willing to share my improvements, but it would be really nice to find other users of the SPH TL3 BH-UP inverters to check some values, that look really broken on my setup.

Br
Christian

wills106, 2023-07-21:
Feel free to submit any improvements either through comment's or PR's.

Unfortunately the Growatt Inverters are a little unloved at the moment as there aren't many users.
Also the growatt_plugin hasn't been fully setup for PV only Inverters which I need to work on.

Hans-chr2, 2023-07-23:
Hi Christian

What changes did you make ? 
last i tried the mapping for the TL3-BH-UP were wrong.

I have a same inverter as you, just 8kW model, if you have done some mapping i could verify some of them on mine.

wills106, 2023-07-23:
Just released [2023.07.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.2) which has Firmware starting 'YA1'

Hans-chr2, 2023-07-23:
I finds the inverter now, nut the mapping seems to be off. 
[image]

wills106, 2023-07-23:
This is the first time a 3Phase Inverter using this set of registers has been used on the Integration.
I'll need to go back over the documents.

Hans-chr2, 2023-07-23:
I don't know if this helps ? 

 sensors:

  - name: "Fault code"
    unique_id: '0a123585'
    scan_interval: 15
    slave: 1
    address: 0
    data_type: int16
    scale: 1
    precision: 1
    input_type: input
    device_class: monetary

# 1 = checking
# 5 = ?
# 6 = ? OK 
# 9 = Battery open 

#GENEREL INFO
  - name: "Inverter_effekt_sum"
    unique_id: '0a1235852'
    scan_interval: 5
    slave: 1
    address: 35
    state_class: measurement
    unit_of_measurement: "W"
    data_type: int32
    scale: 0.1
    precision: 1
    input_type: input
    device_class: power

  - name: "Inverter_frekvens"
    unique_id: '0a'
    scan_interval: 7
    slave: 1
    address: 37
    state_class: measurement
    unit_of_measurement: "Hz"
    data_type: int16
    scale: 0.01
    precision: 1
    input_type: input
    device_class: frequency

  - name: "Inverter_temperatur"
    unique_id: '0a695'
    scan_interval: 5
    slave: 1
    address: 93
    state_class: measurement
    unit_of_measurement: "°C"
    data_type: int16
    scale: 0.1
    precision: 1
    input_type: input
    device_class: temperature

#inverter total energi idag
  - name: "Inverter_energi_idag"
    unique_id: '0a12358526541'
    scan_interval: 60
    slave: 1
    address: 53
    state_class: total
    unit_of_measurement: "kWh"
    data_type: int32
    scale: 0.1
[... 731 lines cut ...]
        verify:
      - name: Status
        address: 1105
        command_on: 1
        command_off: 0

      - name: bat_3_tidslot
        unique_id: '0a135fhdsg3'
        address: 1108
        write_type: holding
        verify:
      - name: Status
        address: 1108
        command_on: 1
        command_off: 0

      - name: Export_limit
        unique_id: 'cfe6e6d3-0e3a-40b7-980d-e266ca3a29ac'
        address: 122
        write_type: holding
        verify:
      - name: Status
        address: 122
        command_on: 1
        command_off: 0

Hans-chr2, 2023-07-23:
I don't know if this helps

https://pastebin.com/XP9kmvRA

I have the newest modbus manual someware for it. 
https://www.scribd.com/document/592776725/Growatt-Inverter-Modbus-RTU-Protocol-II-V1-24-English-new

choenig, 2023-07-23:
Hi, juchu a buddy :)

I just created a WIP-pullrequest https://github.com/wills106/homeassistant-solax-modbus/pull/473 that shows what I changed.

@Hans-chr2 Some of your above errors will get fixed by my changes (e.g. the last two values). Those get fixed by reducing the block_size to 64, don't know why :)

Values vor L2 and L3 look broken here, too, but L1 is mostly correct (just as if it was just a 1 phase inverter)

choenig, 2023-07-23:
This is my state:

[image]
[image]
[image]
[image]
[image]

choenig, 2023-07-25:
Hi, 

I just updated the PR.

I switched back to 100 regs/fetch as I found out, that the issue was just a duplicate reg in the configuration. That resulted in misalignments when assigning values (as it is handled as "set of byte values" then).

Now I don't have any errors in the log anymore during fetch :).

I added some more registers but they don't make too much sense to me.

@Hans-chr2 Did you have a chance to test my patch?

Br
Christian

Hans-chr2, 2023-07-25:
How do i test your patch easy ?

choenig, 2023-07-25:
This is the link to the file of the PR:

https://raw.githubusercontent.com/wills106/homeassistant-solax-modbus/32728ab222979ada4f832e99de326976d13c19e3/custom_components/solax_modbus/plugin_growatt.py

You can download it and replace `./custom_components/solax_modbus/plugin_growatt.py` with the downloaded one. Then you need to restart HA (quick restart is not enough). I'd suggest to create a copy of the original file before overwriting it ... just in case ;-)

The next official update (done via HACS) will overwrite the file again.

Hans-chr2, 2023-07-25:
Just a quick overlook, the following are off: 
[image]
[image]

Hans-chr2, 2023-07-25:
The power reading seems to be the only sensors that are off, but i just looked over it quickly. 
When i look at the code, you use: 
[image]

and i use: 
[image]

as stated in the PDF i'm pretty sure all power readings are int32 and there are high power [kW] reading and low [W]

wills106, 2023-07-26:
> I'm pretty sure all power readings are int32

That's correct if the register is bidirectional it needs to be an int so it can represent positive and negative values.

You will find that positive means import from the grid and negative is exporting to the grid. (Or the other way round)

@choenig once you are happy with the PR let me know and I can merge it.

Hans-chr2, 2023-07-26:
@wills106 i know, but as i see it the registers are U32 and if that is correct what ChatGPT states, then you will get some very low or very high power readings because the first bit is used to determine positive/negative. 

But i don't know how the change the int type from U32 til int32, i tried it briefly last night but no luck.

choenig, 2023-07-26:
Using `signed int`s here makes absolute sense, unfortunately the values are still off for me.

The raw data for 1031, 1033 and 1035 (House Load L1-L3) looks like this:

[image]

And here for 1023, 1025, 1027 and 1029 (AC Power to Grid L1-L3 and Total)

[image]

That does not look like `signed` ;)

Any other idea?

@wills106 Thanks, I think i'm trying to get these things sorted out, then I'll make it ready for you :)

Hans-chr2, 2023-07-26:
@choenig i can't find the modbus manual 1,24 from Growatt on my computer anymore. If you can find that manual you can see the data size. 
As i remember all power and energy readings are 32 bit, and i can see on your screen dump above you pull 16 bit.

choenig, 2023-07-26:
I have the manual here, but where does it say something about signed/unsigned? I only find the information if it is high 16bit and low 16bit like here:

[image]

Or am I missing someting?

> As i remember all power and energy readings are 32 bit, and i can see on your screen dump above you pull 16 bit. 

That's correct, but each 32bit field contains of two 16bit fields. 
E.g. for 1035 and 1036: If we combine 214 (two high bytes) and 226 (two low bytes) we get a strange high number. Or is there any advanced way to combine those two values?

> and i use:

[From the previous post]
Do you have a running application that displays those value correctly?

wills106, 2023-07-26:
If you use GModbus you have the option to read registers as 32bit either a normal number or as a float.

So you would only read:
1031 as "unit = REGISTER_U32," or "unit = REGISTER_S32,"
1033 as "unit = REGISTER_U32," or "unit = REGISTER_S32,"
1035 as "unit = REGISTER_U32," or "unit = REGISTER_S32,"

Depending on which method you need. Pymodbus will read the inbetween registers as part of each 32bit read (reads two 16bit registers to create a 32bit one)

Hans-chr2, 2023-07-27:
When i use the modbus integration in Home Assistant via the config files i only use the int32 bit, the power values can both be negative and positive.  i also don't read multiple registers only 1 addressee at the time. 

I think we either over complicates things, or there is something i have done that is stupidly lucky in the config. :-D

[image]

Hans-chr2, 2023-07-27:
Im out of options now... 

[image]

wills106, 2023-07-27:
> Im out of options now...

Does House Load L1 & L2 seem correct?
It might be possible there is a Firmware fault with L3?

choenig, 2023-07-27:
`L1` shows the same as `Total` for me, so I guess `L2` and `L3` are both broken.

Hans-chr2, 2023-07-27:
[image]
[image]

Regarding L1+L2+L3 i also have the same issue with my generic modbus, so i see the same as @choenig  now.

choenig, 2023-07-28:
I marked the PR as `ready`, as it does not look that we have something new here, now. 

Let's then see what the holding-registers-side has for us :)

wills106, 2023-07-29:
Just released [2023.07.4b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.4b2) with the changes.

wills106, 2023-07-31:
I have just released [2023.07.4b5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.4b5) which redefines all Growatt Inverters. So a lot has changed.

choenig, 2023-07-31:
I just updated to `b7` and I have not found any issues, yet :)

wills106, 2023-07-31:
There is an error but I have just released [2023.07.4b8](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.4b8) I was accidentally reading 'Grid Export Limit' as a 32bit register.

wills106, 2023-07-31:
Corrected another error in [2023.07.4b9](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.07.4b9)

wills106, 2023-08-01:
Does anyone have a 2nd battery system connected to your Inverter?

wills106, 2023-08-01:
Just released [2023.08.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.08.2 which fixes an issue with TL-HX / TL3-HX and adds additional sensor entities for TL-HX / TL3-HX Inverters.

wills106, 2023-09-03:
Can this be closed down now? Or is further work required?

wills106, 2023-09-24:
Closing down as no further response.

closed 2023-09-24 by wills106

igorzavalyshyn-centrica, 2024-04-12:
Just a small comment for Growatt inverters' registers. 
According to 1.24v of official Modbus specs, all registers are U16 (unsigned int16) which means we cannot possibly tell if a given power value is negative or positive. So there is no point in using S32 type for power values instead of U32 because you won't ever get a sign bit.
[image]