wills106/homeassistant-solax-modbus #476: [BUG]

wartungsmonteur, 2023-07-26:
**Describe the bug**
A clear and concise description of what the bug is.

**Mandatory details**
1. Integration Version:
2. HA Version:
3. Inverter Brand:
4. Plugin Used:
5. First 6 digits of Serial:
6. Connection Method:

**Detailed Error Log**
Detailed Error Log, if you press on the individual error further information is displayed.
This is the content we need, the description alone doesn't provide enough context.

If there are no log's showing for SolaX try the following depending on your language settings, you can find the full logs under:

Settings → System → Logs > at bottom of page press “LOAD FULL LOGS”

Now the full logs are loaded. If you scroll down, you will see them.
Once the full logs are shown, you can either use the search function in your browser to search for “solax” related entries or use the search entry field on top of the page.

Search for

solax

and report us the logs. Make sure to replace sensitive information by xxxx (if any)

**Additional context**
Add any other context about the problem here.

wartungsmonteur, 2023-07-26:
Plugin startet nicht, keineVerbindung zum Modbus, 
Anbei ein Logfile

[home-assistant_solax_modbus_2023-07-26T00-32-36.554Z.log](https://github.com/wills106/homeassistant-solax-modbus/files/12166703/home-assistant_solax_modbus_2023-07-26T00-32-36.554Z.log)

wartungsmonteur, 2023-07-26:
[home-assistant_2023-07-26T00-44-33.604Z.log](https://github.com/wills106/homeassistant-solax-modbus/files/12166736/home-assistant_2023-07-26T00-44-33.604Z.log)

wills106, 2023-07-26:
Are you sure that's the correct IP address of the Waveshare?
Most home networks default to 192.168.0.x or 192.168.1.x
Can you ping that address and get a response ok?

wartungsmonteur, 2023-07-26:
Hallo

IP ist korrekt, der waveshare ist unter 192.168.178.107 erreichbar und auch konfigurierbar.

Ich bin am verzweifeln.

Grüße und vielen Dank für eure Hilfen.

wartungsmonteur, 2023-07-26:
[image]

wartungsmonteur, 2023-07-26:
Hier ein Screenshot

infradom, 2023-07-26:
> Hier ein Screenshot

Screenshot did not yet arrive ... Uploading .....

wartungsmonteur, 2023-07-26:
[image]

wartungsmonteur, 2023-07-26:
Es geht wieder, der WR müsste ein Update bekommen, 312 Entitäten sichtbar. Ihr seit phantastisch.. Kann ich euch was spenden und übernehmt ihr auch eventuell eine Fernwartung meines Home-Assistenten ?

Vielen lieben Dank

closed 2023-07-26 by wartungsmonteur

infradom, 2023-07-26:
One remark: in my opinion, the enable-multihost option is in most cases not needed . Screenshot from the wikil
[image]

infradom, 2023-07-26:
Do not forget to disable the DEBUG mode again, otherwise your system might be overwhelmed by messages ..

wartungsmonteur, 2023-07-26:
Alles ok, der neue WR musste ein  Update bekommen.

wartungsmonteur, 2023-07-26:
Hallo

Wenn ich die beiden Werte im waveshare umstelle geht es nicht mehr. Ich muss Enable Multi-Host auf yes setzen.

infradom, 2023-07-26:
Strange, for me, it has worked for years now without multi-host