wills106/homeassistant-solax-modbus #477: Sofar PV Inverter not reporting kWh totals

q1x, 2023-07-26:
A little more digging. I see the cli util supports 32 bit unsigned integer queries through this command:

```
q1x@laptop~ % modbus 192.168.x.2x:8899 1668/I 1670/I
Parsed 0 registers definitions from 1 files
1668: 2214 0x8a6
1670: 1505 0x5e1
```

Looking at your code in `plugin_sofar` from line 1909 , this would evaluate register `0x684` (Daily) to 22,14 kWh and register `0x686` (Total) to 150,5 kWh based on the `scale`parameter. This is inline with the stats in the webportal!

```
    SofarModbusSensorEntityDescription(
        name = "Solar Generation Today",
        key = "solar_generation_today",
        native_unit_of_measurement = UnitOfEnergy.KILO_WATT_HOUR,
        device_class = SensorDeviceClass.ENERGY,
        state_class = SensorStateClass.TOTAL_INCREASING,
        register = 0x684,
        newblock = True,
        unit = REGISTER_U32,
        scale = 0.01,
        rounding = 2,
        allowedtypes = HYBRID | PV,
    ),
    SofarModbusSensorEntityDescription(
        name = "Solar Generation Total",
        key = "solar_generation_total",
        native_unit_of_measurement = UnitOfEnergy.KILO_WATT_HOUR,
        device_class = SensorDeviceClass.ENERGY,
        state_class = SensorStateClass.TOTAL_INCREASING,
        register = 0x686,
        unit = REGISTER_U32,
        scale = 0.1,
        rounding = 1,
        allowedtypes = HYBRID | PV,
    ),
```

Can you help me figure out why the values are returning as 'unknown' in HASS? If needed, I can create a separate issue for this.

_Originally posted by @q1x in https://github.com/wills106/homeassistant-solax-modbus/discussions/187#discussioncomment-6500693_

q1x, 2023-07-26:
Screenshots
[image]
[image]

wills106, 2023-07-29:
If you change:
https://github.com/wills106/homeassistant-solax-modbus/blob/a9712803e5449199866f9ac6f8e522901f9fbeb5/custom_components/solax_modbus/plugin_sofar.py#L2419

to 

`order32 = Endian.Little,`

Do the sensors read ok?

Does that break any other sensor using `REGISTER_U32` or `REGISTER_S32` though?
Or are they starting to read correctly as well?

q1x, 2023-08-24:
Hi @wills106 ,

Sorry for the late reply, I've been away on vacation.
Just upgraded to the latest version and edited the plugin file to the entries below:

```
...
plugin_instance = sofar_plugin(
    plugin_name = 'sofar',
    SENSOR_TYPES = SENSOR_TYPES,
    NUMBER_TYPES = NUMBER_TYPES,
    BUTTON_TYPES = BUTTON_TYPES,
    SELECT_TYPES = SELECT_TYPES,
    block_size = 100,
    order16 = Endian.Big,
    order32 = Endian.Little,
    #order32 = Endian.Big,
    )
```

Unfortunately, all sensors seem to be reporting `unknown` now (also the ones that were working before).
Reverting the change reenables all the sensors (except for the totals).

cschlipf, 2024-02-19:
Works for me with the current version and a HYD10KTL:
[image]

Do you still see this issue with the latest version?

q1x, 2024-03-08:
Sorry, I haven't been able to test. The Modbus sensors work well for me and I don't want to mess that stuff up.
I'll assume it has been resolved or it was a case of PEBKAC and close the issue.

closed 2024-03-08 by q1x