wills106/homeassistant-solax-modbus #479: Core & SolaX Modebus issue

zlakes01, 2023-07-28:
It works but, I have found this message in Core logg:

2023-07-28 13:09:37.317 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 342: return self._client.read_input_registers(address, count, **kwargs)

Biside this, Solax-modebus doesn´t work after Core update to version core-2023.7.3.

Inverter Solax Hybrid X3 10 kW

alienatedsec, 2023-07-28:
Update to the latest beta as it will match the pymodbus version of the core. Check this https://github.com/wills106/homeassistant-solax-modbus/issues/464#issuecomment-1646819950

zlakes01, 2023-07-28:
Update solax-modebus or Core?

alienatedsec, 2023-07-28:
Solax-Modus, please.

wills106, 2023-07-28:
If you are not on Integration 2023.07.2 or newer like @alienatedsec mentions upgrade to that version or newer.
If you are still having issue did you perform a proper restart of HA following updating Core to 2023.7.3 ?

The blocking calls issue is listed [here](https://github.com/wills106/homeassistant-solax-modbus#known-issues) it's already an open issue as well at #379

zlakes01, 2023-07-31:
Solved. Thanks a lot.

closed 2023-07-31 by wills106