wills106/homeassistant-solax-modbus #483: Consumption Accuracy - apply offset?

solarjones, 2023-08-01:
Not a bug but an observation.

I've been comparing the kWh import & export totals (Grid Import Today & Grid Export Today) read in from the inverter (Solis) against those from the utility provider (Octopus) smart meter data. There's a consistent under read in HA, almost accurate but not quite and I obviously know the Solis inverter isn't as accurate as the smart meter itself. 

I'm wondering if there's a way of applying a multiplier (~1.05 in my instance), to raise the values being read in from the inverter into HA at a plugin level? Or am I missing something obvious?
Thanks!

solarjones, 2023-08-01:
Oddly the myenergi Harvi CT clamp is very accurate and positioned directly next to the Solis CT on the same wire.  While the 'inaccurate' Solis readings aren't a big issue, it's a shame there isn't an offset applicable somewhere to tune it.

alienatedsec, 2023-08-01:
Again, similar with my Zappi Grid CT clamp is much more accurate and shows those daily import closer to my Octopus one. Solis is more likely to show 0.0 import. 

[image]

[image]

[image]

closed 2025-10-03 by github-actions[bot]