wills106/homeassistant-solax-modbus #492: Cannot lower control setpoint value when using autorepeat, and decimal notation bug??

frank8the9tank, 2023-08-14:
**Describe the bug**
Unable to lower the control setpoint when auto repeat function is running.

I use my inverter as charger/discharger, I read my energy meter using DSMR, and based on those values i set the control point for charging and discharger, using the auto repeat function with battery control selected.
Range trough automation can be set to -3000 (discharge the battery), and 8000 (charger the battery) ( i have other inverter that generate the energy)
So basicly i read my energy meter and set every 5 seconds what needs to happens on the inverter. (it changes the control point value)
But sinds 2 day is can no longer lower the charging value for this control loop.
when i press the auto repeat button and have a control point value of -1000 for an 3600 seconds and then update the value for control point to -2000 it does not ge updated to the inverter.
Then i switch to disable and enable it again it working fine at -2000.
Strange thing is that i increase it to -1500 and inverter switches to -1500 and to -1000 and inverter to -1000
But never lower again. so -1000 to -2000 no longer works.
On positive side to looks to be working fine. (so charging the battery). increasing and decreasing looks fine.

My setup was stable running for 2 months at least. No changes made to automations or inverter.
I only update the solax integration when a update is available.

it almost looks like that is an internal cap on the control range when i press the autorepeat functions.
And this limit can only increase. so from -1000 to -500 but never back down.
is looks like to stick to the highest value is set in the negative range.
Before my bug i could increase and decrease the control setpoint with what every is in between -3000 and 8000.

Things i have tried. 
- Set the values by hand in autoreapet function works like described above, set value and disable and set new.
- Checked settings in inverter.
- Checked automations.
- Tried previous versions of the integration.

**Mandatory details**
1. Integration Version: 2023.08.3
2. HA Version: 2023.8.2
3. Inverter Brand: Solax hybrid x3 gen 4
4. Plugin Used: [homeassistant-solax-modbus](https://github.com/wills106/homeassistant-solax-modbus)
5. First 6 digits of Serial: H34A10
6. Connection Method: waveshare modbus tcp

**Detailed Error Log**
2023-08-14 08:23:15.645 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-08-14 08:23:27.883 INFO (MainThread) [homeassistant.bootstrap] Setting up stage 2: {'home_connect', 'nibe', 'min_max', 'switch', 'version', 'stream', 'file_upload', 'solaredge', 'entsoe', 'pi_hole', 'person', 'sensor', 'github', 'local_calendar', 'image_upload', 'stookalert', 'solaredgeoptimizers', 'frank_energie', 'timer', 'my', 'onvif', 'bmw_connected_drive', 'cert_expiry', 'system_health', 'whois', 'analytics', 'telegram_bot', 'zone', 'uptime', 'input_number', 'plex', 'hacs', 'knmi', 'google_translate', 'onboarding', 'roomba', 'forecast_solar', 'persistent_notification', 'homeassistant_alerts', 'default_config', 'template', 'wellbeing', 'auth', 'mqtt', 'application_credentials', 'schedule', 'config', 'season', 'script', 'esphome', 'daikin', 'input_text', 'lametric', 'dlna_dms', 'synology_dsm', 'openuv', 'openexchangerates', 'counter', 'dlna_dmr', 'input_select', 'zodiac', 'dsmr', 'conversation', 'zha', 'utility_meter', 'xbox', 'history', 'assist_pipeline', 'solaredge_modbus_multi', 'diagnostics', 'webostv', 'solax_modbus', 'shelly', 'input_boolean', 'stt', 'electrolux_status', 'media_source', 'wake_on_lan', 'sun', 'tag', 'mobile_app', 'group', 'repairs', 'search', 'unifi', 'sql', 'workday', 'device_automation', 'map', 'hardware', 'blueprint', 'husqvarna_automower', 'input_button', 'local_ip', 'ffmpeg', 'automation', 'buienradar', 'scene', 'co2signal', 'xiaomi_miio', 'moon', 'brother', 'media_player', 'lovelace', 'trace', 'tts', 'nibe_heatpump', 'energy', 'jellyfin', 'tasmota', 'radio_browser', 'logbook', 'input_datetime', 'comfoconnect', 'notify', 'zwave_js', 'dnsip', 'luftdaten'}
2023-08-14 08:23:34.421 INFO (MainThread) [homeassistant.setup] Setting up solax_modbus
2023-08-14 08:23:34.421 INFO (MainThread) [homeassistant.setup] Setup of domain solax_modbus took 0.0 seconds
2023-08-14 08:23:34.435 INFO (MainThread) [custom_components.solax_modbus.config_flow] detected HA core version 2023 8
2023-08-14 08:23:34.435 INFO (MainThread) [custom_components.solax_modbus.config_flow] starting configflow - domain = solax_modbus
2023-08-14 08:23:34.435 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solax
2023-08-14 08:23:34.446 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: trying to determine inverter type
2023-08-14 08:23:34.457 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX 0x0 serial number before potential swap: H34A10IA506142
2023-08-14 08:23:42.974 INFO (MainThread) [homeassistant.components.button] Setting up button.solax_modbus
2023-08-14 08:23:42.976 INFO (MainThread) [homeassistant.components.number] Setting up number.solax_modbus
2023-08-14 08:23:42.976 INFO (MainThread) [custom_components.solax_modbus.button] hub.wakeuButton: SolaxModbusButtonEntityDescription(key='battery_awaken', device_class=None, entity_category=None, entity_registry_enabled_default=True, entity_registry_visible_default=True, force_update=False, icon='mdi:battery-clock', has_entity_name=False, name='Battery Awaken', translation_key=None, unit_of_measurement=None, allowedtypes=0, register=86, command=1, blacklist=None, write_method=1, value_function=None, autorepeat=None)
2023-08-14 08:23:42.998 INFO (MainThread) [homeassistant.components.select] Setting up select.solax_modbus
2023-08-14 08:23:43.032 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solax_modbus
2023-08-14 08:23:43.049 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x0 seriesnumber to block with start 0x0
2023-08-14 08:23:43.049 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0x7d
2023-08-14 08:23:43.049 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x7d firmwareversion_invertermaster to block with start 0x7d
2023-08-14 08:23:43.049 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x81 firmwareversion_modbustcp_major to block with start 0x7d
2023-08-14 08:23:43.049 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x82 firmwareversion_modbustcp_minor to block with start 0x7d
2023-08-14 08:23:43.049 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x83 firmwareversion_manager to block with start 0x7d
2023-08-14 08:23:43.049 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x84 bootloader_version to block with start 0x7d
2023-08-14 08:23:43.049 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x85 rtc to block with start 0x7d
2023-08-14 08:23:43.049 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8b charger_use_mode to block with start 0x7d
2023-08-14 08:23:43.049 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8c manual_mode to block with start 0x7d
2023-08-14 08:23:43.049 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8d battery_type to block with start 0x7d
2023-08-14 08:23:43.049 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8e battery_charge_float_voltage to block with start 0x7d
2023-08-14 08:23:43.049 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8f battery_discharge_cut_off_voltage to block with start 0x7d
2023-08-14 08:23:43.049 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x90 battery_charge_max_current to block with start 0x7d
2023-08-14 08:23:43.049 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x91 battery_discharge_max_current to block with start 0x7d
2023-08-14 08:23:43.049 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x94 selfuse_nightcharge_upper_soc to block with start 0x7d
[... cut ...]
2023-08-14 09:46:48.955 INFO (MainThread) [custom_components.hacs] <Integration wills106/homeassistant-solax-modbus> Post installation steps completed

**Additional context**
A other thing i see is the following.
My whole system is in decimal and dot notation. So a decimal comma and a dot for every 1000's. like we have in the netherlands.
My laptop is this, my home assisitant is this. 
but i see this in the integration:
[image]
The control setpoint has a dot instead of a comma.
and the seconds also have this.
Don;t known is this is releated

Frank,

frank8the9tank, 2023-08-14:
looking trough the documentation i found:

1: remotecontrol_active_power: The target active power level. The current implementation will automatically limit the specified value into the range determined by the modbus registers active_power_lower and active_power_upper (readonly sensors that are updated by the inverter continuously).

2: When remotecontrol_power_control = Enabled Battery Control is chosen, the target value refers to the battery interface where positive values mean charge, negative values mean discharge. If target is set to zero, excess PV power will be sent to the grid.

So the control value is in between registers active_power_lower and active_power_upper.

But when i looks at those now in home assistant, they have strange values for the last 2 days:
[image]

Looking at this i see:
When the control loop is disabled, i see the active power lower being updated by the inverter,
But when i enable the loop is stop updating,

So could this be the reason why i cannot lower the control loop value, As is has to be in between the lower and upper, But lower never lowers again.

infradom, 2023-08-15:
I wonder if this is not related to a recent solax update. There is a similar bug report involving the remotecontrol power.
Which firmware versions are you running ? 
I hesitate to upgrade my inverter as I am running on remotecontrol almost all the time ...

frank8the9tank, 2023-08-15:
i have made a video about the problem,
My firmware is the following: DSP: 2.07, ARM 2.03
But i didn't update the firmware on my inverter, or did i notice did. 
Can solax do that without me knowing that?

https://github.com/wills106/homeassistant-solax-modbus/assets/84384888/408d42dd-3c40-480d-a6ba-bddbe4c41ccf

frank8the9tank, 2023-08-15:
The same for me, i run on remote control 24/7.

infradom, 2023-08-15:
I now also noticed you remark about the comma /  dot 
This is strange ...

frank8the9tank, 2023-08-15:
That control setpoint is something from my own Ha installation.
That is the number that i calculate based on my meter readings and other inverters, and that copies to the control setpoint from the solax integration.
It could be that the number are off by watt or kilowatt, i think i already was messing with thing be for i noticed the real problem.
But still that wrong dot/comma notation is there

infradom, 2023-08-15:
Thanks, we could remove the limiting done by the remotecontrol loop, and hope that solax limits itself to reasonable values. But I feel unsure about this. Maybe Solax will issue a bug correction. I will try to contact them, but as I have a bad flue these days, not sure when. ...

             ('remotecontrol_active_power',   max(min(ap_up, ap_target),   ap_lo), ),

frank8the9tank, 2023-08-15:
I think we at least should remover our own limit on that value, so we can see that the control value still gets accepted by the inverter, maybe as a beta? Can you do this?

infradom, 2023-08-16:
I find this dangerous, so I would not dare to test this on my inverter, unless solax declares it is fine.

infradom, 2023-08-16:
But if you are willing to take the risk: following line must be modfied in plugin_sollax.py:

         ('remotecontrol_active_power',   max(min(ap_up, ap_target),   ap_lo), ),

into

         ('remotecontrol_active_power',  min(ap_up, ap_target), ),

frank8the9tank, 2023-08-16:
in the mean time i have,
changed it to`('remotecontrol_active_power',   ap_target, ),` for testing.
And what i see now that i can use the control loop as before.
so increasing and decreasing works again this way in the negative range, so charging the battery.

And what i also see is then when i go from -1000 to -2000, then the value for the limit get updated afterwards .
So it looks like the integration has to change that is first set the new negative setpoint and than get a new limit from the inverter and it needs to round/limit to that. (instead of now check first for within limit and set new value).

I known for sure that are limit within the invert is self. example:
I also ways charge with 8kw to my battery to 100%
But the last 5% the inverter ramps this value down in steps when it gets closer to 100%, no matter what i set the charge value.
So i presume that do that also for the negative range (discharging)

And for safety when i copy my calculated setpoint from automation i round it with -3000 to 8000.

`service: number.set_value
data:
  value: >-
    {{ ([-3000, (states('sensor.solax_control_setpoint')|float ), 8000]|sort)[1]
    }}
target:
  entity_id: number.solax_remotecontrol_active_power`

infradom, 2023-08-16:
Thanks for testing this ! This makes me feel more at ease.
I am still a bit sick these days, so once I feel better I will probably adapt the code

frank8the9tank, 2023-08-16:
okay, will keep you updated.

frank8the9tank, 2023-08-30:
@wills106 What is the status on this? is this going to be fixed in coming updates? asking nicely.

wills106, 2023-09-03:
This should have been corrected in 2023.09.1?

frank8the9tank, 2023-09-04:
I have updated to the latest version, and indeed this is fixed, Many thanks!

closed 2023-09-04 by frank8the9tank