wills106/homeassistant-solax-modbus #493: Add X3 MIC 12 G2 Solax Inverter

wok18, 2023-08-15:
Hello,
I am not fully sure if this is the right way to add, please let me know if I need to do it differently. 

I managed to add my X3 MIC 12 G2 inverter by adding the line:
elif seriesnumber.startswith('MC212T'): invertertype = MIC | GEN2 | X3 # MIC X3 12 G2
In file plugin_solax.py after line 5314.

Would it be possible to add this to the next release?

thank you
Wolfgang

wills106, 2023-08-15:
Just released [2023.08.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.08.5)

wills106, 2023-08-16:
Closing down as [2023.08.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.08.5) includes 'MC212T'

closed 2023-08-16 by wills106