wills106/homeassistant-solax-modbus #495: [BUG] Wrong remote control command

jdedo, 2023-08-16:
After my inverter received an automatic update through the connected Pocket LAN Adapter, I noticed a new option for "peak shaving" in my Solax App. I've been using this integration for several months, and everything worked perfectly until this update. Now, there seems to be an issue. I have a simple automation that toggles between two modes: 'Enable Self Use' and 'Enable Feed-in Priority'. However, since the update, the inverter has begun importing power from the grid even when the battery is almost full. I've tried troubleshooting by checking all settings, reinstalling the integration, and restarting the inverter, but the problem persists. The debug logs suggest that an incorrect power command is being sent.

I'm very thankful for the excellent integration you've provided, I would greatly appreciate your assistance with this issue.

**Mandatory details**
1. Integration Version: 2023.08.5
2. HA Version:2023.8.2
3. Inverter Brand:Solax
4. Plugin Used:
5. First 6 digits of Serial:H34A15
6. Connection Method: Modbus TCP with Waveshare RS485 to RJ45 Ethernet

**Detailed Error Log**
[custom_components.solax_modbus.plugin_solax] peak shaving: old_ap_target:-2968 new ap_target:-3382.0 max: -3382.0
[custom_components.solax_modbus.plugin_solax] Evaluated remotecontrol_trigger: corrected/clamped values: [('remotecontrol_power_control', 'Enabled Power Control'), ('remotecontrol_set_type', 'Set'), ('remotecontrol_active_power', -924), ('remotecontrol_reactive_power', 0), ('remotecontrol_duration', 20)]
[custom_components.solax_modbus] ready to repeat button remotecontrol_trigger data: [('remotecontrol_power_control', 'Enabled Power Control'), ('remotecontrol_set_type', 'Set'), ('remotecontrol_active_power', -924), ('remotecontrol_reactive_power', 0), ('remotecontrol_duration', 20)]
[custom_components.solax_modbus] Ready to write multiple registers at 0x7c: [1, 1, 64612, 65535, 0, 0, 20]

infradom, 2023-08-17:
Seems like a side effect of the software updates solax has been pushing out. 
Issues #488 and #492 are also related to his.
There is a 'simple solution' but we hesitate to push it until further tests are successful
We know this is annoying. For the ones with more technical background: a quick fix is suggested in issue #492 
We hope to implement this in a release soon, but to not want to take any risks

jdedo, 2023-08-23:
@wills106 
It would be great if you could publish a beta version; I'll test it. I have some concerns given the system's criticality. Any advice or things I should be aware of? Thanks for pushing this forward!

wills106, 2023-09-01:
Should be corrected in [2023.09.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.09.1)

jdedo, 2023-09-02:
@wills106 thanks for the update

I tested the remote trigger this morning and noticed that the grid import was much lower than before. However, it's still higher compared to when I switch the inverter to FeedIn mode without using the remote trigger.

I've also observed another strange behavior, as you can see in the chart: when the remote trigger is activated in 'FeedIn' mode, the Grid Export continuously decreases."
[image]

[image]

I hope this information is helpful. It would be great if you could assist me in fixing these issues. Thanks!

iMatte81, 2023-09-25:
Hi,
I've solved my issue using the function Enabled Power Control in Power Control of the master inverter and after calculate the right value of remotecontrol_active_power (by home assistant), this value is:

remotecontrol_active_power = house_load - pv_power_total

but with inverter master and slave system:

house_load = house_load_master + house_load_slave
pv_power_total = pv_power_total_master + pv_power_slave

In this way I can control correctly my plant.

During the configuration of the two inverters with the solax integration I have selected for both the inverters the checkbox Parallel Mode (Master-Slave), is it right?

If I use the enabled_feed_in of the solax integration, the remotecontrol_active_power calculated by the integration for the inverter master is not the total power but the only power of this inverter while my system needs the total power of the two inverters for working correctly.

infradom, 2023-09-25:
In order to be able to implement this in the source code, yould you answer follwoing question (I am not familiar with the master/slave concept):
- how do you red the pv_power of the slave ? which entity?
- same question for the inverter_load of the slave (that entity is used to compute house_load_nett)

iMatte81, 2023-09-25:
> In order to be able to implement this in the source code, yould you answer follwoing question (I am not familiar with the master/slave concept):
>
>     * how do you red the pv_power of the slave ? [...]

1. I read the power of the slave from the slave directly, using pv_power_total. To calculate the total power of the plant, I add the pv_power_total of the master and pv_power_total of the slave
2. The same as before, I read the parameter house_load and I add master and slave readings

peimar_potenza_ftv:
    unit_of_measurement: "W"
    value_template: '{{ states("sensor.peimar_10kw_pv_power_total" )| float + states("sensor.peimar_6kw_pv_power_total") | float }}'

  peimar_casa_load:
    unit_of_measurement: "W"
    value_template: '{{ states("sensor.peimar_10kw_house_load" )| float + states("sensor.peimar_6kw_house_load") | float }}'

I have configured 2 inverter in the Solax integration

[image]

[image]

wills106, 2023-09-25:
Just so you know I released [2023.09.6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.09.6) this reverted houseload back to the old style of calculation which I find more accurate and you don't get spikes from Solar affecting it.

> During the configuration of the two inverters with the solax integration I have selected for both the inverters the checkbox Parallel Mode (Master-Slave), is it right?

If the Inverters are in a true Master/Slave setup you only need to add in the First Inverter with PM selected. It should then read in the second Inverter. But you only get a limited set of data.

Are all the Inverters behind the same Modbus Power Meter?

We might be able to read the measured power (Indicated from the Power Meter) and deduct the Current from Each connected Inverter as that data is per phase is given and then calculate a global houseload from that. We know how many Inverters are connected from a register that is used in PM

wills106, 2023-09-25:
Are you able to give a screenshot of all the entities starting PM on the Master Inverter.

wills106, 2023-09-25:
Just so you know I am going to add in PeakShaving later this week.

iMatte81, 2023-09-25:
> Are you able to give a screenshot of all the entities starting PM on the Master Inverter.

[image]

[image]

[image]

[image]

wills106, 2023-09-25:
I meant with the Values sorry, aren't they listed like:

[image]

On the Integration page?

iMatte81, 2023-09-25:
> Just so you know I released [2023.09.6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.09.6) this reverted houseload back to the old style of calculation which I find more [...]

The inverter are all behind the same modbus power meter connected only to the master one

I have updated this morning to the new version, but the result are the same, if I set feed in priority the system always charge the battery.

iMatte81, 2023-09-25:
[image]

[image]

[image]

[image]

wills106, 2023-09-25:
If you manually enable "PM Inverter Count" does it display 1 for 1 slave. Or does it display 2 for Master and Slave?

We could use that to know how many Inverters are connected for a calculation. But we wouldn't know how you have named them.

> house_load = house_load_master + house_load_slave
> pv_power_total = pv_power_total_master + pv_power_slave

Perhaps we need a text entry to input your templated sensors? Unless there is enough info from all of the PM sensors?

iMatte81, 2023-09-25:
> If you manually enable "PM Inverter Count" does it display 1 for 1 slave. Or does it display 2 for Master and Slave?
>
> We could use that to know how many Inverters are connected for a calculation. [...]

[image]

The names are:
- Peimar_10kW
- Peimar_6kW

wills106, 2024-02-20:
Just wondering if this is an active issue. Or has it been worked around or resolved?

wills106, 2024-08-03:
Closing down as no response.

closed 2024-08-03 by wills106