wills106/homeassistant-solax-modbus #502: Growatt  unrecognized inverter type MIC 2500tl-x

super-bedst, 2023-08-20:
Please add this 1 phase Growatt inverter MIC 2500tl-x it is without battery and 1 mppt serial is : FVJ4CD708B

wills106, 2023-08-22:
Is that what the Integration returns in the error log?
(f"unrecognized {hub.name} inverter type - firmware version : {seriesnumber}")

Or the actual Serial Number? On the Growatt it's using the firmware version and not the serial number.

super-bedst, 2023-08-22:
Here is the string from the log
unrecognized Growatt Mic inverter type - firmware version : GH1.0ZAAA

wills106, 2023-08-22:
Just released [2023.08.6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.08.6)

wills106, 2023-08-29:
Is your Inverter working now?
Can I close this down?

wills106, 2023-09-03:
Closing down as Inverter should be supported now.

closed 2023-09-03 by wills106

super-bedst, 2023-09-03:
HiYes it works great 😊 MobilmailMed Venlig HilsenPeter FriisMobil 20654196

wills106, 2024-02-09:
@super-bedst Is your MIC Inverter reporting Grid Current, Grid Power, Grid Power correctly?

Or do you have the same scaling issue as #727 

```
Grid Current: 0.8A >> 0.08A
Grid Power: 93W >> 9.3W
Grid Power VA: 118 VA >> 11.8 VA
```